FormerLurker/Octolapse #2: Plugin doesn't activate

nbun, 2018-01-20:
I have installed the plugin and followed the instructions, but I must have missed something because I can't get it to work. I restarted Octoprint and enabled the plugin, changed some printer settings and enabled test mode. When starting a print, the printer still heats up and doesn't pause between layers. The printer is a Prusa MK2s.

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/1649220/plugin_octolapse.log)

[10mm_calibration_cube.zip](https://github.com/FormerLurker/Octolapse/files/1649221/10mm_calibration_cube.zip)

[settings.zip](https://github.com/FormerLurker/Octolapse/files/1649222/settings.zip)

FormerLurker, 2018-01-20:
I'm still looking into this.  I noticed a few other issues along the way:

The 'Embed Watermark' option will prevent rendering.  I think I'm missing a resource file for that.

The floating point math issue was preventing detection of the IsExtruding and IsDetracted events.  I finally got around to rounding the extruder values when checking for state changes, and will check this in once I've finished testing.

Regarding the actual issue, I am thinking it is an initialization issue with the settings file itself.  I'm testing this now and will report back soon.

THANKS so much for the error reporting, obviously there are some bugs to work out.

nbun, 2018-01-20:
The 'Restore All Default Settings' button didn't do much, but there are different error messages in the log:
```
2018-01-20 20:45:58,179 Printer event received:PositionUpdate.
2018-01-20 20:45:58,180 Position received by Octolapse while paused, but was declined.
```

FormerLurker, 2018-01-20:
I've started a development branch and have pushed some changes.  There is definitely something fishy happening during initial install that is causing problems.  Maybe try reinstalling from this url: https://github.com/FormerLurker/Octolapse/archive/development.zip

Make sure to do a ctrl+f5 to clear octoprint's cache before changing any settings.  I'll let you know when I've tested the new build.

FormerLurker, 2018-01-20:
OK, I just fixed another small issue and ran a brief test print on my machine and it seems to be working for me.  Feel free to install it again at this link:  https://github.com/FormerLurker/Octolapse/archive/development.zip

FormerLurker, 2018-01-20:
Ok, My extended test failed big time.  Maybe don't bother trying it again until I run a battery of tests to figure out what's going on.

FormerLurker, 2018-01-21:
I may have fixed it.  Will test later tonight.  Late for a dinner.

nbun, 2018-01-21:
No success so far. I could try a clean Octopi image to see if any of my Octoprint settings interfere with the plugin, but I don't have time to do it right now. Any shell commands that would help to find the issue?

FormerLurker, 2018-01-22:
Hey, you wouldn't happen to be trying to print from the SD card, would you?  Octolapse can only work if the files are printed locally, not from the SD card on the printer.  I was working on this all weekend then it dawned on me.  I didn't even write anything about this in the readme.md!

I'm testing a print with the latest build (on the master branch now), and will probably add a popup warning if printing from SD.  What an important feature that I overlooked!

nbun, 2018-01-22:
No SD card in the printer, sorry. I tried the newest version but everything still heats up, so I canceled the print immediately. I don't know if anything interesting is in the log, so just in case: 
[plugin_octolapse(8).log](https://github.com/FormerLurker/Octolapse/files/1652498/plugin_octolapse.8.log)

nbun, 2018-01-22:
Alright, the issue still persists, but I did some debugging myself and hopefully I can point you in the right direction. [I added a few debug lines](https://github.com/nbun/Octolapse/blob/master/octoprint_octolapse/__init__.py#L262) and the log contains:
```
2018-01-22 19:41:50,174 OnPrintStart #1
2018-01-22 19:41:50,174 OnPrintStart #2.5
2018-01-22 19:41:50,177 OnPrintStart #3.5
2018-01-22 19:41:50,185 OnPrintStart #4.5
```
Since "4.9" doesn't appear in the log, `StartTimelapse` appears to never return. I'll do the same in `StartTimelapse` to hopefully find the root cause.

FormerLurker, 2018-01-22:
Excellent.  Looks like a silent failure somewhere in timelapse.py

FYI, if you're having trouble getting the _logger in timelapse.py, use this instead:  self.Settings.CurrentDebugProfile().LogError("Error Message Here")

It will log as long as logging is enabled for the current debug profile.

nbun, 2018-01-22:
I used the unconditional logging method since I suspected a problem with the profiles, but it doesn't look like that's the case. My improvised exception logging tells me `g90InfluencesExtruder` and I know that it reaches [this statement](https://github.com/nbun/Octolapse/blob/master/octoprint_octolapse/__init__.py#L320), so my best guess is that something is funky with `self._settings.settings.get(["feature"])["g90InfluencesExtruder"]`. It's getting late and programming in the Github editor isn't the greatest thing, so I'll try again tomorrow. You should definitely look into exception logging for the event handlers, as this would simplify finding issues like this significantly.

FormerLurker, 2018-01-22:
Excellent, I can deal with that pretty easily.  I need to dig into the source code for the Octoprint settings, cause I've had lots of problems.

I will take your advice regarding the exception handling.  I'll push these two changes ASAP.

FormerLurker, 2018-01-22:
New build checked in on master.  Be sure to ctrl+f5 after Octoprint reboots since it seems to be caching settings client side even after a restart.

So, I changed the way I access settings from octoprint's built in object.  I'm still learning, obviously, and didn't realize there settings.get can take multiple keys at once to retrieve nested settings.  If you use that form it returns a default value.

I can't for the life of me observe the same issue you are seeing with the g90InfluencesExtruder option (in the Octoprint settings within Features at 'G90/G91 overrides relative extruder mode Smoothieware'), but I'm pretty sure any problems there should go away now since I use the default (False) if an exception is thrown trying to retrieve the setting.  That combined with the proper version of settings.get should at least get you to the next issue ;)

Also, I started adding exception handlers around the various events as you suggested.  It's a start!

Thanks again for your assistance!

nbun, 2018-01-23:
It finally works, awesome! That was a tough one to find, now the real testing can begin :grinning:

closed 2018-01-23 by nbun

FormerLurker, 2018-01-23:
Excellent!  Thanks again for helping me to track this down!

FYI, I just made some pretty major changes to the position/extrusion state monitoring as well as the snapshot triggering mechanism.  It should now only retract/lift as much as it has to in order to achieve the printer lift and retraction distances.  Please let me know if you notice any incomplete/excessive retraction/lifting, or any double lift/retraction.

Sandman0815, 2018-03-25:
Hallo There, I'm using the Creality Ender 2 Printer.
Is there a way for me to Use Octolapse with my Printer?
I Tryed it out with CR 10 but it doesnt work :-(

Please can you help me ?
Sandman