FormerLurker/Octolapse #5: Add units to number input fields

nbun, 2018-01-23:
Especially the printer configuration page has many number input fields like retraction length that need a unit to make sense. The Octoprint printer configuration page has input boxes with units, so I assume these could be used here?

FormerLurker, 2018-01-23:
That's a good idea.  I'll mimic the Octoprint styles.  Marking as enhancement.  I'll add soon and will commit the changes.

FormerLurker, 2018-01-23:
Ok, I've added units to the menu options where it seemed appropriate.  I also re-arranged and re-worded a bit.  I'm open to suggestions if you see any settings that are confusing, or could use better names/descriptions.

I'll leave the issue open.

FormerLurker, 2018-01-29:
I'm going to go ahead and close this.  If you think other menu items should be renamed/made clearer, or if you notice any missing/buggy validations, please open a new ticket.

closed 2018-01-29 by FormerLurker