FormerLurker/Octolapse #6: Time lapse comparison

nbun, 2018-01-26:
I did my first real print and it worked quite well. The only issue that I ran into was that the camera lags behind a bit because of the limitations of the Pi, so I increased the time to wait for the photo to 1 second. I also created a "on z change" time lapse for comparison.  Both look pretty bad image quality wise because my enclosure isn't lit very well and thus the ISO value needs to be pretty high, but your plugin does a really good job of moving the head to the exact same spot without much overhead.

[Comparison.zip](https://github.com/FormerLurker/Octolapse/files/1668901/Comparison.zip)

FormerLurker, 2018-01-27:
Wow, that looks great!  Do you know what framerate your camera is set to?  I'd recommend setting the delay to 1/framerate, which should be all the delay you need. 

It looks like something is up with the z-axis stabilization, because it shouldn't be bouncing up and down like that.  I have done a bunch of updates in the last week, so if you haven't done so, try upgrading to the latest version on the master branch.  It could fix the bouncy z axis.  

I've added an optional navbar icon, as well as a tab on the main page that displays status, and allows you to stop the timelapse mid print.  I've also updated the settings so that they only load once a user is signed in, added access restrictions to all web-service calls, and fixed cross client synchronization, in case there are multiple clients.  The menu has changed somewhat as well.  Let me know what you think.

Thanks again for the testing and great results!  Here is a link to a recent timelapse I made for comparison:  https://youtu.be/er0VCYen1MY

nbun, 2018-01-27:
The most recent version produces really great results! [https://youtu.be/2cSztSnohZQ](https://youtu.be/2cSztSnohZQ) I changed a few settings to get rid of the noisy image and it's perfectly smooth now. Really mesmerizing to watch.

FormerLurker, 2018-01-27:
Wow!  That is an awesome video.  I'm so glad it worked so well!  I'd like to use your video, if you don't mind.  Also, if you had the Octoprint default timelapse video, I'd love that as well, for comparison.

Regarding the firmware bug, I've been battling with it.  Prusa has confirmed the bug.  They are going to fix it first on the MK3, then will release a patch for MK2.  I've been avoiding it by not using the linear advance profile in slic3r and by manually changing the 'Start G-Code' in the filament settings to: M900 K0

I've not had any lockups or out of order execution since.  I really hope this gets fixed soon.

nbun, 2018-01-27:
I forgot to enable the standard timelapse function for this one, but I'll send you both versions for upcoming prints. If the MK2.5 upgrade is released in the next days, I'll have to do a lot of printing 😀

closed 2018-02-12 by FormerLurker