FormerLurker/Octolapse #13: Support non-cartesian printers

FormerLurker, 2018-02-12:
These are not currently supported.  I hope to include support for these at some point, but I don't own one to test on currently.

FormerLurker, 2018-03-30:
At least one user has actually gotten one of these to work!  I added the beta profile to v0.2.4

FormerLurker, 2018-04-03:
Note:  Octolapse cannot yet determine in or out-of-bounds positions when using a circular bed.  I understand lots of deltas have these.  For now you will need to use your bed diameter and define a rectangle of this size within your printer volume settings (either in the Octolapse Printer Profiles or within the OctoPrint printer profile, depending on your Octolapse Printer Profile settings).

etrigan63, 2018-10-24:
> Note: Octolapse cannot yet determine in or out-of-bounds positions when using a circular bed. I understand lots of deltas have these. For now you will need to use your bed diameter and define a [...]

By this do you mean a rectangle that is circumscribed by the bed diameter (ie, fits inside of the circle) or a rectangle that contains the bed diameter (in essence a square that is as big as the circle)?

FormerLurker, 2018-10-24:
You'll need to use a square that completely encloses the circle.  However, I strongly recommend that you use the 'center' stabilization since some of the other stabilizations (back left, back right, front left, front right) would attempt to put your extruder outside of your bed which could be bad depending on your firmware (note that some firmware will not allow the nozzle to go out of bounds, which would lessen the severity of the issue).

I can't believe I haven't fixed this yet.  Soon I'll add a drop down box for 'bed shape' that should take care of this once and for all.

Thanks for your post!

FormerLurker, 2019-10-12:
This has been solved and will be included starting with V0.4!!!

closed 2019-10-12 by FormerLurker