FormerLurker/Octolapse #15: Test Center Orign Printer

FormerLurker, 2018-02-12:
Some printers have their origin's in the middle of the build plate.  This is another thing I learned while writing this plugin.  I'm pretty sure these will work if the settings are correct, but I don't have one to test it out.

Shadowen, 2018-05-02:
Just a note: OctoLapse doesn't like negative coordinates. I found this out when I configured my dual extruders the wrong way.

Then again, I'm not exactly sure if OctoLapse even knows that I have multiple toolheads. Perhaps a separate issue.

FormerLurker, 2018-05-02:
It is likely due to the dual extruders.  In order to get these working, I need to add tool tracking and multiple extruder states as well as some additional gcode parsing.  I think this is on the feature list, and I've already implemented a new parser to help get it implemented.

That being said, I can't be 100% sure unless I know exactly what happened.  If you want, send me your gcode file and I'll take a look when I have time.  I've been swamped for the last couple weeks, but hopefully I'll have a few evenings or weekends soon.

Shadowen, 2018-05-02:
It's not a big deal for me since I've fixed my firmware since then. No more negative coordinates. I just wanted to make a note for anyone dealing with center origin printers that OctoLapse seems to error out when any coordinate is negative.

FormerLurker, 2019-10-12:
This has been fixed and solved.  Octolapse now supports center origin printers (and circular beds), so...  CLOSED!!!  Wood :)

closed 2019-10-12 by FormerLurker