FormerLurker/Octolapse #21: Create a default profile for the Prusa MK3

FormerLurker, 2018-02-13:
After receiving word that Octolapse is working with at least one MK3, I want to add a default profile.  I just need the settings!

danecreekphotography, 2018-02-23:
Just by looking at Slic3r with the Prusa i3 profile provided with the printer here's what I've come up with so far:

Retraction Length: Printer Settings > Extruder 1 > Retraction > Length: 0.8mm
Z-Hop Height: Printer Settings > Extruder 1 > Retraction > Lift Z: 0.6mm (is this the right property?)
Movement Speed: Print Settings > Speed > Speed for non-print moves > Travel: 250 mm/s * 60 sec = 15000 (???)
Retraction Speed: Printer Settings > Extruder 1 > Retraction > Retraction Speed: 35 mm/s * 60 sec = 2100
Detraction Speed: Printer Settings > Extruder 1 > Retraction > Detraction Speed: 0 mm/s * 60 sec = 0 (???)
Z Movement Speed: ???
Auto-Detect Home Position: enabled
Override Octoprint Printer Profile: disabled
Abort Snapshot When Out-Of-Bounds: enabled
Printer Position Tolerance: 0.1
Snapshot Command: snap

I'm not sure where to get the Z Movement Speed from in Slic3r. Also does a movement speed of 15000 seem reasonable?

FormerLurker, 2018-02-23:
Retraction length looks good.  Lift looks good.  250mm/s sounds awfully
fast for movements, especially if you're printing something big.  It is
possible, but sounds high.  I'd cut it in half and watch your printer to
see if it's too slow. Retract speed looks good.  Set detract to the
retraction speed (that may be what 0 means in that case, use previous
speed).

I think z speed is somewhere odd.  Maybe with z lift?  If you can't find
it, send me your profile.  You can try something slowish if you want to be
safe (4000).  This controls the speed of z travels when lifting and moving
back.

danecreekphotography, 2018-02-23:
This is an export from Prusa's version of Slic3r with the 0.15mm OPTIMAL MK3 print settings, Prusa PLA filament settings, and Original Prusa i3 MK3 printer settings selected.

[Original Prusa i3 MK3.zip](https://github.com/FormerLurker/Octolapse/files/1750643/Original.Prusa.i3.MK3.zip)

FormerLurker, 2018-02-23:
Wow that printer is fast!  Looks like your original movement speed (travel
speed, I need to rename that field) was correct.

Not sure about z axis speed.  When I figured this out for my printer, I
guess I looked at some sliced GCode to find a z hop and used the f value.
4000ish would work fine to get you started.  Judging by the speeds in that
profile, you might be able to go quite a bit higher, but it won't make all
that much difference.

Tomorrow I'll import that profile, slice something and take a look at the
speeds.  Getting pretty late here now unfortunately.  I'll mail you the
results.  Then I'll add the new profile to the default settings.

Let me know how it goes!

danecreekphotography, 2018-02-23:
Ran a print last night, here's what the timelapse looks like. I've attached the printer profile I used in octolapse.

[octolapse video.zip](https://github.com/FormerLurker/Octolapse/files/1751928/octolapse.video.zip)

[Settings.zip](https://github.com/FormerLurker/Octolapse/files/1751925/Settings.zip)

FormerLurker, 2018-02-23:
That's so cool, and thanks for the settings!  I'll take a look ASAP and will add this to the default settings json.

FormerLurker, 2018-02-27:
Default settings for the MK3 have been added, but z-speed and detract speed need to be researched.  The movement speed has been set to 'stealth mode' speed.

I'm changing this to 'Needs Testing' and 'Help Wanted'

Make sure you restore the default settings, restart octoprint, then clear your cache (ctrl + F5), else things won't work right due to a bug in the restore default settings routines.

adam6806, 2018-02-28:
Hey @FormerLurker it's PirateOfDW from Reddit. Sorry it took so long to get around to sending my settings but here they are.
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/1765348/Settings.zip)

FormerLurker, 2018-02-28:
Hey man, thanks!  Where did you find the speed for z-hop(lift)?  All of our other settings pretty much agree, which makes me feel a lot better.  I chose to use the stealth mode travel speed, but it's not much different than yours.

I think the 0 detract speed should work (use previous axis speed), but I was thinking it would probably be better to set it to the retract speed anyway (funny, the mk2's retract and detract are different speeds).

Unrelated - Have you tried out the new beta development branch?  Lots of bug fixes, some optimizations, new features (see issue #40 )  I've run lots of prints through the virtual printer, but have only run prints in test mode on my actual printer so far.  I want to lock this down and push V0.2 ASAP so I can work on the no-pause snapshots.  Let me know if you want to try it out and test drive some features.  

Later, and thanks again!

FormerLurker, 2018-03-03:
I believe this issue is closed!

closed 2018-03-03 by FormerLurker