FormerLurker/Octolapse #25: Fix All Tests

FormerLurker, 2018-02-15:
After a major rewrite I got lazy and didn't update my tests, now most of them are completely broken.  Fix all these tests and keep them up to date, especially since the code is starting to stabilize.

Shadowen, 2018-02-24:
Hey, I'm going to fix up your existing tests! No guarantees on any new tests being written though...

FormerLurker, 2018-02-24:
You rock!  Today has been a good day.

FormerLurker, 2018-02-24:
You might want to hold off fixing any tests of timelapse.py, btw.  It's going to change a lot soon when I remove the pause/resume from the snapshot process.

Shadowen, 2018-02-24:
Can you please post the contents of `c://test`? I'm on a Ubuntu machine and this is completely breaking most of your tests anyways. I'll see if I can trick it with some temp directory magic.

Brb, reading up on how settings are stored to begin with.

FormerLurker, 2018-02-24:
It's just a log path.  We can change it to a a subdirectory of test and add to gitignore?

Shadowen, 2018-02-24:
I need an Octolapse profile. A lot of the tests are erroring out because the printer homes, but all positions still return `None`.

In `position.py`, the plugin detects `G28 Home` and does:
```
pos.X = self.Origin["X"] if not self.Printer.auto_detect_position else None
```
`self.Printer.auto_detect_position = True` (I think this is correct, as default). So it sets `pos.X` to None. And no homed position/coordinates is ever generated. Am I missing something?

FormerLurker, 2018-02-24:
I see.  I'd probably just test to make sure the coordinates are set to null
after a g28 with autodetect position enabled.  Timelapse.py is responsible
for delivering the actual position in that case, I believe via a call to
update position.

If you want to test position after a home, you could do something like this
to set the home position:
self.Position.UpdatePosition(x=x,y=y,z=z,e=e,force=True,calculateChanges=True)

Or, set self.Settings.CurrentPrinterProfile().auto_detect_position = false
and set the origin, also in the printer profile.

Let me know if that made sense.  I need to get some sleep, but I'm going to
be working on this probably most of tomorrow.

Shadowen, 2018-02-24:
I suppose I should give a clearer example:

In `test_trigger_gcode.py`
```
		# home the axis and resend the snap command - will be false because the PREVEIOUS state must be homed
		position.Update("G28")
		trigger.Update(position, "snap")
		self.assertFalse(trigger.IsTriggered(0))
		self.assertFalse(trigger.IsWaiting(0))
```

The first `trigger.Update(...)` parses the home command, so following commands should be homed.

The test continues with:
```
		# try again, should now work
		position.Update("Snap")
		trigger.Update(position, "snap")
		self.assertTrue(trigger.IsTriggered(0))
		self.assertFalse(trigger.IsWaiting(0))
```
Somewhere in the `trigger.Update(...)`, we have the following `if-else`:
```
			if(not position.HasHomedPosition(1)):
					... (skip command)
			else:
					... (parse command)
```

However, it appears `if(not position.HasHomedPosition(1))` is `True` (branch is taken), so it never even attempts to parse the command. Is this working as intended?

FormerLurker, 2018-02-24:
Ok, so right before line 50 (# try again, should now work)

Add this to simulate a position update :

position.UpdatePosition(x=0,y=0,z=0,e=0,force=True,calculateChanges=True)

You could also change the settings in setUp()

Self.Settings.CurrentPrinterProfile().auto_detect_position = False
Self.Settings.CurrentPrinterProfile().origin_x =0
Self.Settings.CurrentPrinterProfile().origin_y=0
Self.Settings.CurrentPrinterProfile().origin_z=0

FormerLurker, 2018-02-24:
I have fixed most of the tests that were throwing exceptions.  Most of this (and probably most of the other breaking changes) was due to the addition of the circular buffers you noticed.  These were added to position.py and extruder.py so that I could easily 'Undo' position and extruder updates.  I figure they might be useful in the future to allow for more complicated triggers/snapshot gcode.  I hadn't planned initially on doing this, but realized at some point that it was necessary.  You can imagine how much that change affected everything.

Since you mentioned working more on the tests, I'll push these changes as soon as I run a couple quick test prints.

Shadowen, 2018-02-24:
Would be great! I'm mostly looking at `position.py`. By the way, the structure I'm looking at using is [`collections.deque`](https://docs.python.org/3/library/collections.html#collections.deque).

FormerLurker, 2018-02-24:
Ok, Just this AM i standardized most of the calls to Position that retrieve something from the PositionState (GetPosition), and I did something similar to extruder.py (GetState).  I didn't change it everywhere, but hopefully that will make what you're doing easier.

BTW, I'm having some trouble with Git recognizing my credentials.  This has happened before and it's something to do with the credential manager.  I'll let you know when I've pushed the changes I've been making this AM.

FormerLurker, 2018-02-24:
K, those changes are in BetaDevelopment now after applying a patch to my IDE (apparently there was an issue with TLSv1.2).  

I'm really sorry if I stepped on your toes with this update, but I didn't expect us to both be working in the same file (position.py) this morning!

FormerLurker, 2018-02-24:
I'm super confused.  I pushed those changes but I don't see the commit on github.com.  However, when I clone betadevelopment, the changes are there!

Can you let me know if you see this line in utility.py after pulling the latest code from BetaDevelopment branch?

Line 281:  return None if v is None else min(max(v, v_min), v_max)

If it's there I"ll assume it's some glitch on github, otherwise I messed something up somehow.

Shadowen, 2018-02-24:
Looks fine on my end.

FormerLurker, 2018-02-26:
Aside from test_timelapse.py (which will be a real bear), there are 7 tests left to fix in the following files:

test_utility.py - 2 fail
test_timerTrigger.py - 4 fail
test_layerTrigger.py - 1 fail (I thought I fixed this one...)

Getting close!

FormerLurker, 2018-02-27:
All of the tests except those from test_timelapse.py are fixed!  There are probably some flawed tests in there, but we'll find them eventually.  At some point many probably need to be re-written, split or removed.  I'm going to create a separate issue for test_timelapse.py, since it will probably need to be completely rewritten anyway, but probably won't do this until timelapse.py is re-written to include Gina's newly implemented lock that will allow me to take snapshots without pausing the print.  I fear we will need a testing master to write useful tests for timelapse.py, though....

FYI, I changed trigger.py so that it looks at the current trigger state instead of the previous to determine when to trigger.  Then in timelapse.py I look at the PREVIOUS trigger state.  This REALLY simplified a lot of the tests by preventing me from having to send a dummy command (or other such nonsense) to read the previous trigger state.  Honestly I have no idea why I did it that way in the first place, but it's fixed now!

I deem this issue CLOSED!  Thank you so much for your help.  I was DREADING fixing these up, but knew it would be good medicine.

closed 2018-02-27 by FormerLurker