FormerLurker/Octolapse #26: Sometimes fails to capture any snapshots

djeddleman, 2018-02-19:
Sometimes Octolapse fails to capture any snapshots. After completing the print, the following error appears:

_Octolapse has failed to render a timelapse. Reason:No frames were captured._

I have attached a log file where Octolapse properly captures snapshots until the print is cancelled and restarted. During the second print, no snapshots are captured. 
[plugin_octolapse_fail2.log](https://github.com/FormerLurker/Octolapse/files/1737766/plugin_octolapse_fail2.log)

I have not been able to determine the conditions where Octolapse captures snapshots properly and when it does not. If I can help by performing any tests, please let me know.

FormerLurker, 2018-02-19:
djeddleman,

Thanks for the issue report!  I'm so glad to get them so others don't have to experience the problems you have.

I've reviewed the log, and cannot see any obvious errors, but I may have already fixed this bug in the latest development build.  I will try the steps that caused your issue (start print, cancel, try again) on both the current beta build and the development build to see if it's been fixed, or if this is a new issue.

In the meanwhile, perhaps turn on 'snapshot save' and 'snapshot download' logging.  If you want to do a timelapse between now and when I fix this issue, reboot Octoprint right before starting your next print and after cancelling (that is really a terrible workaround, I'll admit, but it's the only way to be sure).  

This will be my priority issue, fyi, since it's such a terrible bug.  Not sure how I missed this while testing since I cancel more prints than I complete when testing stuff, but it's probably because of an issue I had fixing 'restore default settings' that was requiring a reboot.  boo.

FormerLurker, 2018-02-19:
I'm a doofus.  You had all of the logging enabled, including snapshot download/save, but I started from the bottom of the log and didn't see it, even though you mentioned it.

FormerLurker, 2018-02-20:
I believe I have tentatively confirmed that the bug has been fixed already in the Beta build, but will wait for confirmation from you before closing the ticket.  I've run several tests within the virtual printer and now three tests on my actual 3d printer.  I've tried canceling at all of the possible important moments: before a snapshot, during a snapshot, after a snapshot.  I've also tried all of these after several snapshots have been taken, and in each case things continued to work after the next print!  The master branch failed a few of these tests, though I can't pinpoint exactly what is failing and why.

In order to make this build usable, I had to hastily finish up some new items that were in-process:  Additional location detection commands in the Printer Profile settings, and Position Restriction in the Snapshot Profiles (layer trigger setting).  Please don't use these yet, especially since the client side validations have not been implemented all the way.  I have tested the rectangular position restrictions (x1,y1,x2,y2)on my printer and they seem to work, but have not yet tested the circular ones (x,y,radius).  In any case, they are not battle hardened.

You should notice better pop-up notifications for the renderings within the GUI.  There was some re-work of the pre and post-roll processing, which was causing some issues in some cases (this could actually have been the root of your error, if you were using pre/post roll), and the rendering process/notifications in general.

Here is a link you can use to install the new development branch:   https://github.com/FormerLurker/Octolapse/archive/BetaDevelopment.zip

You shouldn't need to uninstall the plugin before you reinstall (same steps as in the readme.md file on the repo home page), but make sure you clear the cache after you install, or else you may be using old javascript/html due to caching in octoprint (ctrl+f5 after the required reboot should do the trick).

Let me know if you have any additional issues, and thanks again for submitting the error report!

djeddleman, 2018-02-20:
Great. I'll try it out tonight. 

Before I install the development build, I thought it would be good to let you know that I've attached a log file to this message that shows a print completing and the snapshots were taken properly. Then, I ran the print again, and the second time it failed to take the snapshots. This confirms that cancelling the print did not cause the error. Both the first good print (snapshots taken) and the second print (snapshots failed) are included in this log file.
[plugin_octolapse (4).log](https://github.com/FormerLurker/Octolapse/files/1740689/plugin_octolapse.4.log)

Hopefully, the new build will solve this bug. This is an incredible plug-in. It would be an understatement to say that I'm impressed with your work on this.

FormerLurker, 2018-02-20:
Yeah, I think it was actually a bug in the rendering.py file.  It seems that in some conditions it was failing and leaving Octolapse in a 'Rendering' state, which prevents it from switching to the 'WaitingForSnapshot' state.  If you find that the BetaDevelopment branch is still broken, maybe you can also attach the octoprint.log file.  It may have some exceptions within it if my code didn't capture and log them properly.  There are a few places still where exceptions can occur that I'm not trapping, but they should always propagate to the main octoprint.log file.

Thanks for the compliments!  I hope to continue to improve the plugin.  Gina (octoprint creator) has just added a new feature for me to the latest development branch that should allow Octolapse to capture the snapshots without pausing the print at all!  That should shave a few MS off of each snapshot and improve print quality somewhat in some cases.  Also, I'll be able to prevent other plugins/terminal code from messing up the snapshot (and thus the print) due to a new gcode tagging system that was also just added at my request (Thanks Gina!!!).  Gcode tagging should also improve logging since now we'll know where each gcode came from (file, octoprint, octolapse, other plugins...) Unfortunately it is a fairly substantial modification that may delay the release of V1.0, but it will be more than worth it, I think.

djeddleman, 2018-02-21:
I installed the development branch and restarted. This time it did not take any snapshots, even during the first print. Feel free to let me know if there is anything else I can do to help debug this.
[plugin_octolapse (5).log](https://github.com/FormerLurker/Octolapse/files/1742710/plugin_octolapse.5.log)

FormerLurker, 2018-02-21:
I'm very sad to hear that, especially since I've completed several prints on this build already :(

Can you do three things for me - post the gcode from the file you were printing, export your settings.json file (you can do that in the Octolapse menu), and post octoprint.log.  As soon as I have this I'll run some tests with my setup as close to yours as possible.

I COULD get the settings from the log, but there are some character conversions that make it difficult, unfortunately.  In any case I should have time tonight to work through this.

Thanks again!

djeddleman, 2018-02-21:
Here is the gcode file, the settings.json, and the octolapse log.

[Octolapse21FEBB2017.zip](https://github.com/FormerLurker/Octolapse/files/1744577/Octolapse21FEBB2017.zip)

FormerLurker, 2018-02-21:
Excellent, thanks for the files and for your patience!

FormerLurker, 2018-02-21:
Hey, can you also post octoprint.log (not plugin_octolapse.log, I already have that)?

djeddleman, 2018-02-21:
No problem. Here is the octoprint.log file.
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/1744603/octoprint.log)

FormerLurker, 2018-02-21:
I've done a compare of your octoprint.log and mine and am sad to say I didn't find anything obvious.  The only difference I can see is that you are using pip 8.1.1 (mine is 9.0.1), a slightly different version of the octopi_support plugin (I'm using the dev branch right now), and you have Repetier_0.91 firmware, while mine is Marlin V1.0.2; Sprinter/grbl mashup for gen6.  I see this line in Octoprint.log:  Detected Repetier firmware, enabling relevant features for issue free communication

I need to see if these 'relevant features' are related to your issue.  I'd also like to know what kind of printer you are using, mostly out of curiosity, but also so I can read about any gcode differences.  I was able to print using your exact settings (with a slightly modified camera URL), so it's not likely to be Octolapse settings related.  The hardest thing to figure out is how it could have worked at all earlier if this was related to firmware/octoprint settings.  This might be a dead end, honestly.

In the meanwhile you could try updating pip and reinstalling (let me know if you need instructions for this), and maybe disabling some of the other plugins you are running to see if we can isolate the problem further?  It also might be good to do a full reboot instead of restarting Octoprint, just in case.  When in doubt, reboot three times, lol!

FormerLurker, 2018-02-21:
I've figured it out, I think.  I obviously didn't get enough sleep last night, lol!  It looks like the home command was sent to the printer before my plugin had a chance to initialize.  You'll notice even at the end of the plugin_octolapse.log the reported positions are not known - X:None,Y:None,Z:None

Waiting for a home command before starting was was added as a safety feature, and to handle printers that don't home to the same location every time (mesh bed leveling and other stuff seems to change the home position on my Prusa), and to reduce setup for printers with varying origins.  Normally Octolapse would pause the print and quickly take a location measurement after getting the G28, but it never saw it.  You can see in the beginning that the G28 was sent to the printer, but Octolapse never saw it in the Queuing phase, which is where the magic happens, unfortunately.

This bug could possibly affect any print, but it's unlikely since there are warm-up commands that take some seconds to complete.  [There's already an issue up for this portion of your problem.](https://github.com/FormerLurker/Octolapse/issues/12)  It's been a particularly irritating issue.  I'll probably update it based on your most recent problem.  I'll up the priority of that issue now that I know it's not just a problem in the virtual printer (I've never seen it when using my Prusa, probably due to the lengthy start Gcode).

I suspect if you try a print without using test mode that things will work.  Alternatively, use the following command to force a brief pause to give my plugin a chance to start up:  G4 P1000

Put that right above the G28 command and test mode should work.  If it doesn't I'll eat my hat (not really).

Let me know if you need any assistance at all, and IGNORE my advice to update pip and reinstall (if it mattered I would have expected pip to fail).  That shouldn't be necessary.  FINGERS CROSSED!

FormerLurker, 2018-02-21:
OK, I pushed an update to the BetaDevelopment branch, and tested your gcode file.  I was quite nervous about running it on my printer, but I looked through the commands and it all seemed fine.  The first few snapshots worked fine, and worked again after a cancel.

The fact that g28 is the very first command in your Gcode file is interesting.  I actually don't think a live print would work either on the last release.  In fact, it might have even been causing the original issue you reported!  Maybe some random lucky thread scheduling made it work sometimes, or maybe the prints that worked didn't have G28 first (or close to the front).  I'll take another look at the original logs and see if I can confirm the latter suspicion.

So, definitely update to the newest version of the BetaDevelopment branch, then try that print again (the gcode you sent me) in test mode.  You can cancel it after a few layers once you're sure it's taking snapshots, then try it for real.  That should take care of several test cases all at once.

Thanks again for your help, this is going to improve Octolapse for everyone!

FormerLurker, 2018-02-21:
Didn't mean to close...

djeddleman, 2018-02-22:
Good news! That seemed to work. 

As for your question about my printer, I'm using a Monoprice Maker Select v2/Wanhao Duplicator i3.

Let me know if there is anything else I can do to help.

FormerLurker, 2018-02-22:
Whew!  THANK YOU SO MUCH for helping me track that down.  You definitely wouldn't have been the only person to encounter that issue had it not been fixed (well, not fixed but much reduced)...

I do have two small requests if you don't mind:  

1.  When you have the printer profile all dialed in for your printer, would you send me a new copy of your settings.json?  I might also want this for your camera depending on what kind of webcam you are using, especially if it's a raspi cam and the settings vary from the default.  I am trying to add common printers/cameras to the default profiles to make it a bit easier to setup. I'm in the process of getting an MK3 profile too, so that would bring the number of tested printers up to 4!

2.  I'd love to get of a timelapse from you that clearly shows the model of your printer.  I want to put together a montage of timelapses showing all of the printers that are known to be currently compatible for when I launch V1.0.

I'm going to create a new issue now for the 'Monoprice Maker Select v2/Wanhao Duplicator i3' profile, and you can post the settings here or there if/when you have a chance.  

Again, thanks so much!  I mark this issue closed!

closed 2018-02-22 by FormerLurker