FormerLurker/Octolapse #28: Plugin can't detect home GCode

Shadowen, 2018-02-22:
Hi,

I wanted to get started with OctoLapse so I downloaded and configured everything. I started a print in test mode and noticed the plugin never seemed to detect the homing (It was still waiting for coordinates).

I am using a Wanhao Duplicator i3 v2 modded with a RAMPS 1.4. I use Slic3r (Prusa Edition) to slice. The home command is `G28`.

Here are the first few lines of GCode:

```
M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F80 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F80
;Put printing message on LCD screen
M117 Printing...

; Filament gcode

M109 S210 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-2.50000 F2400.00000
```

---

Also, I noticed that "View as larger image" and the button to its upper right corner are not working. What are they supposed to do? I know my camera is working as configured (both your plugin and the Octoprint timelapse plugin confirm the camera works.

FormerLurker, 2018-02-22:
Will you try installing the new beta development branch, I just fixed a similar issue with another printer yesterday!  Replace master.zip with BetaDevelopment.zip.

It could also be that the homing command is split.  I wrote my code to handle this, but it's not something that appears in any of my test prints (time to add that).

If I have time I'll take a look later tonight, if not tomorrow night (I have a thing in a few hours).  If you don't mind, can you post plugin_octolapse.log and your GCode file?  Fyi, the log may be empty depending on your debug settings.

FormerLurker, 2018-02-22:
Forgot to mention, the button in the upper right corner plays the most recent N frames.of the timelapse.  You'll see it work after a few frames.  I need to make that more clear.

The pop up is all kinds of messed up in chrome (works fine in ff).  Working on that, as well as the poor dialog css (bootstrap 2, ug)  Both items were a last minute (last day actually) addition to the beta build.

Let me know if the preview animation works for you once we get snapshots working.

Shadowen, 2018-02-23:
I installed from `BetaDevelopment` and it still doesn't seem to detect homing.

`/home/pi/.octoprint/logs/plugin_octolapse.log` doesn't have anything interesting:
```
2018-02-22 22:29:31,161 Octolapse - loaded and active.
2018-02-22 22:29:31,834 Octolapse - is loading template configurations.
2018-02-22 22:36:44,296 Saving Settings.
```
I tried once with verbose logging and it didn't seem to have anthing still. Am I looking at the right log?

GCode file is a cube.
[cube.txt](https://github.com/FormerLurker/Octolapse/files/1750369/cube.txt)

---

Also, I'm wondering how you usually do development? How do you deploy the development build to your Pi? Right now I have to uninstall and reinstall through plugin manager, but that only gets me the tip of a branch. I'm assuming you have a way to just sync source code across to the Pi and run it immediately?

Shadowen, 2018-02-23:
Oops. Found the rest of the log.
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/1750397/plugin_octolapse.log)

FormerLurker, 2018-02-23:
Sad that didn't fix the issue, so I'm guessing there is a bug homing x and
y separately from Z.  I'll test my theory tomorrow, so I may have a patch
by tomorrow evening.

Regarding development, I test first with a virtual printer on my PC, then I
either upload a package to the pi, or I use the plug-in manager like you.
99% of the time if it works in the virtual printer it will work on my mk2,
but not always.  It's not practical to debug on the pi itself, at least I
haven't found a way to do that.

If you're interested in doing some plugin development, let me know what ide
you use.  I've been able to get the debugger working in visual studio, and
pycharm works well also.

FormerLurker, 2018-02-23:
Looking at the log, it looks to be a problem with my regex (I am not so good with regex, but getting better).  I can send you a visual studio solution with some instructions on getting a virtual environment for python to work if you want to tinker.  Let me know, and thanks for reporting the issue!  I'd live to integrate a pull request sometime.  So cool.

FormerLurker, 2018-02-23:
Ok, last message tonight :)  If you're good with regex, the [problem code is here on line 275.](  https://github.com/FormerLurker/Octolapse/blob/master/octoprint_octolapse/command.py).  I've never seen numbers after the X, Y or Z, so that's definitely the problem.  For example, for the y parameter I have:

(?i)^(y)

I probably need something more like this:
(?i)^y(?P<y>-?[0-9]{0,15}.?[0-9]{1,15})$

Somehow those numbers must be made to be optional.  What do those numbers even do?  Is it saying y = 0 after the home command?  I'll have to do some research to figure that out.  I'll also have to modify position.py (you can search for G28) and figure out how to handle the parameters.  I might be able to just ignore them since I send an M114 (get position) command right after homing to auto detect the origin, and I allow an origin to be manually entered in the printer profiles.

I'm going to make this a priority issue.  It will be fixed prob tomorrow.  In the meanwhile, just get rid of the numbers, or just use plain old g28.  Not sure why your sliced code homing the way it is unless it's something specific to your printer.  I worked an issue yesterday, I'm pretty sure it was the same model printer, and it was homing with just g28, so I think it should be OK.

Shadowen, 2018-02-23:
```
The X, Y, and Z parameters act only as flags. Any coordinates given are ignored. For example, G28 Z10 results in the same behavior as G28 Z.
```
http://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29

I'll take a look at the regex tonight. My slicer profile might be doing something funny to the homing code, or it might be Slic3r Prusa Edition.

FormerLurker, 2018-02-23:
According to the info you sent (reprap wiki?), It looks like your slicer is
generating valid GCode.  It's just that any coordinates given are ignored.
I think we can just add on some regex to match positive or negative
decimals to the end and leave the capture group alone.  Then position.py
can stay the same.  Nice.

Shadowen, 2018-02-23:
By the way, I usually use PyCharm, if that's an option. If you can point me to this virtual printer setup you have, I can see if I can resolve some of your issues.

ninjamojo, 2018-02-23:
Hey, I think this regex will work:

(?i)^(x)(?:-?[0-9]{1,15}(?:\.[0-9]{1,15})?)?$

Here it is in action:  https://regex101.com/r/oKsQdE/2

I've not tested 100% yet, will get to that today still hopefully.

Shadowen, 2018-02-23:
I was using something similar.

Still working my way through your parsing logic (You split, then regex each part? I would've written a single monstrous regex with capture groups for each thing o.O)

Got OctoPrint working on my computer though, so that's nice.

FormerLurker, 2018-02-23:
I tried using a monstrous regex for each command, but they quickly became
pure evil once I started adding all the parameters.  At the suggestion of
some nice folks on Reddit, I split them up.  It was a hasty change (as many
things were since I didn't know what I was doing), so it's probably not
well commented.  I'll go back and add in better comments too, later today.

Awesome that you got it working so quickly!  It took me a few days to
figure out.  I really had no idea what I was doing then, lol.  I'm glad to
have someone with more python experience taking a look at things.

Shadowen, 2018-02-23:
Haha, I can really tell you're not a Python developer. Brackets everywhere, weird variable naming schemes... I'm not exactly a Python developer either, but let me see if I can get you this PR.

FormerLurker, 2018-02-23:
I REALLY struggled with the naming conventions.  I started out using one more familiar to me, but it RADICALLY conflicted with the parameter, function, member conventions I was seeing.  I'm used to being able to do 100% safe project wide refactoring, and found this very difficult in Python, else I'd have tried to fix this embarrassing stuff.

How well can pycharm refactor?  I'd love to run something similar to resharper on this project and just make everything consistent.

Shadowen, 2018-02-23:
I don't think **Python** as a language supports project-wide refactoring. You don't know which class you're referring to until runtime...

FormerLurker, 2018-02-23:
Given a choice I prefer strongly typed languages, but I am starting to see some of the power (and potential disorder in the wrong hands) of python.

Shadowen, 2018-02-23:
I prefer Java. C++ feels like they want to become Python, but they're still carrying around the C baggage. Oh well.

FormerLurker, 2018-02-23:
I've done a lot of Java and c++ (in the distant past) coding.  I haven't
used plain old c since high school.  I really don't enjoy c++ as much after
being able to use a garbage collector.  They spoil you completely.

These days I do mostly C#, and Java(mostly with android these days).  Those
two languages are so similar they could be twins.  Unfortunately I am
forced to look at VB from time to time, which saddens me greatly.  Recently
I've been struggling with python (I should actually read a book, or do a
tutorial sometime, lol), as you are by now well aware.

FormerLurker, 2018-02-23:
Hey, it looks like that code fixed the issue!  I'm running it through the virtual printer as we speak, and it's happily snapping away again!  I'll leave this open until you confirm.

So were you able to get everything working properly in pycharm (debugger included)?  If so I would love to get some quick start type instructions to add to the repo for people interested in assisting with the plugin.

This issue went very well!  It's just fun to cooperate after working alone on this for so long.  Thank you so much!

Shadowen, 2018-02-23:
Yea, I followed the instructions on the Octoprint wiki, then I just pasted
the entire Octolapse repo into the Octoprint/src/octoprint/plugins/ folder.
I had to compile steps from a few different places, but it wasn't too bad.
I'll see about writing setup instructions if you want.

FormerLurker, 2018-02-28:
I believe this issue is fixed.  I've not seen an issue detecting home in a while.  Closed!

closed 2018-02-28 by FormerLurker