FormerLurker/Octolapse #29: Silent failure when no webcam selected

Shadowen, 2018-02-23:
When running without **OctoPrint** or **OctoLapse** webcam configured (properly), you may get some of the following errors:

```
2018-02-23 12:08:54,313 Traceback (most recent call last):
  File "/home/wesley/workspace/OctoPrint/src/octoprint/plugins/Octolapse/octoprint_octolapse/__init__.py", line 299, in CopyOctoprintDefaultSettings
    o = urlparse(snapshotUrl)
  File "/usr/lib/python2.7/urlparse.py", line 143, in urlparse
    tuple = urlsplit(url, scheme, allow_fragments)
  File "/usr/lib/python2.7/urlparse.py", line 182, in urlsplit
    i = url.find(':')
AttributeError: 'NoneType' object has no attribute 'find'
```

```
2018-02-23 12:22:57,992 Printer event received:PrinterStateChanged.
2018-02-23 12:22:57,993 Printer event received:PrintStarted.
2018-02-23 12:22:57,993 State Change:PrintStarted.
2018-02-23 12:22:57,994 Traceback (most recent call last):
  File "/home/wesley/workspace/OctoPrint/src/octoprint/plugins/Octolapse/octoprint_octolapse/__init__.py", line 437, in on_event
    self.OnPrintStart(origin)
  File "/home/wesley/workspace/OctoPrint/src/octoprint/plugins/Octolapse/octoprint_octolapse/__init__.py", line 481, in OnPrintStart
    result = self.StartTimelapse()
  File "/home/wesley/workspace/OctoPrint/src/octoprint/plugins/Octolapse/octoprint_octolapse/__init__.py", line 507, in StartTimelapse
    if(not os.path.isfile(ffmpegPath)):
  File "/home/wesley/workspace/OctoPrint/octoprint_venv/lib/python2.7/genericpath.py", line 37, in isfile
    st = os.stat(path)
TypeError: coercing to Unicode: need string or buffer, NoneType found
```

Nothing is displayed on screen at all, so you should probably display some sort of warning...

FormerLurker, 2018-02-23:
Nice catch.  I'll add a check to the initialize section (Start timelapse).

FormerLurker, 2018-03-05:
Rolling this into issue #46 since it will be solved by that enhancement.

closed 2018-03-05 by FormerLurker