FormerLurker/Octolapse #32: Apply style guidelines

FormerLurker, 2018-02-23:
Refactor all code to align with the following style guidelines, particularly the naming conventions.

https://www.python.org/dev/peps/pep-0008/

I probably should have read this BEFORE starting the project, lol!

ppessi, 2018-02-24:
Working on it. What maximum line length do you want? Some variables and strings are very long and hard to fit.

FormerLurker, 2018-02-24:
100, 120 maybe something around there?  My IDE wraps pretty nicely, and I have a big enough display that I didn't notice.  Honestly I'm SO appreciative of any help here that I'm more than willing to let you do what you want.  I just sort of started coding without knowing anything about python, and that will probably be pretty apparent.  I have been spoiled by safe, project wide refactoring...

Could you please pull from the BetaDevelopment branch?  Shadowen and I have been working on that branch, and will continue to until the next beta version is released, which should be a few days at least.

Lastly, I think many smaller commits would be best, too, since so much of the code will be touched.  What do you think?

Thanks again ppessi!

Shadowen, 2018-02-24:
I'm using whatever my IDE autoformats to. I think PyCharm will obey `.editorconfig` though, so just keep that updated with whatever guidelines you choose.

Shadowen, 2018-02-24:
Might want to change the Python setup:
```
[**.py]
indent_style = tab
```

Spaces are preferred.

FormerLurker, 2018-02-25:
I believe I have fixed the build issues.  Sorry if my fixes messed up some of your style changes, I'm having trouble preventing my IDE from making changes.  Now that the build is working again I'll whip my IDE into shape.

I'm going to leave this issue open since I still want to take a look at the naming conventions functions/variables.

Things are starting to look a lot better, thanks again!

ppessi, 2018-02-25:
You're welcome :) I'm glad to help

FormerLurker, 2018-03-05:
Just a few minor issues to take care of in the .py files.  Pretty much only the files in \test need to be refactored.

FormerLurker, 2018-03-05:
There are a few issues still, but to the best of my knowledge our .py files comply with PEP8!

Thanks to everyone who contributed to this issue.  It was a huge task and I'm so glad I had help.  I think this will make it easier for new people to contribute and read the code.  Well, without as gagging as much, lol!

I declare this issue CLOSED!

closed 2018-03-05 by FormerLurker