FormerLurker/Octolapse #33: Make GCode parsing case-insensitive

Shadowen, 2018-02-24:
Some places in the code are still case-sensitive. I know the parsing front-end is mostly proof against this, but some of the internals will crash if provided a lowercase GCode (one of the tests is failing because of this :( ).

Low priority unless a user-affecting issue is discovered.

> The original NIST GCode standard requires gcode interpreters to be case-insensitive, except for characters in comments. However, not all 3D printer firmwares conform to this and some recognise uppercase command letters and parameters only.

FormerLurker, 2018-02-24:
I'll review command.py and see what I can find.  Let me know the nature of the failed test and I'll pay extra attention to that.

Shadowen, 2018-02-24:
test_trigger_gcode.py was sending `g28` rather than `G28`. I'm pretty sure that was causing a problem, but I've got bigger problems to fix before it passes.

Shadowen, 2018-02-24:
Let's not worry about this until it's a real issue.

closed 2018-02-24 by Shadowen