FormerLurker/Octolapse #38: If print is canceled, no need to capture post roll

cabbey, 2018-02-26:
If a user cancels a print, especially before the first layer is complete, then there's no reason to do a post roll. This is particularly painful because if you fix the issue that caused you to cancel the print in a hurry and start a new one, then the second print doesn't get a timelapse created.

FormerLurker, 2018-02-26:
Post roll is only a single repeating frame, unfortunately, though I'd love to get a post roll movie working too!

The problem you are seeing with the missing snapshots might be due to a bug in the timelapse initialization that keeps it from detecting the home position.  To verify, turn the position state info panel on (main settings, feel.free to try the other panels too, they are oddly satisfying).

If you see the bug, install the newest beta development branch and see if that solves it.  If so, let me know so I can relate the issus and hopefully close it out.

If it still fails, maybe you could turn on diagnostic mode (debug settings), and send a log (plugin_octolapse.log, you can download it in the octoprint logs settings page) and your GCode file?

cabbey, 2018-02-26:
I need to poke at this some more... and will grab logs when I figure something out. It looks like it's a little different than I originally thought. It's not capturing post roll... it's just wedged waiting for the print to finish. Except the print has been cancelled. (turns out I may have had a bad USB connection to my printer... I couldn't get any long prints to work right this weekend.)

FormerLurker, 2018-03-05:
@cabbey - Should I close this issue, or are you still having problems?

cabbey, 2018-03-05:
This issue remained in the BetaDevelopment branch when I last encountered a failed print, thursday of last week. If you've made changes since then that you think will resolve it I'll re-fetch the branch and force a failure.

In summary: if you have to cancel a print, or one fails, octolapse doesn't seem to get the message and thinks it is still working on capturing that print, which means it doesn't start a new time-lapse when you start your next print.

Workaround seems to be restarting Octoprint.

FormerLurker, 2018-03-05:
@cabbey, I finally experienced this after many tests.  It seems this is a race condition caused by the event handlers.  I can't seem to recreate this within the virtual printer, so it's going to be difficult to solve.

A feature is coming out in the next version of Octoprint (rlock that can be used to stop gcode processing)  that could solve these problems 100%, but for now I'll try to create a workaround that can hopefully eliminate or reduce this issue significantly.

FormerLurker, 2018-03-09:
@cabbey - I got the no-pause version working!  I think all the race conditions have been eliminated, but am doing some final testing to be sure.  Once that's finished, I can send you instructions on how to upgrade Octoprint so you can upgrade Octolapse to the new NoPause branch.  Let me know.  If so you'd be the third person, including myself, to try it out.  It will NOT work without the latest maintenance branch of Octoprint (not RC maintenance) because there are several fixes and additions in there that were made specifically to get this version to work.

My initial results are very promising.  The quality went way up (even for my horribly configured bowden setup), and using a snapshot position far away from the print (50% of the bed)!  The time required to take a snapshot went down by around 0.25 seconds (not 100% sure how much faster, but I can tell a difference just by watching the printer).  The code even got simpler.

As you can tell, I'm pretty excited.  Anyway, thought you'd want to know since it might fix this error once and for all.

cabbey, 2018-03-10:
Yeah, I'll look at getting my Octoprint moved up to foosel's maintenance branch, then pull BetaDevelopment branch here and see how it goes. I actually have a great test case ready to go for it form my experimentation with the restrictions. Turns out combing a carefully crafted restriction with vase mode printing is... interesting.

https://cabbey.smugmug.com/Fun/3D-Printing/i-2Lk75Sh/A

Sadly, it does some nasty things to the side of the print where it pauses:

https://cabbey.smugmug.com/Fun/3D-Printing/i-8qCkCd8/A

but if we don't have to pause.... 😀

FormerLurker, 2018-03-10:
Yeah, not too pretty.  It should improve with this release, but it won't be perfect.  I have been thinking about adding a wipe feature to the lift and retract.  That should help a bit too.

I still like the timelapse ;)

FormerLurker, 2018-03-10:
Oh, be sure to pull the NoPause branch, not beta development, after updating octoprint.  I can send instructions if you want.  I wrote them up already for someone else.

cabbey, 2018-03-10:
Ah yeah, I’d forgotten seeing that branch show up.

FormerLurker, 2018-03-11:
I noticed that Octolapse was getting locked up if the printer was disconnected during a print.  I had not been handling these events previously.  I think this is now fixed on the latest NoPause branch.  Let me know if you ever get a chance to try it out.

FormerLurker, 2018-03-13:
I found an issue in the NoPause branch (#64) that is related, but was a new problem caused within the new branch.  Turns out canceling/disconnecting is a real pain to handle :)  Anyway I fixed this bug too by resetting some threadding events in timelapse.py.  I'm going to close this issue.  If you have any problems, please report them in issue #64.

Thanks!

closed 2018-03-13 by FormerLurker

cabbey, 2018-03-13:
Yeah, sorry... I downloaded the NoPause branch friday night, kicked off a print and went to bed. Woke up sick saturday morning and haven't gotten back to comment on it since. The NoPause branch worked a charm as for as not pausing went.... shaved several minutes off the print time with zero impact on the print quality. 
[image]

FormerLurker, 2018-03-13:
Sorry to hear that you got sick.  I hope you are recovering.

Well, that is positive!  Thanks for trying it out for me.  Too bad there was no improvement in quality, though.  Hopefully a spiral vase is a worst-case scenario, though I'll attempt to improve it if I can.

I added one cool new feature today that you might like.  You can now switch profiles on the Octolapse tab if you are signed in as an admin.  I haven't tested it without access mode enabled.  Come to think of it, I've never tested Octolapse without access control at all, lol!

cabbey, 2018-03-13:
The quality issues are 100% due to the lift. I have an idea on how to handle that I'll file as a feature request once I have it fully baked in my mind how it should work. (or maybe I'll pull the code up in an editor and try to remember python enough to create a PR. ;)

FormerLurker, 2018-03-13:
Sounds great.  Feel free to post pseudocode if you don't feel up to pull requests just yet.  Feel free though!