FormerLurker/Octolapse #48: unable to create stabilization profile with all disabled

cabbey, 2018-03-03:
Attempting to create a new stabilization profile that has both X and Y stabilization options set to 'Disabled' silently fails.

Workaround: You CAN create it with either set to another value, then edit them later to get the desired "just take the picture wherever you are" setting.

FormerLurker, 2018-03-03:
If I'm understanding correctly, after you click save the dialog close without actually creating a new profile?

FormerLurker, 2018-03-03:
I have a feeling that this was a dialog close issue, where the dialog was closing instead of saving due to a jquery issue with another popup (the timelapse/snapshot preview popup).  I think I have fixed this.  I'm going to close after I check in the fix, but if you notice it again, let me know.

FormerLurker, 2018-03-03:
This issue is related to issue #49

closed 2018-03-03 by FormerLurker