FormerLurker/Octolapse #54: Printing with Octolapse enabled and no ffmpeg path silently fails

adam6806, 2018-03-04:
It would be nice to throw an error to the UI if a print is started with Octolapse enabled but no ffmpeg path defined.

FormerLurker, 2018-03-04:
This issue is related to #29 and #46.  Solving #46 should solve all 3 issues.

Currently I am checking for ffmpeg, but during init (starttimelapse).  It should happen probably right before a print.  Unfortunately this will complicate issue #12 since Octoprint doesn't wait for me to finish with Init before starting the print.  The more time spent in init the higher the chance of missing some gcodes at the start of the print.  The next release of Octoprint will allow me to fix issue #12 by letting me acquire a lock that must be released before the print can continue.

adam6806, 2018-03-04:
Ya that lock would be very beneficial. I didn't realize that it wouldn't wait. Race conditions amirite.

FormerLurker, 2018-03-04:
Yeah, it will save the day in several cases.  I would write it in right now, but I am doing so much stuff outside of timelapse.py, and I'm not sure when Gina will release the next build of Octoprint.

FormerLurker, 2018-03-05:
I'm rolling this into issue #46, and am closing this one.

closed 2018-03-05 by FormerLurker