FormerLurker/Octolapse #73: Commands get sent after SD print cancel

FormerLurker, 2018-03-19:
When printing from SD with Ocotlapse enabled, some commands get sent to the printer before the print is cancelled and a warning message is sent to the user (cannot start octolapse when printing from SD).

This might not be fixable.

sarf2k4, 2018-04-26:
Hi, is it possible to add in a condition for this one if printing from sd, will temporarily disable octolapse and enable it again after sd print has finished automatically.

FormerLurker, 2018-04-27:
So, there is already an option to disable Octolapse on the 'Octolapse' tab, but there currently is no way to automatically re-enable it.  What if there was an option to disable automatic cancellation of the print when trying to print from SD?  Currently it sends a message saying Octolapse can't print from SD and cancels the print.  What if it just stopped the octolapse and let the print continue instead?  Optionally it could send a quick info popup saying that the timelapse is disabled that closes itself after a few seconds.

Let me know what you think of this solution.

sarf2k4, 2018-04-28:
Hi,

Both also works as long as octolapse did not interfere with the sd print. I
think the stock timelapse also disabled by itself when doing sd print.

Is it possible to add something like a service that will monitor the sd
print has start and has finished?

I think this way it might solve your issue of re-enabling the octolapse.

FormerLurker, 2018-08-11:
I think this issue has been solved.  Octolapse will NOT start when printing from SD, and will optionally cancel the print (see main settings) if it is enabled.  It will warn the user in both cases.

Let me know if you think I should reopen.

closed 2018-08-11 by FormerLurker