FormerLurker/Octolapse #83: Brainstorming - Rendering Features

FormerLurker, 2018-03-24:
Comment below if you have any ideas to improve rendering with Octolapse.

systemik, 2018-03-24:
Maybe not related but as a first time user, I spent 30 minutes trying to solve my camera issue because I assumed it was not working. I assume this because of the icon you chose for the main octolapse screen (red strike camera). I was thinking that this would display the snapshot even before print. After 30 minutes of tinkering I tried and it worked as a charm ... Maybe you can change the icon or add a notice that this will display image only when print is active ...

FormerLurker, 2018-03-24:
Great idea, I hadn't thought of that!  I'll create a separate issue and mark it as an enhancement.

FormerLurker, 2018-03-25:
The camera icon issue was resolved in issue #87

Boymeetsmill, 2018-04-04:
I have a couple of ideas for stabilization modes...
1. "Animated - Circle" - Similar to the "Animated - Orbit" stabilization, but instead of many circular paths made throughout the time-lapse, a single circle is made.  
2. "Animated - Square" - Like the circle idea above, but a single square (or perimeter) path

FormerLurker, 2018-04-04:
I made the existing animation's with a spreadsheet.  They are pretty easy
to make!  I did have plans for a fast and slow orbit animation.

It would be difficult to make exactly one orbit.  Somehow I'd need to know
ahead of time how many frames there will be (GCode pre-processing probably,
which will allow lots of cool things).  I will ponder this part.

Thanks for the suggestion!

FormerLurker, 2018-08-11:
Since rendering has changed dramatically over the last several months, I'll close this thread.  We can always create another one!

closed 2018-08-11 by FormerLurker