FormerLurker/Octolapse #84: Cannot set settings

Tjololo12, 2018-03-24:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

I cannot load the plugin, or any of the options, because of this error. It creates a popup as soon as I login stating that there's an error, and all of the tabs say I need to load a profile before I can do anything, but I cannot load a profile either because of a different error (Cannot read property 'guid' of undefined), which I assume is due to the settings not being configured or enabled.
1. Install Octolapse
2. Try to edit settings, or reset default settings
3. Click save

#### What should have happened?
Settings were updated

#### What happened instead?
500 server error with the following stack trace in octoprint.log:

2018-03-24 15:52:45,351 - octoprint - ERROR - Exception on /plugin/octolapse/restoreDefaults [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1132, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 244, in restore_defaults_request
    if self.Settings is not None:
AttributeError: 'bool' object has no attribute 'current_debug_profile'
2018-03-24 15:52:45,355 - tornado.access - ERROR - 500 POST /plugin/octolapse/restoreDefaults (127.0.0.1) 103.58ms

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 4.9.35-v7+

#### Printer model & used firmware incl. version

Printer Model: Do not get to this point
Printer Firmware Version: n/a

#### Browser and version of browser, operating system running browser

Browser: Chrome 65.0.3325.181
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

#### Link to settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/u5t0mTxR

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Tjololo12/ba2d18dbd43b30b6ce34135ed6f13e77

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://gist.github.com/Tjololo12/bc587c22710d8ea78754f35622c03305

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://i.imgur.com/ULEjluM.png

FormerLurker, 2018-03-24:
I was responding on gist, sorry about that.  Going to add my comments here:

> Can you try re-installing? I have made some updates to the default settings.json file that might help. I'm checking a few more things in the meantime.

FormerLurker, 2018-03-24:
This has been fixed, I believe.  There was a problem (typo) when creating a new settings file from the defaults.  I'll leave the issue open until I hear back from you

systemik, 2018-03-24:
I had the same issue and just re-installed. I do not get the error anymore on start and I'm able to go to the setting page. Thanks for the quick turnaround and for the plugin creation (looks perfect on the description).

Tjololo12, 2018-03-24:
Looks like it's good now, thanks! I have a maker select v2, if I get a
working profile I'll shoot it over for you. Thanks for the rapid response!

FormerLurker, 2018-03-24:
You are too nice @systemik!  I"m going to leave the issue open at least until I hear back from @Tjololo12, but maybe a bit longer since this essentially broke the plugin.  Unfortunately I didn't detect the typo causing the bug because my settings file was not being deleted on uninstall.  Glad I found it before OctoPrint 1.3.7 stable dropped.

FormerLurker, 2018-03-24:
@Tjololo12 - Check this issue and let me know if your settings match:  https://github.com/FormerLurker/Octolapse/issues/27

mikevindler, 2018-03-26:
I am having this same issue when I try to reset any settings. "Unable to restore the default settings.  Status: error.  Error: INTERNAL SERVER ERROR"

FormerLurker, 2018-03-26:
@mikevindler , can you try a reinstall?  This looks to be a first time installation issue.  If you still have problems can you post your logs (plugin_octolapse.log and octoprint.log)?  Thanks!

FormerLurker, 2018-03-27:
I just read your edit, glad it's working!  Closed!

closed 2018-03-27 by FormerLurker