FormerLurker/Octolapse #85: Setup Problem

interceptor70, 2018-03-24:
Error message : Unable to restore the default settings.  Status: error.  Error: INTERNAL SERVER ERROR

FormerLurker, 2018-03-24:
@interceptor70 , I think this was fixed just a few moments ago.  If you installed Octolapse more than about 2 hours ago, please re-install and try again.  Very sorry for the inconvenience :(

interceptor70, 2018-03-24:
Ok Thanks :) now it works.... :-) and now i will add an Irapid Black

FormerLurker, 2018-03-24:
Superb.  And another printer profile, you're the best!  That a sweet looking printer, btw...

FormerLurker, 2018-03-24:
I'm marking this as closed, please create a new issue to create the irapid black profile, upload your settings, and I'll add it to the defaults with a 'beta' tag.

closed 2018-03-24 by FormerLurker