FormerLurker/Octolapse #86: [Request] Printer Profile - LulzBot TAZ 6

b-morgan, 2018-03-24:
#### If this is a feature request describe it here
What do you need from me to create a Printer Profile for the LulzBot TAZ 6?

I looked at https://github.com/FormerLurker/Octolapse/wiki/Printer-Profiles but there are values in there that I'm not sure where to obtain them. I use both CuraLE and S3D to slice (plus other slicers at times). Do I need a profile for each Printer/Slicer combination? I also have multiple extruders (both single and dual). Are the profiles going to change depending on which extruder is being used? There are some filaments for which I use more or less retraction. Do I have to have a profile for each of these?

#### Version of Octolapse
Octolapse Version: v0.2.2

#### Version of OctoPrint
OctoPrint Version: 1.3.7rc2

FormerLurker, 2018-03-24:
You may want to use multiple profiles depending on your slicer settings. If you shoot me a copy of your slicer settings, preferably those provided by the MFG (I'd like the out-of-the box settings to work for new users if possible), I'll post some setting values you can test.  A small sliced gcode file that has retract/lift commands would also be helpful.

Unfortunately I've not thought far enough ahead to support different retraction values for each extruder.  I don't even detect tool changes at the moment, unfortunately.  Retraction detection isn't a requirement to get a timelapse, but it helps quality in most situations.  So if one extruder is retracting to .5mm and the other at .8mm, you MIGHT be able to get away with setting retraction to either .5 or .8, but you have to understand the effects.  If your extruder is retracted 2mm, but your printer profile is set to retract to 0.5MM, it will not retract further when taking snapshots.  The other side of the equation is the [extrusion state requirements](http://github.com/FormerLurker/Octolapse/wiki/Snapshot-Profiles#extruder-state-requirements).  Here you can define the required extruder conditions for taking snapshots.  Pay close attention to the PARTIALLY and FULLY retracted settings, which will be determined by the retraction settings in the printer profile.

I hope that helps!  Unfortunately it's hard to support all of the myriad settings in all of the different slicers, but I will sure try!

b-morgan, 2018-03-24:
Let's start with these files. A .gcode of two simple squares sliced for the LulzBot TAZ 6 with a Dual Extruder V3 and the Marlin Conditionals_LulzBot.h which defines all the settings for all of their printers.
[Temp.zip](https://github.com/FormerLurker/Octolapse/files/1844837/Temp.zip)

FormerLurker, 2018-03-25:
This printer has some complicated starting gcode, but I'm getting through it.  Your cura profile would be helpful since I can't find any good examples of zlift or retraction outside of the start/end gcode.

The min/max x, y, and z positions are pretty hard to extract from that .h file since there are so many defines spread about and lots of superfluous info.  Can just tell me what your octoprint profile settings are for print volume?

Also, I noticed the following in your gcode:
G1 X -18 Y95 F1000                   ; slow wipe
G1 X -18 Y90 F1000                   ; slow wipe
G1 X -18 Y85 F1000                   ; slow wipe
G1 X -17 Y90 F1000                   ; slow wipe

Notice the spaces after the X.  Is this standard start gcode from your cura profile?  My printer doesn't seem to like these commands, and it kind of breaks my gcode parsing.  Octoprint's virtual printer accepts the command but doesn't change positions when I send it.  I removed the space between X and - for my testing, at least until I know more about why there are spaces there.

Now for the good news.  I was able to get snapshot generation within the virtual printer.  It captured 14 images using the default layer trigger.  However, I'm not confident that I understand the behavior of your printer enough to say that Octolapse will work.  Maybe you can answer the following questions:

If you move to a known position and change tools, does one axis move to compensate for the nozzle offset, or does the gcode compensate for this somehow?

T0
G1 X100 Y100
T1; is there movement here?

If you move to a known position and change tools, does this change the result of an M114 command?

T0
G1 X100 Y100
M114; should report position x100 y100
T1
M114; What position doe it report now?

I wish I had one of these bad boys here, I'd get it running this weekend.  Thanks for helping with a fruitful, but somewhat slower process!

b-morgan, 2018-03-25:
Attached is a .zip of the OctoPrint printer profiles for the TAZ 6 with the dual extruder V3. The one labeled with (simplify3d) does not have a nozzle offset because the start gcode contains an "M218 T1 X13 Y0" at the end of the start gcode. I have not changed the speeds from the OctoPrint defaults but here they are from the .h file (I wish there was a way to run the C pre-processor on a .h file and have it output only the lines that will be compiled):

    #define LULZBOT_DEFAULT_MAX_FEEDRATE          {300, 300, 3, 25}      // (mm/sec)
    #define LULZBOT_DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}
    #define LULZBOT_Z_STEPS                       1600
    #define LULZBOT_Z_MICROSTEPS                  16

The printer volume is 280 x 280 x 250 with travel outside those bounds for the nozzle wipe.

The space between the "X -" is there from the manufacturer but I can easily remove it as there are a couple of bugs in the start gcode. I believe the firmware is pretty much stock Marlin so the space is probably allowed. I'll do some research on that.

I'll change to the dual extruder and perform the tool change experiments tomorrow. I expect that the behavior will depend on whether or not I have a "M218 T1 X13 Y0" active or not. CuraLE handles the offset by adjusting all of the gcode values as it outputs the gcode. Simplify3D relies on the firmware to handle the offsets.

[lulzbot_taz_6_dual_v3.zip](https://github.com/FormerLurker/Octolapse/files/1844945/lulzbot_taz_6_dual_v3.zip)

FormerLurker, 2018-03-25:
Thanks for the files, and the info.  Ill research the m218 command a bit
and review the new info.  I will probably work on this tomorrow unless
something comes up.

b-morgan, 2018-03-25:
> This printer has some complicated starting gcode, but I'm getting through it. Your cura profile would be helpful since I can't find any good examples of zlift or retraction outside of the start/end gcode.

I'm not sure how to give you my Cura profile. The values presented to me in CuraLE are assembled from a rat's nest of json, xml, and cfg files (which are all text files). If I try and save a profile, its binary.

If you have a model that should generate the gcode you are looking for, I'd be happy to run it through Cura LulzBot Edition (CuraLE). You could download CuraLE from https://www.lulzbot.com/cura, tell it you have a TAZ 6 and a Dual Extruder V3 and then look at the list of parameters. I believe the retraction stuff is defined per filament type. You can also load a model file, slice, and save to file to look at gcode.

FormerLurker, 2018-03-25:
Thanks, I'll download the slicer.

b-morgan, 2018-03-25:
BTW, both Ultimaker Cura 3.2 and Cura LulzBot Edition 2.6 (or alpha 3.2) are open source and have definitions for more than just Ultimaker or LulzBot printers (Ultimaker doesn't supply LulzBot definitions, I think there's some rivalry between the two companies).

b-morgan, 2018-03-25:
@FormerLurker wrote:
> Notice the spaces after the X. Is this standard start gcode from your cura profile? My printer doesn't seem to like these commands, and it kind of breaks my gcode parsing.

I think you have to fix your gcode parsing :-)

From https://www.cnccookbook.com/g-code-basics-program-format-structure-blocks/:

> Spaces in G-Code Programs
> Now is a good time to mention that g-code ignores spaces in many cases. They’re there simply to make the code more readable by us humans, so use them to your benefit. You can stick spaces between a word and the number that goes with the word (that number is called the “address”, see Word Address Format below):
> 
> G00 can be G 00
> 
> X0 can be X 0
> 
> Decide what you like to see in terms of readability and stick with it. Depending on your g-code editor, it may have tools to help you manage this sort of formating automatically. G-Wizard CNC Simulator and Editor certainly does.

From http://linuxcnc.org/docs/html/gcode/overview.html:

> Spaces and tabs are allowed anywhere on a line of code and do not change the meaning of the line, except inside comments. This makes some strange-looking input legal. The line G0X +0. 12 34Y 7 is equivalent to G0 x+0.1234 Y7, for example.

FormerLurker, 2018-03-25:
Thanks for the heads up.  Shouldn't be too difficult to fix.  I'm kind of
learning as I go here.

b-morgan, 2018-03-26:
I changed to my TAZ 6 Dual Extruder V3 and performed your experiments. The detailed output is attached but to summarize:

> If you move to a known position and change tools, does one axis move to compensate for the nozzle offset, or does the gcode compensate for this somehow?
> 
> T0
> G1 X100 Y100
> T1; is there movement here?

There is no movement with or without an "M218 T1 X13 Y0" command.

> 
> If you move to a known position and change tools, does this change the result of an M114 command?
> 
> T0
> G1 X100 Y100
> M114; should report position x100 y100

It does.

> T1
> M114; What position doe it report now?

With "M218 T1 X0 Y0": x100 y100
With "M218 T1 X0 Y0": x113 y100

 
> I wish I had one of these bad boys here, I'd get it running this weekend. Thanks for helping with a fruitful, but somewhat slower process!

[serial.log](https://github.com/FormerLurker/Octolapse/files/1849706/serial.log)

b-morgan, 2018-03-26:
I have a question for you... Is the reason you need Retraction amounts, Z-lift, etc. so you can figure out if a change in Z is a layer shift? Or is it so when you move the extruder out of the way, you can retract some filament to prevent oozing?

Don't most slicers put in comments between the actual layers? I know Cura does. I know Simplify3D does. Would specifying the string to look for help in any way?

FormerLurker, 2018-03-26:
That is pretty good, actually.  It looks like the only settings that need to be added are the offsets (I'm guessing X and Y. What about Z?) for each tool and maybe a separate retraction distance.

I'm not going to lie, this will take some time for me to implement.  I'm going to try and find a dual extruder printer to test with.  However, it makes sense to support these, and I definitely will.

Again, thank you so much for you help!

FormerLurker, 2018-03-26:
> Is the reason you need Retraction amounts, Z-lift, etc. so you can figure out if a change in Z is a layer shift? Or is it so when you move the extruder out of the way, you can retract some filament to prevent oozing?

There are several reasons:

1.  Retraction mm is used to retract when moving to take a snapshot.
2.  Retraction mm is used to determine if the extruder is fully/partially retracted.  This is used in the snapshot settings under each trigger type.
3.  Z-hop height is used to determine how much to lift when moving to take a snapshot
4.  There is a flag in the snapshot settings to prevent snapshots EXCEPT when lifting.

All of these options are configurable.  You can disable retraction, or choose to ignore the retraction state (partially/fully retracted) when deciding when to take snapshots on a layer.  You can do the same with z-hop.

Does that help?

b-morgan, 2018-03-26:
The offsets are just X and Y. For most printers with dual extruders the two nozzles need to be level (i.e. at exactly the same Z) or the lower one could drag over the layer that the higher one just put down. 

There are some printers that have a servo to shift the inactive nozzle higher to avoid this issue. Some printers have two independent extruders (Ultimaker 3?) so I suppose there could be different Z offsets.

With different materials in each nozzle (i.e. PLA and PVA) the settings (including retraction) could be separate for each nozzle.

One annoying thing I've found with dual extruders

FormerLurker, 2018-03-26:
OH. Also, I learned that many implementations of Marlin do NOT obey the spacing rules for gcode.  I started on a new parser, and have had some difficulties because some parameters REQUIRE that spaces be preserved (file names, messages to LCD, etc).  Simply trimming all whitespace from a line will not cut it.  So far I have implemented a routine to handle all kinds of comments (';' and '()'), line numbers, and does a greedy search for the command and address.  Now I'm left with the parameters.  I've been stuck dealing with several critical issues (initial install problems, lag due to python logger, freeze in the middle of a snapshot), but am testing fixes and will hopefully be clear to continue with the parser next.

b-morgan, 2018-03-26:
I'm guessing that if you just accept / handle spaces in the parameter part (number part) of the line, you can leave the spacing as it arrives.

I believe LinuxCNC is open source so I wonder if you could "steal" their parser or at least look at it to see how they handle it.

FormerLurker, 2018-03-26:
I'll take a look, thanks!  I've checked out a couple of other open source
solutions, and none that I've found actually work.

FormerLurker, 2018-04-04:
So this feature request turned out to be quite difficult.  I had to completely re-write the parser so that those gcodes in your file would parse properly.  It's been added to the unit tests.

Here are some examples of gcode that will now parse (surrounded by 's):
```
g  01 x  10 0  y2 00 .0z  3.0 001 e1. 1 f72 00 .000 

g0x-100y + 200.0z+  3.0001e -1.1f  +  7200.000; Here is a comment

g1x100y200.0z3.0001e1.1f7200.000

g1 x100 y200.0 z3.0001 e1.1 f7200.000; Here is a comment

(comment in the front)g(comment in middle)2()8x(another comment in middle)x(comment between address and value)1.00()1 . 1(another); Here is a comment
```
The good news is the new parser is actually faster than the old one since it doesn't use any regex.  I did a performance test on my desktop (mileage may vary) and was able to parse one gcode in 0.006 to 0.0559 milliseconds depending on the number and length of the comments and parameters.  Not too shabby :)

Also, some kind soul created a Taz2 profile and got it working without the new parser.  This is all in the devel branch.  I've added lots of fixes and improvements.  It does NOT work with multiple extruders at the moment, but if you do a single extruder print, it should be OK.  I need to REALLY roll up my sleeves to multiple extruders to the mix, but that will hopefully be here in the next couple of months.

If you are feeling brave and want to test before I release v0.3.0, here is the installation URL for the OctoPrint plugin manager:

https://github.com/FormerLurker/Octolapse/archive/devel.zip

I recommend updating OctoPrint to rc4 before you start, and if you have installed OctoLapse before [RESTORE THE DEFAULT SETTINGS](https://github.com/FormerLurker/Octolapse/wiki/Settings#restoring-default-settings), else you won't see the Taz 2 profile.  Hopefully I'll have an upgrade routine for the settings soon so that people won't have to do this each version change.....  Add it to the list, lol!

FormerLurker, 2018-04-11:
So this profile (single extruder slices only though) is now included in v0.3.1.  You'll still need to restore the default settings if you already have Octolapse installed.  Let me know if you have a chance to try it out, but be sure to review the settings first, since it's a brand new profile.

b-morgan, 2018-04-13:
Octolapse v0.3.1 seems to be working fine on my TAZ 6 after I adjusted the print bed size to -20 in X, -25 in Y, and 300 in X and Y. I'm testing with the Dual Extruder V3 because that's what I'm currently printing (experimenting) with.

The start code for the Dual Extruder V3 hangs the nozzles off the front edge of the build plate to prime the nozzles and hangs the nozzles off the back of the build plate when using Change Filament (an M600). I believe the Single Extruder V2.1 will do the same thing off the back for Change Filament, the prime off the front is part of the start gcode.

I also believe that the Aerostruder will be identical to the Single Extruder so you can claim support for that one as well. The Dual Extruder V2 has the second nozzle offset by -50mm in Y from the first. The Dual Extruder V3 has the second nozzle offset by 13mm in X from the first.

b-morgan, 2018-04-13:
> I'm not going to lie, this will take some time for me to implement. I'm going to try and find a dual extruder printer to test with. However, it makes sense to support these, and I definitely will.

If there's anything I can do to help, just let me know. I'm retired and the printer is a hobby so it often sits idle. I can set up a VPN if you want more direct access.

CaylaDenning, 2018-06-18:
Hello! I have been trying to get my duel V3 print head to work and I have been having a lot of problems! I am using the cura software that is suggested by lulzbot and I am using clear and lime green filament. When  I try to print anything with two colors the clear filament stops printing as soon as it reaches the layer before the green is supposed to start. I was wondering if you could try to print a simple cube that is one color on the bottom and another on the top switching halfway through to see if your head works better. I have tried to change just about every setting to try and fix it but nothing has worked. I am using PLA and the suggested settings for that.

b-morgan, 2018-06-18:
My Dual Extruder V3 on my TAZ 6 is working fine. I've probably printed over 20 dual color objects including the gear thing using the filament (PLA + Polysolve) that came with the tool head using the default settings. However, the fact that mine works isn't really going to help you.

My suggestion is to call or email [LulzBot Support](https://www.lulzbot.com/contact-us) and they should be able to identify the problems you are having and get you on the right track.

FormerLurker, 2019-10-12:
I'm going to go ahead and close this issue out since there is now a default lulzbot TAZ profile (starting in V0.4, see the profile repo [here](https://github.com/FormerLurker/Octolapse-Profiles/blob/master/0.4.0rc1.dev2/printer/lulzbot/taz_6/profile.json)).  There is a separate issue for dual extruders, which has not been solved.  At some point I plan to tackle that too.

Thanks for participating everyone!  I really appreciate it.

closed 2019-10-12 by FormerLurker

FormerLurker, 2019-11-01:
@b-morgan, 

I thought you might be interested to know that I have finally added preliminary support for multiple extruders!  See Issue #95.  I will need testers soon if you are interested.

I've only tested it via the debugger so far, and don't own an actual multi-extruder printer.  I can only test the shared nozzle mode, which doesn't include any extruder offsets, which is the bit that I'm most concerend about.  My hope is that I've taken all of the relevant things into consideration, and that in the end the extruder will stabilize in the appropriate spot no matter what tool was being used at the time.

I'll let you know when I've run a few MMU prints and single material mode prints.  It will probably be a week or so before I can get all of that done, FYI.  I'll be looking to make sure that per-material settings are honored when performing the stabilization, that per-extruder tracking is working correctly, and that slicer settings (both manually entered and automatically extracted) are correctly applied when running live prints (already verified in the debugger).  Once all of that is complete the only thing left will be verifying that extruder offsets (manually enetered and extracted via G10/M218 codes) work as expected.

Thanks!