FormerLurker/Octolapse #88: Maker Select v2 Profile Unrecognized Command

Frankenberrypi, 2018-03-24:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.2.2

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.7rc2 running on OctoPi 0.13.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Upload gcode for 0.5mm_thick_wall.stl from https://www.thingiverse.com/thing:5573
2. Click print

#### What should have happened?
The test file should print and Octolapse should do something. 

#### What happened instead?
1. Click go in Octoprint
2. Printer goes to home in XYZ
3. Does the normal priming up to about 15mm after the Z home step
4. Translates in XY to the start position (more quickly than normal), remains at about 15mm in Z
5. Translates back to 0,0 in XY at the rapid speed, remains about 15mm in Z
6. Re-homes in in XY, remains at about 15mm in Z
7. Print stops, Octolapse popup states that a video can't be made from 1 frame

#### Operating System running OctoPrint and Octolapse

OS Name: Linux octopi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: maker select 3d printer v2
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: 64.0.3282.186
Browser OS: Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/LTuFZE5q

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/EXtpXfPc

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/wHLzULx5

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

Frankenberrypi, 2018-03-24:
Forgot the setup.json file.  Here it is:
https://pastebin.com/j7K5ycaL

FormerLurker, 2018-03-24:
OK, excellent!  Thanks for reporting the issue, you gave me everything I needed to solve it, but I'd like one more thing if you could:

Your gcode file does not have any code other than the start up/end gcode.  Probably because you sliced without tje 'detect thin wall' setting if you are using Slic3r (could be true for others).  However, I used your code and was able to get a print running, so I think I've fixed it.  Re-install to test, but please send me a complete gcode file too so I can add it to my test library.

Also, I noticed that your printer might have the coordinate 0,0 in the center of the bed, no?   If your print volume is not correctly configured you will probably get lots of 'out of bounds' errors.  Let's say your bed is 100mmx100mm.  If 0,0 is in the center of your bed then your print volume should be x1=-50, x2=50 y1=-50,y2=50.  I'm not sure about this, but if you want to find out you could home all axis then issue G0 X0 Y0 and see where the extruder/bed go.

Frankenberrypi, 2018-03-24:
OK, reinstalled and it's working now.  You are right, something is wrong with that gcode file I sent you.  I'll have to debug my slicer settings on that one.  In the meantime, here is a gcode file that I more typically use for dimension stability testing.  
https://pastebin.com/LPJv8D31

Frankenberrypi, 2018-03-25:
Will do. I've got an overnight print going right now. I expect the timelapse to work just fine based on my testing.  Assuming that turns out to be true, I'll post my settings tomorrow morning.  Thanks a bunch for the help.

FormerLurker, 2018-03-25:
Awesome, I'll wait for your confirmation before I close the issue.  Please do keep an eye on things just in case, and good luck on your print!

skorokithakis, 2018-03-25:
I'm getting the same error, although my file isn't wrong (it printed fine). Here are the logs and the gcode file: https://www.eternum.io/ipfs/QmSoobYZF6UVYHA1NCgB8auQZAHNQdHAZe2QR4kqHiZq5n

skorokithakis, 2018-03-25:
And here are my printer settings:

```json
{
  "movement_speed": 9000,
  "guid": "e8c13c35-c22b-4825-bcd1-1c1808912787",
  "min_x": 0,
  "min_y": 0,
  "min_z": 0,
  "detract_speed": 2400,
  "z_hop": 0.5,
  "max_z": 0,
  "max_x": 0,
  "max_y": 0,
  "description": "",
  "e_axis_default_mode": "require-explicit",
  "g90_influences_extruder": "use-octoprint-settings",
  "origin_z": null,
  "origin_y": null,
  "origin_x": null,
  "priming_height": 0.5,
  "abort_out_of_bounds": true,
  "retract_speed": 2400,
  "auto_position_detection_commands": "",
  "name": "wanhao duplicator i3",
  "retract_length": 0.6,
  "snapshot_command": "snap",
  "override_octoprint_print_volume": false,
  "z_hop_speed": 9000,
  "printer_position_confirmation_tolerance": 0.001,
  "units_default": "millimeters",
  "xyz_axes_default_mode": "require-explicit",
  "auto_detect_position": true
}
```

FormerLurker, 2018-03-25:
Your slicer is assuming that your axes are already in absolute mode and is not adding a G90 to the start gcode.  There are 2 ways to handle this:

1.  Recommended - Add g90 to your start gcode.  This is the safest way to go.
2.  In Printer Profile settings change XYZ Axis Mode from 'Require Explicit...' to 'Default to absolute'

Let me know if your error goes away.

Also, if you want, delete plugin_octolapse.log and restart OctoPrint.  That way you can clear out the unnecessary logs.  I'm going to try to update the logger so that it will detect and recreate a missing log on the fly so this won't require a restart.

skorokithakis, 2018-03-25:
G90 added! Will let you know how it goes, but you may want to add that in the "prerequisites" section.

skorokithakis, 2018-03-25:
Yep, works fine now. However, it is doing two captures before starting to print the skirt, one after the other (after priming and moving to the center, but before extruding anything else).

FormerLurker, 2018-03-25:
Interesting.  Send the GCode please and I'll try to replicate.

skorokithakis, 2018-03-25:
Here you go: https://www.eternum.io/ipfs/QmZBtm3FyGCiLGS7og9WvWxozbEE5S9yNBWvyNjDqZdCNb

skorokithakis, 2018-03-25:
FYi, it stopped again (moved the bed to take a shot and then didn't resume). I wasn't logging, so I can't open another issue yet, but I will.

skorokithakis, 2018-03-25:
Here's the log from the double-snapshot-before-print bug: https://www.pastery.net/hfwvtf/

skorokithakis, 2018-03-25:
Getting the log for the no-resume bug now. *Of course* it would only fail when my model gets perfect adhesion and not when it got unstuck...

skorokithakis, 2018-03-25:
This is where it didn't resume after a snapshot: https://www.pastery.net/dvgwrh/

skorokithakis, 2018-03-25:
It's odd because it doesn't happen sometimes, even though I use the exact same file/settings/everything.

FormerLurker, 2018-03-25:
I don't think changing settings will help.  Seems to be a serial issue of
some kind, or a race condition maybe.

skorokithakis, 2018-03-25:
I'm leaning more towards race condition, as I've never had a serial problem in the years I've been printing.

FormerLurker, 2018-03-25:
Well, we are using a brand new feature to stop commands from being sent
from the GCode file to the printer while snapshot codes are being sent.
Could be an issue with that.

I'm trying to find a way to replicate the issue consistently now.

skorokithakis, 2018-03-25:
Oh, this reminds me: I have it set to take the bed 75% of the way forward, which causes it to take more time to complete the snapshot. Maybe there's a race condition related to that.

FormerLurker, 2018-03-27:
I believe this issue has been resolved.  Will reopen if not.

closed 2018-03-27 by FormerLurker