FormerLurker/Octolapse #89: No stream from Logitech C615

ieatacid, 2018-03-24:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Trying to get my camera working (Logitech HD WEBCAM C615)

#### What should have happened?
I should see the stream from my camera

#### What happened instead?
The OctoLapse tab just shows 'no_snapshot.png'.
I tried using "Default Webcam" & "Logitech Webcam" profiles, but I still get the 'no_snapshot.png' icon  instead of the view from the camera (Logitech HD WEBCAM C615). In the OctoPrint 'Control' tab I can see the camera stream but I can't get it to show in the OctoLapse tab. I've tried both 'Default' & 'Logitech' profiles with the same result. I've tried restarting OctoPrint, rebooting my RasPi.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/PuYHC4UA

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/JKvjzxNJ

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/GD0KMV2c

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://i.imgur.com/uRSA1oP.png

FormerLurker, 2018-03-24:
Hey, THANK YOU for submitting an issue!  Please check out Issue #87 [here.](https://github.com/FormerLurker/Octolapse/issues/87).  This isn't an error, but a tragic oversight on my part.  That big error looking image is just the placeholder meaning 'no snapshots have been taken yet'.  I need to replace it with a 'welcome to octolapse' image or something.

Try running a print and see if this resolves itself.

Oh, and sorry my plugin doesn't play nicely with your skin :(.  I haven't tried any of those plugins yet, but maybe I will at some point if there is enough demand.

ieatacid, 2018-03-24:
Ok, thanks. I'll try a print.

ieatacid, 2018-03-24:
It works. Sorry about that.

closed 2018-03-24 by ieatacid

FormerLurker, 2018-03-24:
No problem, I'm very glad it's working for you.  I'd love to hear how your timelapse turns out!