FormerLurker/Octolapse #91: Add mm/s in addition to mm/min in printer configuration.

systemik, 2018-03-25:
#### If this is a feature request describe it here

Many person use mm/s to configure their printer / slicer. Can you add a new setting in the printer configuration page to chose between mm/min and mm/s? This will ease the creation/modification of printer templates for those using mm/s.

#### Version of Octolapse
 
Octolapse Version: V0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-03-25:
I'll definitely consider this!  There are a lot of units on the various settings pages, so it's not a small task.  Maybe there's an easy way to accomplish this that I'm not thinking of at the moment.

I'll think about this for a bit and will mark this as an enhancement now.

systemik, 2018-03-25:
Quick win for you, waiting for full implementation, is to add a mm/s calculation info. No need to have mm/s entry at first but at least when you input mm/min, you'll know immediately what is the mm/s equivalent. This would need to ba added to the printer config page only (e.g. automatic divide by 60 info).

FormerLurker, 2018-03-25:
Good idea!  Will probably do this.

FormerLurker, 2018-04-01:
Well, I ended up adding a unit selector on the printer profile page.  This will be available when v0.2.6-beta is released.  It should be pretty soon unless I run into snags testing.

FormerLurker, 2018-04-11:
Hey, this is now in version 0.3.1.  Let me know if you like it, and thanks for pushing me to do it.  The axis feedrate section is much nicer now :)

Closing the issue.

closed 2018-04-11 by FormerLurker