FormerLurker/Octolapse #92: Add horizontal and vertical flip option for camera.

systemik, 2018-03-25:
#### If this is a feature request describe it here

Horizontal and vertical flip option for camera exists in the regular camera option but not in octolapse. I do use pi camera mounted upside down and a flip setting would be interesting (to avoid post processing of the video to put it at 180°).

#### Version of Octolapse
 
Octolapse Version: v0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc2

Funkenjaeger, 2018-03-25:
I second this.  Many of the (popular) raspberry pi camera mounts for the Prusa on Thingiverse mount the camera with the ribbon cable coming out the top, which makes the image upside down by default.

FormerLurker, 2018-03-25:
Thanks for the suggestion!

I am using the Pillow module, which I think supports rotation.  I think this belongs in the camera profiles.  Would it make sense to add the following option:

Rotation Angle (0-365, where 0 and 365 are equivalent) 

or would it be better to keep it consistent with the rendering settings and have the following

Flip Horizontal (true/false)
Flip Vertical (true/false)
Rotate 90 Degrees (true/false)

I think the rotation angle would be more powerful (and easier to implement), but is unfortunately different than the rendering settings.

I have a few bugs to work on right now, so there is time to ponder this for a bit.

Funkenjaeger, 2018-03-25:
Either way, I think it makes sense to keep it in increments of 90 degrees, so image quality doesn't change due to interpolation.  It looks like Pillow's transpose method can do everything we'd need:
https://pillow.readthedocs.io/en/3.1.x/reference/Image.html#PIL.Image.Image.transpose

FormerLurker, 2018-03-25:
You're probably right.  If someone needs 15 degrees of correction it's probably best to adjust the camera itself.

Since this is a pretty easy thing to implement I'll probably have it in the next release, which should be out very soon.

FormerLurker, 2018-03-31:
This has been implemented and is in the master branch.  Enjoy!

closed 2018-03-31 by FormerLurker

Funkenjaeger, 2018-03-31:
Works great.  Thank you!

varazir, 2018-04-01:
I have set my to flip Top and Bottom, tested also 180 that would be the same but the test/thumbnail on the Octolaps tab do not show the change.

Funkenjaeger, 2018-04-01:
Are you sure it has taken a new thumbnail snapshot since you changed the setting?  Immediately after changing (to rotate 180) the thumbnail was still upside down for me, but was clearly a stale image - as soon as I started a print it updated and has been fine since.

varazir, 2018-04-01:
You'er right it worked fine when I started to print.

karthik-KSK, 2018-06-11:
where do i get the code

FormerLurker, 2018-06-11:
@karthik-KSK, I'm not sure what you are asking for exactly, but if you're looking to rotate/flip your snapshots, just install the latest version of the plugin via the OctoPrint plugin manager and check the camera profile settings for the image transformation drop down.  Let me know if you have any problems.

icancwhatusay, 2018-09-06:
>  just install the latest version of the plugin via the OctoPrint plugin manager

Perhaps I'm missing the obvious, but what is the name of the plugin?

FormerLurker, 2018-09-07:
@icancwhatusay, what are you trying to accomplish exactly?  I think you might have intended to comment on the OctoPrint repository.  This is the OctoLapse repository (a timelapse plugin for OctoPrint).  If I'm misunderstanding you, I apologize, but if you have a problem with Octolapse I want to help!

icancwhatusay, 2018-09-10:
Hi,

Thanks for the reply but I actually figured out my issue. I couldn't find
the "flip the image" buttons. I didn't realize they would be in the
settings page instead of on the Controls page. Thanks for your help though!

Regards,
*Brian Janiszewski*

   -

mweitzel2005, 2022-01-14:
I went into `Wrench` (Settings) > `Webcam & Timelapse` and checked both boxes for

1. `Flip webcam horizontally`
2. `Flip webcam vertically`

While these settings work and can be verified on the `Control` tab of the OctoPrint interface they don't take effect on the `Octolapse` tab and don't reflect in the resulting videos from Octolapse either. 

Did I misunderstand or do something wrong?

FormerLurker, 2022-01-14:
@mweitzel2005, those options only affect the stream display, and cause some CSS to rotate the image on the browser screen.  They do not actually modify the stream in any way, only the way it's stuffed into the webpage.

Octolapse has transformation options in the camera settings, and has more capabilities than the 'Webcam & Timelapse' settings.  I've circled the option you want (I assume your camera is upside down):

[image]

Octolapse will physically transform both the full res image and the thumbnail, so it does take a very small amount of time to accomplish this (very very little for 1080P images, and more for massive DSLR images).  At some point I will move this to the 'rendering' phase so that there is zero performance impact while printing.

mweitzel2005, 2022-01-15:
@FormerLurker thank you so much for the detailed response! Awesome project!!