FormerLurker/Octolapse #93: [Request] Enable/Disable through main panel

normen, 2018-03-25:
#### If this is a feature request describe it here
Being able to enable or disable Octolapse without having to go through the settings panel would be nice. The green button in the Octolapse tab looks like the ideal candidate to bind that function to imo.

#### Version of Octolapse
0.2.2

#### Version of OctoPrint
0.3.7rc2

Thanks for this awesome software!

FormerLurker, 2018-03-25:
Good idea, and thanks for the compliment!  I JUST added the settings controls on the main tab over the last week or two, and have been planning on adding some additional control, so this fits into my plans nicely :)

Do you think it will be confusing that the other buttons around the power button will not be clickable if I attach it to the green power icon you mentioned?  Maybe I should put something within the 'current settings' expandable section instead?  I have a few bugs to fix first, so we have a bit of time to think about this.

normen, 2018-03-25:
Well it was just an idea, given that you can click the ones below which enable the different monitor options I don't think it would be too confusing. You could also add them where the monitor options are I suppose, kind of mirroring the other settings tab. Or make the settings accessible directly in the main tab through checkboxes or somesuch - but that'd obscure the fact that you can still open a "big" settings panel over there.

Btw, I seem to have found a small bug where when Octoprint is started with Octolapse disabled Octolapse won't display all of it's panels - even if it's enabled later. It seems to be reproducible - shall I make a new issue for that or do you know about that already?

FormerLurker, 2018-03-25:
No, that's news to me.  Please do post a new issue.  Just fill in the steps to repeat and I bet I can get it solved quickly.

I had thought about including a button to open some of the settings pages directly from the tab.  It's nice to know this is a desirable feature.

So many options!  I'll think it over and will let you know what I decide.  Thanks again for your suggestions and error reports!

FormerLurker, 2018-03-30:
Added to v0.2.4

closed 2018-03-30 by FormerLurker

FormerLurker, 2018-03-31:
62a462a