FormerLurker/Octolapse #97: Watermark option broken

FormerLurker, 2018-03-25:
The watermark option has never worked properly.  Fix this and allow custom watermark upload.

Shadowen, 2018-05-02:
Watermark option seems to be working fine. See my branch https://github.com/Shadowen/Octolapse/tree/watermark.

I added a test for rendering (with and without watermark). You'll need to run the test and pause it with a debugger before it cleans up (in `tearDown`) to see the rendered video.

Shadowen, 2018-05-13:
Lol, I built that progress bar half from scratch :/
I'm pretty sure there's a better one in a JQuery plugin included somewhere already...

You might need to package the OctoPrint watermark into out `/data/octolapse/watermarks` directory, or wait for me to implement the `config.yaml` config for the directory, since half the feature is hard-coded to load the watermark from that directory...