FormerLurker/Octolapse #99: Octolapse panel broken when disabling Octolapse

normen, 2018-03-26:
#### Version of Octolapse
Octolapse Version: 0.2.2

#### Version of OctoPrint
1.3.7rc2
OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Set Octolapse to disabled in the settings
2. Restart Octoprint
3. Open the Octolapse tab

#### What should have happened?
The Octolapse panel is displayed with all options

#### What happened instead?
The Octolapse panel only has the first tab (settings).

#### Operating System running OctoPrint and Octolapse
OS Name: Raspbian
OS Version: Stretch

#### Browser and version of browser, operating system running browser
Browser: Safari 11.0.3
Browser OS: MacOS 10.13.3

FormerLurker, 2018-03-26:
Could be a safari issue, can you post the JavaScript console output if there are errors there?  Also, there seems to be a bug in initial installs, so maybe try a reinstall first?

Thanks for submitting this report!

normen, 2018-03-26:
I reinstalled the plugin and it still seems to happen. No console log output thats out of the ordinary afaics...

Heres a screenshot of the settings panel when this happens:

https://i.imgur.com/u2HJkJy.png

FormerLurker, 2018-03-26:
Can you edit the main settings?  What happens if you enable then disable Octolapse in the main menu?

If that doesn't do anything, can you try restoring the default settings and see what that does?

Finally, if none of this helps, try re-installing from the plugin manager using this URL:

https://github.com/FormerLurker/Octolapse/archive/master.zip

I'm a newb when it comes to the plugin manager (first time plugin here), so I'm not sure if it's pulling from the latest release or from the latest commit.  If that URL works, then I'll know at least :)

We'll get it working!

FormerLurker, 2018-03-26:
Hey, I've finally been able to replicate the issue.  It will take a javascript change.  I'll let you know when the code is up, then you can re-install and try again.

Sorry about that!

FormerLurker, 2018-03-26:
Oh man, that's one of those bugs I was ashamed to push into production :)  Thanks for spotting it early for me!

I'll leave this open until you get a chance to reinstall and try again.

FormerLurker, 2018-03-26:
Oh, I forgot to mention that after you reinstall and reboot you'll need to clear your cache.  OctoPrint combines all of the plugin js files into one, and it stays in the cache even after an install/reboot cycle (boo...).  If memory serves you'll need to use CMD + R to force a refresh for safari.

normen, 2018-03-27:
I (re-)installed from that URL you gave me, no idea which version that is/was as the panel still says 0.2.2 and the bug still seems to be present.

normen, 2018-03-27:
Wait, I misspoke - I didn't realize that the panels not showing was a feature when octoprint is disabled :) They DO show now when I simply enable octoprint, without a restart.

FormerLurker, 2018-03-27:
And you reset the cache with cmd + r (screen should reload)?  If so, I'll push another branch and will add console logging so we can figure out what is happening.

FormerLurker, 2018-03-27:
Oh man, you made my day!  Would you say the issue is closed?

normen, 2018-03-27:
Looks like it yes :) Thanks!

closed 2018-03-27 by FormerLurker