FormerLurker/Octolapse #102: Unable to restart Octoprint Server after install

Crowlord, 2018-03-27:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.3.7RC2 on OctoPi 0.13

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

Updated from 1.3.6 to 1.3.7RC2 to accommodate plugin
Removed Navbartop plugin to tidy up
Removed TouchUi to tidy up
rebooted server
installed plugin
server would not reboot
last entry in log is reboot command
octoprint server will not start under any command

#### What should have happened?
plugin should have installed

#### What happened instead?
server broke

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.13

#### Printer model & used firmware incl. version

Printer Model: POriginal Pruas i3 Mk2
Printer Firmware Version: 3.1.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

Crowlord, 2018-03-27:
May be a non-issue but thought it was worth mentioning in case it isnt.

FormerLurker, 2018-03-27:
That's certainly something worth looking into, thanks for the report!

So, if you power down and back up does OctoPrint load now?  Also, after the reboot command what happened exactly?  Did Octoprint shut down without the pi actually rebooting, or did Octoprint just continue running?

Can you send me the boot log?  I can't really remember where it is, maybe /var/log/syslog?  If you have trouble finding it I'll do some digging and figure out where it is.

Thanks again!

Crowlord, 2018-03-27:
I'm really sorry the logs are all now gone in the reinstall.

A full reboot of the pi didnt fix it, octoprint shut down as expected by internal command to load the new plugin and never came back. Pi continued to function as normal. 

One difference of note is that my setup runs the OctoPi desktop environment that you install by script. It also runs a 5" touchscreen. 

Hopefully its just me, I am sure if its a major issue that you will have heard more by now as I expect the plugin is getting some love :D.

As my install is nice and clean now I will try installing again and see what happens.

Crowlord, 2018-03-27:
Is there an enhanced log argument you would like me to enable before I jump in

FormerLurker, 2018-03-27:
I understand about the missing logs.  Often you only know you need them after the fact.  I run into this frequently.  When OctoPrint 1.3.8 releases asynchronous logging will be introduced, which should make it easier to keep some logs running all the time with little or no performance impact.

If you have install problems again see if you can get to this file:  plugin_pluginmanager_console.log

If you have problems you won't be able to access the web app, but it should still be available within this directory: ~/.octoprint/logs

Good luck!

Crowlord, 2018-03-27:
Worked as expected this time. Hopefully, my install had just gotten a bit rusty!

FormerLurker, 2018-03-27:
Such good news!  I half expected to spend the evening flashing and re-flashing an SD, running install after install to track this down.   Now I can spend the evening looking at other maddening issues instead, lol!

Thanks for your issue report and for trying out the plugin!  It's been tested pretty well on the MK2 + the default slic3r PE profiles (I printed 4 benchy's just yesterday), so you should be able to get pretty decent results by leaving everything set to the defaults, and choosing the Mk2 profile (check retraction length if you've made any changes, or aren't using the most recent version of Slic3r PE).  Let me know how it goes.

closed 2018-03-27 by FormerLurker