FormerLurker/Octolapse #103: Octolapse can miss some gcodes at the beginning of the file

FormerLurker, 2018-03-27:
In some situations Octolapse may not see line 2,3 or 4 of the gcode file.  This is either a problem in the startup routine (timelapse.py gcode_queuing), or has something to do with with asynchronous event processing within OcotPrint.

I'm working on this issue and will post a workaround if I can't solve it soon.

skorokithakis, 2018-03-27:
Would it be possible for you to insert dummy codes in the beginning, just to make sure the real ones are pushed farther down?

skorokithakis, 2018-03-27:
Inserting a bunch of:

```
M82
M82
M82
M82
M82
```

at the start of everything seems to fix this issue.

FormerLurker, 2018-03-27:
It should work.  However, I think I have this one licked!  Try reinstalling.

Additionally I put a check in to make sure EVERY line of the gcode file is received in order.  If not, an error message is displayed and the timelapse will (hopefully) end, but the print will (hopefully) continue.

skorokithakis, 2018-03-27:
Will do now! From master, right?

FormerLurker, 2018-03-27:
That's the one!  I probably should be pushing all of this onto a separate
branch, but I figure while we are in RC versions of Octoprint it's probably
fine.

FormerLurker, 2018-04-01:
It looks like this has been solved!  I'm going to mark it closed and will consider any similar issues to be new.

closed 2018-04-01 by FormerLurker