FormerLurker/Octolapse #104: Octolapse Timelapse Stopped ???

interceptor70, 2018-03-27:
The Printer does not recognize the End of the printing process.
This is the Error Message:

Octolapse Timelapse Stopped
A timeout occurred when attempting to acquire the printer's current position.The current timeout is 600.0 seconds. Stopping timelapse.  

What can I do against it?

FormerLurker, 2018-03-27:
Thanks for submitting the issue.

Can you post the contents of octoprint.log and plugin_octolapse.log?

Did the error occur after the print was completed entirely, or somewhere in the middle?  Also, was a timelapse generated?

Also, there have been some changes to the plugin.  It might be a good idea to reinstall.

FormerLurker, 2018-03-29:
Are you by chance using a printer that runs repetier firmware (Wanhao I3/maker select, etc..)?  If so, this issue may have been fixed in the newest version of Octoprint - 1.3.7rc3.  Please let me know because I'm trying to gather reports for this firmware for Gina so we can confirm this bug is fixed.

Again, thanks for reporting the issue!

FormerLurker, 2018-04-11:
Any update on this issue?  This very well could have been solved in one of the updates to Octolapse that has occurred within the last two weeks.  Can you please try updating to v0.3.1 (should be able to do this in the software update plugin) and let me know if it's working?

Thanks!

GeekDad63, 2018-05-02:
I got the same thing and here are the (droids) er, logs you're looking for.
[plugin_octolapse.2018-05-01.log](https://github.com/FormerLurker/Octolapse/files/1969192/plugin_octolapse.2018-05-01.log)
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/1969194/octoprint.log)

FormerLurker, 2018-05-03:
@GeekDad63, can you post your gcode?   It looks like the parser detected some problems, but whether these are parser or gcode related, I cannot yet tell.  The good news is that it will be pretty easy to debug.

While I'm at it I'll update the error message to include the line/gcode command that threw the error if possible.

FormerLurker, 2018-09-10:
@GeekDad63  and @interceptor70, 

Have you tried updating to OctoPrint v1.3.9 and updating Octolapse to the latest release by using this URL in the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4rc1.dev2.zip

Here is a video explaining how to configure the new build (things have changed quite a bit):  https://youtu.be/sDyg9lMqMG8

There were some deadlock issues that were resolved, and that's probably what you were experiencing.  Please let me know if that helps, and sorry this issue has taken so long to resolve!

FormerLurker, 2019-10-12:
This issue has been solved, so I'm going to go ahead and close it down.  If anyone continues to have this problem on Octolapse V0.3.4+ and is using Octoprint 1.3.9 or higher, please create a new issue.

Thanks for posting!

closed 2019-10-12 by FormerLurker