FormerLurker/Octolapse #105: OctoLapse and mesh data

Thisismydigitalself, 2018-03-27:
Very impressive plugin.
does octolapse collects mesh data? what data, if any, octolapse collects?

FormerLurker, 2018-03-28:
Hi!  Can you email me at FormerLurker@protonmail.com and we can have a discussion?  I'd like to keep non-issue issues to a minimum if possible ;)

FormerLurker, 2018-03-30:
Hey, if I'm understanding you correctly, I don't collect any mesh data (like in .stl files and such, right?).  Octolapse watches the gcode as it's sent and keeps track of the position history for each axis(the last few positions are kept) as well as some additional state information (is zhop, is retracting/retracted, etc.).

Does that answer your question?  If not perhaps you can elaborate a bit.

Thisismydigitalself, 2018-03-31:
Thank you for clearing this up. I admit that when you wanted to continue the discussion outside of github I thought the plug might collects private data such as stl/gcode/etc. the reason I was asking was following my understanding that octolapse, very smart plug i should say, must parse the gcode in order to know what's going on. I will reinstall the plug but after OP 1.3.7 will get released. these days we know nothing about what a piece of software is doing and I always try to learn if personal data is collected and shared without user explicit consent.

Thank you and again, - Chapeau.

FormerLurker, 2018-03-31:
OH, I see, lol!  No, I'm not hiding anything.  Octolapse is not sending your data anywhere, and the source is available to look through.  The only data it collects is what you enter into the settings and whatever gets captured from your camera, unless I'm forgetting something.  You can see everything it stored in the ~/.octoprint/data/octolapse directory.  

I might note that you can save camera passwords in the camera profiles, but this is not very secure.  The passwords are stored in plain text within the settings.json file.  I hope to improve this at some point, but it's noted with a warning within the camera profile settings.

FormerLurker, 2018-04-03:
I'm going to go ahead and close this issue.  Will reopen if there is anything else you need for this issue.  Thanks again!

closed 2018-04-03 by FormerLurker

Thisismydigitalself, 2018-04-06:
> The only data it collects is what you enter into the settings **and whatever gets captured from your camera, unless I'm forgetting something**

Sorry mate. Truly amazing plugin but I will pass. I am willing to pay for products that don't collect private stuff but I guess I'm not leaving in such a world.

FormerLurker, 2018-04-06:
Is this supposed to be a joke?  The point of octolapse is to take snapshots and create a timelapse, how could it work without saving them?  NO DATA IS HARVESTED.

Thisismydigitalself, 2018-04-07:
No worries mate. No harm no foul.
The fact that you've had your wife read our conversation clearly means both of us practice the same mantra: "Whenever there is any doubt, there is no doubt".

We all live in a world which we know nothing about. some live with this fact just fine. others like myself are merely trying to learn what's going on under the hood. So thank you for clearing up this issue once and for all. You owe us nothing and so I thank you for sharing your great work which seems so many are enjoying daily. 

Keep up the good work.