FormerLurker/Octolapse #106: Cam position set after new layer is started

normen, 2018-03-28:
#### Version of Octolapse
Octolapse Version: 0.2.2

#### Version of OctoPrint
OctoPrint Version: 1.3.7rc2

### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Download the 50mm tower as suggested in the wiki
1. Slice the tower with 0,3 layer / 0,48 width / 2 outlines (GCODE file below)
2. Print with Octoprint enabled

#### What should have happened?
The camera position should be set before the new layer is started.

#### What happened instead?
When the layer switch happens one side of an inner wall of each new layer is printed before the "camera position" is set.

#### Operating System running OctoPrint and Octolapse
OS Name: Raspbian
Os Version: Stretch

#### Printer model & used firmware incl. version
Printer Model: RigidBot Modded
Printer Firmware Version: Marlin 1.1

#### Browser and version of browser, operating system running browser
Browser: Safari
Browser OS: OSX

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://www.dropbox.com/s/nixgx79pjyrgy9q/50mm-tower.gcode?dl=1

#### Link to settings.json
Link to settings.json with all passwords removed: https://www.dropbox.com/s/29872emq969mxu6/Settings.json?dl=1

FormerLurker, 2018-03-28:
Can you elaborate a bit, or maybe post a video?  I'll print the GCode with a virtual printer in a bit, but clarification is always good!  Maybe post a video of your print as it happens?  You can use test mode (debug settings) so you don't have to wait as long.  Unload your filament just in case, and check out the wiki pages (debug settings, test mode).  Thanks for your report!

ejjenkins, 2018-03-28:
Mine does the same thing.  It starts the next level...then moves the camera to take a picture.  This seems to cause the retraction not to happen so the stringing is crazy as well.  I can do another print and post all of my files...just not tonight.

normen, 2018-03-28:
Yeah, I didn't have my phone handy when it happened the first time and I'm not sure when I'll do another print with Octolapse again as I intend to do some non-test prints and they'll take time..

But basically it just did the first straight line of the next layer before going to the cam position. I'll make a video when I can.

FormerLurker, 2018-03-28:
@ejjenkins , are you using the same printer as @normen , or something else?  

I'll wait for your logs/videos, but in the meanwhile I'll run the gcode I received through the virtual printer with your settings.json file and see what happens.  I'm sure it's some subtlety I overlooked :)

Thanks to you both for helping me improve the plugin!

ejjenkins, 2018-03-28:
I am running a D-Bot (CoreXY) with Marlin 1.1...it is a beast 335mm x 635mm heatbed.  I'll do a small print tomorrow and post the files.

normen, 2018-03-28:
In the meantime heres the actual timelapse... If you have pixel-interpolating eagle eyes you can make out the thin additional line on the side of the camera :)

https://www.dropbox.com/s/l87uw2kj64pxtbc/50mm-tower_20180325134952.mp4?dl=1

ejjenkins, 2018-03-28:
Mine does the same...it takes some shots while over the object as well...not all pics are from the desired location.

normen, 2018-03-28:
You can increase the time the head rests in position if your cam is late with the picture.

FormerLurker, 2018-03-28:
Wow, that is really odd and illustrates what you are talking about.  Thanks for your video @normen !  I'm about to run your gcode but I have to hit the sack here soon.  Might be tomorrow before I can figure out what's going on.

FormerLurker, 2018-03-28:
Well, when I ran your code through the virtual printer it takes snapshots here on every layer:

X247.500 Y105.000

This could be something particular to this style of printer.  Do you know if you have inverted axes or if the home position is in the center of the bed?  I haven't tested these two classes and have been wondering how they respond, though I didn't expect to see this!  In any case, I think I'll need all of your logs (plugin_ocotlapse.log and a copy of the terminal output), at least for the first dozen or so layers, so I can see both layers that take the snapshot correctly and those that don't.

The log files got what programmers crave, so before you try anything check your debug profile and make sure it's set to Diagnostic (you can use test mode if you want for this debug).  Look inside of the debug profile settings and scroll to the very bottom.  Make sure 'log gcode received' is there and enabled (unchecked by default).  If it's not there reinstall and it should appear.

K, I'm heading away for the evening, later!

normen, 2018-03-28:
Home is front left for that printer, not center. The Y axis is "inverted" on my printer but I don't think that plays much of a role here. But I'll enable logging when I next try a print with Octolapse.

Also, the problem isn't so much the position for the cam when the photo is made but the fact that the printer starts the next layer before making the photo.

ejjenkins, 2018-03-28:
I removed and reinstalled OctoLapse, swapped the USB cable on my PI to a better cable with ferrite rings, and setup the plugin again from scratch.  The only setting change I made (that I can think of) was to add more time for the camera (snapshot delay).  I ran the same g-code file today using Full Diagnostic - Live Print, and it worked just like it was supposed to.

FormerLurker, 2018-03-28:
Wow, sounds like issues with resend requests then!  There is another issue I'm dealing with right now caused by resend requests.  Thanks for reporting.  @normen , maybe this could be your issue too?  I'd really like to see the octoprint.log file for this (and serial.log if it's available) to confirm.

You can calculate the optimal delay for your camera if you know the framerate.  In general you want to wait 1/framerate seconds (multiply by 1000 to get MS), maybe a few MS more.  Autofocus can mess with this, so be sure to manually focus (see the camera profile settings, check out the camera profile wiki for details).  You'll get way better timelapses with manual focus.  Pretty much everything should be manual ideally.  Not sure what camera you're using.  I know lots are compatible with the camera preferences (contrast, zoom, focus, etc), but maybe not yours.  Let me know if you have problems.

normen, 2018-03-28:
Again, the resulting video isn't the issue here, the issue is that the print is interrupted mid-layer by the cam position movement. I'll look into trying this again tomorrow, both with and without logging and with a phone handy to make a video of what the actual issue is.

FormerLurker, 2018-03-28:
Sorry @normen, I was just explaining the optimal snapshot delay to @ejjenkins since he mentioned turning it up.

I believe I understand what's going on from the timelapse, but a video of the issue as it occurs would be perfect.  Logs are the lifeblood debugging, so I'm always happy to get those too. 

Thanks!

normen, 2018-03-29:
Sorry myself, didn't want to sound demanding, just wanted to make sure the actual issue is pinned in this issue thread :)

normen, 2018-03-30:
So heres a video showing the issue with the layer being started before the cam position is set.

https://youtu.be/8994z1cRd2o

Heres a link to the log, I canceled the print not long after the video ends.

https://www.dropbox.com/s/koh1hfaweyqseqo/plugin_octolapse.log?dl=1

FormerLurker, 2018-03-30:
I really didn't understand what you were talking about before, but the video makes it clear.  Thanks!  Also, please excuse the long response, but this is complicated to explain.

I think I know why this is happening, and it's not exactly a bug, though maybe is not desirable behavior depending on the print.  Fortunately the behavior can easily be changed via settings.  Let me start of by explaining how Octolapse detects layer changes.  Octolapse watches for extrusion and keeps track of the maximum height on which extrusion has so far occurred.   There are some subtle reasons for doing this, but I won't get into that right now.

So here's what I think is happening.  The default snapshot profile - Layer Change - has some settings that are causing this behavior.  Look at the 'Layer Trigger Extruder State Requirements' (see the snapshot profile settings-> layer/height trigger -> Advanced Layer/Height Trigger Options ) has 'On Detracted' set to Forbidden.  This is to prevent Octolapse from doing a double retract/detract, which it would do without this setting enabled in this case  (difficult to explain why this is, but if you want I can give you some gcode examples, also lift and retract can be disabled too!).  So right at the end of a layer your printer is doing a lift and retract, followed by a detract/extrude.  Octolapse doesn't detect the layer change until it sees extrusion at a height above the previous height.  Since Octolapse is 'forbidden' from taking a snapshot right after a detract, it finishes the extrude before taking the snapshot.

Fortunately there are several things you can do about it.  This is the easiest:  Your slicer is probably lifting and retracting after each layer, correct?  In this case use the 'Layer Change - ZHop & Retract only' profile.  Look through the layer trigger settings to see what's going on here.  Basically AFTER it detects a layer change it will wait for a lift/retract before taking a snapshot.  It even works if 'wipe' is enabled in your slicer!  If you turn the trigger state info panel on while running your next test you'll see just what's going on with this profile.

In the worst case scenario using the ZHop & Retract only profiles you might miss a snapshot during the first layer since there might be no hop + retract before the first later completes.  If you prime directly on the bed this shouldn't be a problem, but you might notice some odd behavior here too (I'm working on this one):  Your printer may lay a priming line down, then take a snapshot, then return to the end of the priming line before moving to start printing your first layer.  This is because Octolapse always returns to the previous position before continuing.  This is so any extrusions start in the right place.  I am working on improving this behavior by NOT returning to the starting position if the command that triggered the snapshot was a travel only (no extrusion).  It turns out you can fix this behavior too with a mix of changes to your priming gcode and your Octolapse printer profile.  I can give you more info about this if you care.

The reason the 'zhop and retract' profile isn't set as the default when you install Octolapse is because some gcode files wouldn't produce a single snapshot with this profile (gcode without any lift/retract).  I wanted the default profile to work for the maximum number of cases.

The second thing you could do in this case (varies from print to print) is to use a position restriction to force Octolapse to ONLY take snapshots within a certain area.  I would use a rectangle and set it to 2mm or so narrower than the object you are printing in this case.  Enable intersection calculation so that Octolapse can take snapshots even if it's in the middle of an extrusion if it intersects with the position restriction.  This way you'll be able to take all of the snapshots inside of the infill (except for the top and bottom surfaces).  You can also create areas where snapshots are forbidden, and you can mix and match the two.  It's really a pretty powerful feature, but it is also a REALLY REALLY new (week or two) and could (probably does) have some bugs that might ruin your print.  It is currently a little cumbersome to use, but I've had some pretty fantastic results.  For example I printed in spiral vase mode (MUST USE A HEIGHT TRIGGER) and restricted the snapshots to the INSIDE of the vase.

I'm sure that was all as clear as mud, so please ask for clarifications.  I need to explain this stuff in the wiki and settings pages better, but I'm not very good at that sort of thing.  Any help is appreciated!

normen, 2018-03-30:
I see, I'll try that.

FormerLurker, 2018-04-13:
I'm going to close this issue.  Let me know if you're still having problems and I'll open it back up!

closed 2018-04-13 by FormerLurker

normen, 2018-04-14:
Yeah sorry, my printer broke down so I couldn‘t try again yet.

FormerLurker, 2018-04-14:
Ouch!  I hate it when that happens.  Especially when your printer is down and you cant print a replacement part because your printer is down.

normen, 2018-04-14:
Luckily I still have an Ormerod but that has it's own web interface so I can't test Octoprint there. New power source is coming from China apparently - I really have to check better when ordering on Amazon :)