FormerLurker/Octolapse #107: Installation issue on 1.3.7rc2

ganey, 2018-03-29:
#### Version of Octolapse
 
Octolapse Version: Master

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc2 running on OctoPi 0.14.0

 
--

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: N/A

#### What were you doing when the problem occurred

1. Visit plugin repo
2. Search Octolapse
3. Click Install

#### What should have happened?
Octolapse should install

#### What happened instead?
Installation fails, have attached plugin installation log.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Browser and version of browser, operating system running browser

Browser: Chrome (Latest)
Browser OS: Windows 10 x64

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/G83gz8CN

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/NN6z9X5G

FormerLurker, 2018-03-29:
It looks like maybe you ran out of either space or memory while Pillow (image manipulation library for python) was being installed.  Double check your free space for now, and when I'm in front of a PC I'll send instructions for installing pillow manually, and we'll see what happens with that.

You could also try rebooting your pi and reinstalling if you haven't tried that yet.

Thanks for reporting this!

ganey, 2018-03-29:
I've tried a reboot with no luck unfortunately.

How big is the install? If needs be I can swap to a larger SD and re-setup OctoPi. 

Will it be okay if I just `pip install pillow` and try with that?

```
pi@octopi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.8G  1.4G  289M  84% /
devtmpfs         55M     0   55M   0% /dev
tmpfs            59M     0   59M   0% /dev/shm
tmpfs            59M  4.4M   55M   8% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs            59M     0   59M   0% /sys/fs/cgroup
/dev/mmcblk0p1   41M   21M   21M  51% /boot
```

FormerLurker, 2018-03-29:
289MB should be more than enough.

Yeah, try pip manually (I wasn't sure how familiar you were with the
command line).  If you get a memory error try stopping the octoprint
service, that should free up quite a bit.  Once it's installed it shouldn't
be a problem.

If you have problems, send me the console output of the pip command, thanks!

ganey, 2018-03-29:
pip install pillow was still failing with the MemoryError.

So I tried this which usually helps with other libraries on older RaspberryPi's lacking in memory.

`pi@octopi:~/oprint/local/bin $ ./pip --no-cache-dir install pillow`

Octoprint then installed successfully through the settings - plugin manager

Thanks for your help!

FormerLurker, 2018-03-29:
Excellent!  I'll see if I can add that to setup.py somehow.  I didn't
realize how much memory it took to install that.  It's ONLY used right now
to render a thumbnail, though I'm going to add some other stuff (rotate,
filter, mirror snapshot, etc..) soon.

Hopefully the plugin will work for you now.  If not, create another issue
and I'll hop on it!  I want to fix as much as possible before OctoPrint
1.3.7 stable releases.

FormerLurker, 2018-03-31:
@ganey , have you posted a pull request with OctoPrint for your commit here?  I'm wondering what Gina has to say about your proposed solution.  I'm going to keep this issue open for a while to see how that turns out.

ganey, 2018-03-31:
Not yet, I need to sit down at a pc rather than trying to do it from my
phone.

I still need to run the tests and figure out which octoprint branch to
submit the PR to, but I think this might sort it:

https://github.com/ganey/OctoPrint/commit/38295cfebfdd05db407e01addc846aae8d0295e2

ganey, 2018-03-31:
I have now submitted a [PR#2535](https://github.com/foosel/OctoPrint/pull/2535) so will see how it turns out.

FormerLurker, 2018-04-01:
Excellent!  I'll leave this issue up until we hear back about that.  Maybe other people will run into this.

ganey, 2018-04-03:
Just an update: https://github.com/foosel/OctoPrint/pull/2535 is staying as 'needs work' while I / someone figures out getting this to run on lower memory devices only / via a config.

The change has been merged into `maintenance`, hopefully to come out in 1.3.8 😄  https://github.com/foosel/OctoPrint/pull/2543

FormerLurker, 2018-04-04:
Excellent!  I really appreciate the effort!  Making octoprint better makes octolapse better too and vice versa.

Honestly, I'd rather be able to control these sorts of things within settings.py, but the default should work with low mem devices if possible!  Some improvements are needed there.  For example, and it may be my own ignorance talking here, but I can't figure out any way to limit which versions of OctoPrint are OK to use.  I can do this with other packages, just not OctoPrint.

FYI v0.3.0 will drop soon.  If you want to test drive try installing from the devel branch within the OctoPrint plugin manager with this URL: https://github.com/FormerLurker/Octolapse/archive/devel.zip

It's pretty stable, but not 100% debugged.  I've got about 5 more prints in the queue to test, and am waiting to hear back from a few other users to make sure nothing major is broken.  LOTS of changes including:

* Brand new faster/better gcode parser
* Open the current profile and edit directly from the TAB (no going into settings)
* More printer profiles profiles (need to restore defaults to get them, boo...) with tweaked settings.  
* Less bandwidth usege
* Fixed some VERY annoying issues that were requiring a page reload after print cancel and disconnect to start another timelapse.
* Snapshot post processing (transpose, thumbnail and latest snapshot save) now happens in a thread
* Implemented a rendering queue so that if you manage to finish a print before the previous rendering is complete things should still work.  I've not actually been able to accomplish this so I'm not 100% sure it will work, lol!  It does in the virtual printer, but lots of things work there that don't work on another printer, so hard to tell.
* More stuff too, but I can't remember :)

Let me know if you get a chance to try it out before v0.3 drops.  I'd love to get some additional feedback from some more advanced users!

Closed!

Nuttavoot, 2018-07-13:
I am not using Octopi but install Octoprint from scratch and cannot install the plug-in, this is my log.

[plugin_pluginmanager_console.log](https://github.com/FormerLurker/Octolapse/files/2192422/plugin_pluginmanager_console.log)

abol4671, 2018-07-21:
Install is failing for me on the latest Octopi image as well. Attached is the plugin log:
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/Octolapse/files/2216378/plugin_pluginmanager_console.log)

FormerLurker, 2018-07-21:
This is all related to pillow.  Drat, that library is giving me some trouble.  Does anyone have experience with these sorts of setup.py problems?  I think I need some assistance, especially since the install seems to work for many.  At this point I'm not sure if I should open an issue on the Pillow github or OctoPi.  Maybe both.

abol4671, 2018-07-24:
Sorry this ones beyond me. I spent the better part of an evening trying different things with no luck. I didnt see anything on the Pillow GitHub that was helpful either.

FormerLurker, 2018-07-24:
@abol4671, I'm working to get to the bottom of this.  I am running down some leads.

FormerLurker, 2018-07-25:
I'm reopening this issue since I've gotten a few other reports of Pillow installation issues.

FormerLurker, 2018-09-10:
I'm closing this issue down and have created issue #234 to cover this problem.  If anyone has any further issues installing due to Pillow (PIL) errors, please comment on the new issue.

Thanks!

closed 2018-09-10 by FormerLurker