FormerLurker/Octolapse #108: Repetier firmware (Monoprice Maker Select v2/Wanhao Duplicator i3, etc) bugfix test

FormerLurker, 2018-03-29:
A bug was reported when using Octolapse with Repetier firmware causing lockups during the snapshot process.  A fix was introduced in OctoPrint v1.3.7rc3, and I need someone with one of these printers to try it out.  If the print completes with no extremely long pauses (600 seconds to be exact), then we can celebrate!

If you're interested in helping, upgrade to OctoPrint v1.3.7rc3.  [Here is a blog about the release, including instructions on how to upgrade.](https://octoprint.org/blog/2018/03/29/new-release-candidate-1.3.7rc3/)

If you do run into a long pause while Octolapse is taking a snapshot, you can go to the terminal tab in OctoPrint and send the following command:

M114

This is what Octolapse is waiting forever to get, and sending the command manually will resume the print.  If you do experience a lockup, just be sure to give it a minute or so before issuing an M114 so we can rule out delays in the printer's response.  

Please be sure to select one of the diagnostic profiles for your debug settings before starting your print.  That way if there IS a problem we'll be able to debug it.  Just post your plugin_octolapse.log file, octoprint.log, and if possible either the terminal output (terminal tab) or serial.log (if you've enabled serial logging).  

Also, be sure to check out [test mode](https://github.com/FormerLurker/Octolapse/wiki/Debug-Profiles#test-mode) in the wiki if you want to use the 'Full Diagnostic - Test Mode' debug setting.  It's a pretty cool feature.

Thanks in advance for helping make Octolapse and OctoPrint better!

FormerLurker, 2018-03-29:
I've gotten one report of success on the Wanhao I3!  More feedback here would be extremely valuable, so please post your success/failure with Repetier firmware.

nickm324, 2018-04-01:
Testing maker select v2 on a 12 hour print. Will report results.

cityeyes, 2018-04-02:
Maker Select V2 here - I've got a few successful runs with no huge issues. I HAVE noticed that a small column forms at the location where the print head returns for the timelapse. It extends from the top of wherever the print surface is all the way to the top level of the print, like some sort of filament stalactite. I presume something might be wrong with retraction settings. Not sure if it's an issue on my end, but I haven't had issues with other prints.

Outside of that though, nothing worldbreaking. My prints have ranged from ten or so minutes to a few hours.

FormerLurker, 2018-04-02:
I think I need to separate that profile into two.

nickm324, 2018-04-02:
Maker Select V2 16 hour print no issues that I am aware of.

edsvcs, 2018-04-03:
Wanhao Duplicator i3 v1 here. Several successful prints ranging from 30 minutes to 9 hours. Plenty of fun getting OctoPi to recognize and properly capture HD video from the Pi HD camera, but once I got that working I've had no issues attributable to Octolapse.

FormerLurker, 2018-04-03:
@nickm324, did you change the retraction in the printer profile for the Maker Select? I've heard the beta profile is way off for the Maker Select V2, and am curious.
@edsvcs, Did you find the retraction to be correct for the Wanhao?

I'm trying to figure out if I should split these profiles into 2.

nickm324, 2018-04-03:
I did not change any settings, all defaults.

cityeyes, 2018-04-03:
I should probably clarify - I technically had the wrong model listed when I first commented - I've got a Wanhao Duplicator i3 _plus_ - otherwise known as the Maker Select Plus.

They Duplicator i3 and the duplicator i3 plus are the same machine, just configured a bit differently. I don't think there would be a disparity between the two settings-wise, but who knows?

That said, my filament leaking issue could 100% be caused by the type of filament I'm using and my other settings, if it's not happening to anyone else. It can't hurt to bump the retraction settings up a tiny bit, though, because most people who use this category of 3D printer say they've got it at 2-3mm, still a decent measure above the default .6mm.

FormerLurker, 2018-04-03:
Thanks, that does clarify a lot.  Regarding bumping up the retraction, it
should really be set to the same value as the default slicer profile for
retraction detection to work.  Do you know what the default value is off
the top of your head, or can you double check for me?

edsvcs, 2018-04-03:
I started printing with PETG at the same time that I connected my OctoPi and Octolapse ... so yes I am having a lot of stringing and some globbing, but I assumed it's just because I don't yet know how to work with PETG... I won't have another chance to print PLA or ABS to see if Octolapse is causing any issues until at least the weekend.

FormerLurker, 2018-04-03:
@edsvcs, I"m not saying it isn't Octolapse (it could be), and I've not printed with PETG, but have heard it can adsorb water pretty quickly.  If you have a vacuum chamber or a dehydrator you could try drying it out and see if that helps.  Do some research first, but I think it needs to dry at about 60C.  Or you could try printing with some of your fresh ABS/PLA.

It's not a good idea to do this in an oven or anything else without precise temperature control, btw.

I'm planning to do a vacuum chamber experiment to see if I can dry out some super old PLA and will let you know how that goes.  I already had the chamber for other reason.  Fun side project :)

edsvcs, 2018-04-03:
I opened this PETG fresh from factory vacuum sealed bag and I live in Utah where we have 17% humidity today but only because it's been raining... haha. Great climate for 3D printing.

cityeyes, 2018-04-05:
Sorry for the late reply - I've actually found the default cura profiles to overdo the retraction settings a bit. Since Wanhao i3 isn't a default printer with Cura, a lot of people suggest using the Prusa i3 settings. For that profile, retraction is 6mm, which most people agree is overkill.

Most people online suggest a retraction setting between 2 and 3 for the Wanhao/Maker family, going up to 5 or so if stringing is still out of hand doing test prints. Mine is at 3 currently.

I've found that putting retraction to 2 or 3 in the Octolapse settings gets rid of the little column of printer medium that was forming under the "timelapse" location, so that's what I stopped on.

edsvcs, 2018-04-10:
Following up on this. I have now printed more than a dozen successful prints in PLA. Great time lapse videos and no issues.

FormerLurker, 2018-04-10:
@edsvcs, that's great!  I'd love to see some of the videos if you don't mind posting them.  I love looking at these timelapses, which is why I built the plugin :)

FormerLurker, 2018-04-10:
I'm going to mark this issue as closed since things seem to be working.  I may open another issue for tweaking the default profiles (thanks again for the info @cityeyes) so that they work better in the average case.

Thanks for all of your help!

closed 2018-04-10 by FormerLurker

nickm324, 2018-04-10:
have a couple of question about the stabilization options.
1. Does these extra movements add a substantial amount of time to a print.
2. Is there a setting that would not use stabilization at all so that extra movements are not included and the OctoLapse just takes the snapshot no matter where the print head is.

edsvcs, 2018-04-10:
@FormerLurker Happy to post the videos. Where should I put them? The forum software won't allow .mp4 to be attached here.

ninjamojo, 2018-04-10:
@edsvcs  Maybe youtube?
@nickm324 :

> 1.  Does these extra movements add a substantial amount of time to a print.
It can calculate the exact time for you (check Settings->Octolapse->Main Settings) if you want, but generally it does not add a substantial amount of time.  Somewhere between 1 and 3 seconds per snapshot is pretty typical, though you can get it down below 1 second in some cases if your settings are tweaked for the sliced gcode.
>2. Is there a setting that would not use stabilization at all so that extra movements are not included and the OctoLapse just takes the snapshot no matter where the print head is.
Yes, you can edit the stabilization settings and turn off stabilization for one or both axes.  I find timelapses with y stabilization but no x stabilization to be pretty interesting.

I recommend looking through each setting (I know there are tons) to get a feel for what is possible.  At some point I will put up guides, but I have a full time job and am already spending all of my free time on this project, lol!  Check out the [wiki here](https://github.com/FormerLurker/Octolapse/wiki) for more details.  I'm trying to keep it up-to-date, but things are changing so fast it's difficult to keep up!

nickm324, 2018-04-10:
Cool, I will check out the wiki.

FormerLurker, 2018-04-10:
I thought I would add to @ninjamojo's comment and say that the snapshot delay (camera profile settings) can add a lot to this time.  My camera nearly works with 0ms delay, but not quite (a few frames not stable), so I've settled with 125MS delay.  Some people have reported their cameras require 1-3 seconds delay!  Obviously there is not much Octolapse can do about snapshot lag from any individual camera, but it should be taken into account when discussing how much time Octolapse can add to a print.