FormerLurker/Octolapse #109: [Request] Add a new type of stabilization : Corner (?)

gege2b, 2018-03-30:
Hi there
I'm discovering Octolapse (great plugin, good job !)
I came with a suggestion

#### If this is a feature request describe it here

TLDR;
Add a "Corner of print" option in the Stabilization tab so the snapshot is triggered only when the head/bed is at the min (or max) X/Y position of the current layer

Let me explain : in the early days of 3D printing, someone created a Skeinforge plugin to control a Canon DSLR to mke timelapse

https://eclecti.cc/bytes/reprap-controlled-time-lapse-photography

And one of the available options gives pretty good result : Corner. The goal is to take the snapshot when the head is in the corner of each layer. 

The end result show the head at (mostly) the same position during the timelapse. Visually, it's similar  to the "fixed" stabilization but with one neat advantage : the head don't need to be moved at all for the snapshot => no waste of time, less retraction and an no impact on the print quality

if this could help, here is the snippet doing the maths :  
https://github.com/nrpatel/Sprinter/blob/bf676ab3a5d375517891117cc32f1efde6af3158/photograph.py#L134

I don't know if something like this could be easily doable but it would be great to have such an option! 

#### Version of Octolapse
 
Octolapse Version: 0.2.2

#### Version of OctoPrint

OctoPrint Version: 1.4.0.dev926+gfb4e95bf

FormerLurker, 2018-03-30:
Believe it or not, you could use that script with octolapse as is!  Check out the GCode trigger.  You could set the snapshot command in the printer profile to m240 and run your GCode through the plugin you mentioned.

Currently octolapse doesn't do any pre-processing, so adding this feature will take some doing.  I already have plans to do this, though, and have ideas for other pre-processing features.

Thanks for submitting the idea, and let me know if you need help with the GCode trigger!  To my knowledge nobody has tried this feature yet.  In fact, it was the first trigger I wrote, and I haven't tried it in a long time, so there could be some bugs that have crept into the code.

Also, take a look at the position restrictions.  Depending on your print, you can control where octolapse takes snapshots more precisely.  The 'calculate intersection' feature is really cool too, but takes a bit more processor power (not too much depending on the number of position restrictions).

gege2b, 2018-03-30:
Oh damn ! 
Ok I'll give a look there thank you for pointing me this
I guess I could use the Slic3r post-processing scripts to add the M240 (or whatever) to the exported GCode and voilà

I admit I don't have looked at all the feature of Octolapse yet
Let's experiment :)

FormerLurker, 2018-03-30:
I decided to throw everything but the kitchen sink into Octolapse's settings for better or for worse.  I figure the default profiles are good enough for beginners, and the rest of us might appreciate the more subtle settings :)

BTW, be careful when you test the gcode trigger.  Use some small thin wall test or something first.  I seriously haven't used it in a long time, lol!  Maybe I'll do a test today.

gege2b, 2018-05-15:
Hi there, back from some testing

Well, it seems to work like a charm !
The only caveat I found is that apparently, we can't use ";" in the GCode command and that's mainly why I take so much time to post back here (being busy at work, too)

You can [get here](https://pastebin.com/0G7A8FJS) the python code I use to insert the GCode command using slic3r's post processing script, but it could be used as a standalone script too

I'm not a pro python so it's ugly; but it just works :)

I'll close this feature request as this is not relevant anymore !

(oh, and BTW, I *LOVE* the Test mode logging ! )

closed 2018-05-15 by gege2b