FormerLurker/Octolapse #110: Can't make it startr

jmceara, 2018-03-30:
Hi. I have installed this plugin in the latest vesion of OctoPrint, but I can't make it work. All settings looks fine and I can see plugin enabled, all options are checked, but nothing happens when print starts. Lookingat log's, I found a looooooootttt of this line in the octolapse log file:

`2018-03-30 01:48:17,841 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:22,844 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:27,847 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:32,850 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:37,853 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:42,868 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

2018-03-30 01:48:47,871 Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 958, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'
`

But I don't know how to fix this. Any idea?

jmceara, 2018-03-30:
I almost forgot: my printer is an CoreXY custom made, running smoothieware firmware and printing from OctoPrint (not SD Card).

FormerLurker, 2018-03-30:
It looks like you probably are using Octoprint v 1.3.6 or lower.  Try upgrading to v1.3.7rc3.  You can do this by going into the OctoPrint settings->Software Update.  Click on the wrench icon.  Make sure 'OctoPrint version tracking' is set to Release and OctoPrint Release Channel is set to 'Maintenance RCs'.  Save the changes, and you should be prompted to upgrade to 1.3.7rc3.  Do this and reboot when prompted.

Now it's probably a good idea to test drive v1.3.7 rc3 for a while before you try Octolapse to make sure everything is working properly.  You'll want to disable octolapse before doing this.  You can disable Octolapse from Settings->Octolapse in the 'Main' tab.

Once you're done trying out the new version of OctoPrint, enable octolapse and follow the [steps after installation.](https://github.com/FormerLurker/Octolapse/wiki/Installation#steps-after-installation)  It's a good idea to read through the wiki before use.  

Let me know if that helps!

BarnumD, 2018-03-31:
I had a similar issue with the latest version.  1.3.7rc3 seems to have made it start.

FormerLurker, 2018-03-31:
@BarnumD, What version of octolapse?

jmceara, 2018-03-31:
Yep, almost working now. The 'original' problem is related to octoprint version and is solved. Now, I'm getting another error.....and only one snapshoit are taken. After start printing, I got this:

```Octolapse - Position Error
Position - Coordinates (X:130.00000,Y:130.00000,Z:-15.12350,E:12.76600) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received````
My printer profile (in OctoPrint ocnfiguration) is defined with X:270, Y286, Y:300 (positive!) and I can't find where I define printer area in OctoLapse printer profiles, so my guess is that octolapse uses octoprint profile, right?
What am I doing wrong now? Also, there is any support forum? I'm sure here is not the correct place for this new problem, but I could not find where o ask for help.

FormerLurker, 2018-03-31:
It might be a gcode parsing issue since Octolapse seems to think your z axis is negative!  Can you export your settings.json (can be done in Octolapse settings screen on any tab), and post your gcode file?  You can use gist.github.com for this.  I'll run the code through the debugger and see what's happening.  I'm going to be working on the plugin today so it's not a problem.

Regarding the user forum, there isn't one at the moment unfortunately, so I'm happy to respond to anything in the issues for now.

jmceara, 2018-03-31:
even with error, seens to work fine! I have only one suggestion: separate final renderind from octoprint timelapse. If you have a lot of files/works with same name, is hard to find what you need. If is not possible to create a separated tab/list, at least put a preffix on filename, like 'OctoLapse'.

About the problem, here is the files that you asked for: https://gist.github.com/jmceara/7327b7b5bff3c5552c6b19fbfceae97a

FormerLurker, 2018-03-31:
So, you have two options.  You can disable synchronization (copying timelapses to OctoPrint timelapse folder) OR you can change the rendering filename template (it's pretty powerful).  Check out the rendering profile settings for more details, or let me know if I am not understanding.  

I'll look through that gcode as soon as I'm done with some other fixes (actually working on the rendering section right now).

BarnumD, 2018-04-01:
@FormerLurker Octolapse 0.2.3

FormerLurker, 2018-04-01:
Try updating via the software update plugin, it uninstall and reinstall.  Lots of fixes since 0.2.3 to try!  Let me know if you still have issues.

FormerLurker, 2018-04-01:
Sorry, on my phone..  I meant update OR reinstall

FormerLurker, 2018-04-04:
@jmceara I'm closing the issue.  If you need any help with the rendering template so you can tell which timelapses come from Octolapse and which come from OctoPrint send me an email (formerlurker@protonmail.com), or create another issue and I'll help you out!  Regarding your out-of-bounds error I finally debugged it.  Take out these two lines:
```
G91
G1 Z5
```
from the start gcode.  If your print volume is correct you shouldn't get any additional out-of-bounds errors.  In general it's probably not a good idea to move your z axis until you first home it.  I'm not really sure what the advantage of moving your extruder up 5mm before you home your axis is, but please let me know if you are aware of anything I am not :) If you continue to have problems open a new issue and post your gcode and settings.json file, and I'll take care of it.  Thanks for submitting the issue!

FormerLurker, 2018-04-04:
@BarnumD, if you are still having problems can you create a new issue and fill out the template?  I will definitely need your plugin_octolapse.log file.  It's best if you turn on one of the diagnostic debug profiles first and start a print.  Feel free to cancel after the first 3 or so layers are finished.  That will give me enough to figure out what's going on.  Also, make sure you first upgrade to v0.2.5 (0.3 is coming very soon, contains many bugfixes).  Thanks for adding your input to the issue!

closed 2018-04-04 by FormerLurker

BarnumD, 2018-04-04:
Not having issues since I updated octoprint

FormerLurker, 2018-04-05:
Thanks @BarnumD !  v0.3.0rc1 just dropped, it has lots of fun stuff.