FormerLurker/Octolapse #112: Octolapse not Capturing Timelapse

iaglet, 2018-03-31:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse Vv0.2.6

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.7rc3 running on OctoPi 0.14.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Configured Octolapse
2. Ran Print Job
3. Octolapse only Captured one Image

#### What should have happened?
Octolapse should have captured images every two minutes. Look, I hope whatever I am doing wrong is stupid obvious..

#### What happened instead?
Octolapse only captured one image.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14

#### Printer model & used firmware incl. version

Printer Model: HICTOP CR-10
Printer Firmware Version: Marlin V1

#### Browser and version of browser, operating system running browser

Browser: Microsoft Edge 41.16299.248.0
Browser OS: Windows 10 Home; Version 1709; OS Build 16299.309

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/N209Dcug

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/0My0VSjt

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/NBDP87M1

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-03-31:
Try this, modify your startup gcode and ADD a  right after the M83 command stick a G90 (set absolute coordinates).  It's usually a good idea to have a g90 before any commands that change the axis position (if you are using absolute coordinates).

OR you could go into your printer profile and set X/Y/Z Axis Mode to 'Default to Absolute'.  I don't really consider that option to be safe, but lots of slicers seem to leave out a G90/G91, I guess because they trust that the current mode will be the correct one.  I am more paranoid than that :)

Let me know if that helps and THANK YOU FOR FILLING OUT THE ENTIRE REQUEST..  Yes I shouted that with joy :)

iaglet, 2018-03-31:
If I add the G90 immediately after the M82, I am able to capture an Octalapse driven by time interval (say, every 30 seconds). If I try to use layer change or height change, it won't capture the images. I am really unsure the printer profile is the right one. This is a HICTOP CR-10, which I believe is a Prusa clone.

FormerLurker, 2018-03-31:
Can you attach your plugin_octolapse.log file please?  I'll probably see what is happening from that.

iaglet, 2018-04-01:
https://pastebin.com/Mhu2cQUs <-- That is the log file

https://youtu.be/t86SP-tqu5g <-- This is the generated timelapse

I used the CR-10 beta profile and what’s interesting is that at maybe 15mm it starts making captures. This is supposed to be triggered though off Layer changes.

FormerLurker, 2018-04-01:
That is easy to fix!  Go to your printer profile and set your priming height somewhere between where your printer is priming and your first layer height.  Let's say 0.5MM (that should be good, you will probably never print 0.5mm thick layers!).  Then try again and you'll have the whole timelapse!

iaglet, 2018-04-01:
Alright. Now I need some simple help. Where do I do that? Set the printing profile. Are you talking in my slicing app? I might need a touch of guidance.

FormerLurker, 2018-04-01:
So, go to settings, Octolapse, then click on the printers tab.  Find your currently selected printer, then click on the edit pen.  You can find the priming height setting there (be sure to save the profile once you're finished).

[I recommend that you look at the wiki here regarding where to find the printer profile.](https://github.com/FormerLurker/Octolapse/wiki/Printer-Profiles)  If you look there are a couple expandable sections (click to see the tab, click to see the profile) that will show you pictures that are more helpful than a description!

I recommend that you look through the rest of the settings too while you're there.  Also, look through all of the other profiles to get a feel for what kinds of options you have.  

Let me know if you need any additional help, and do let me know if you get it working!

iaglet, 2018-04-01:
Thank you. I need to play around with this. The printer raises the extruder to 15mm for priming. Which now explains what I was seeing in the timelapse. The captures didn’t start until 15mm. In fact, the first photo happens with the bed clear, after the extruder is raised to 15mm.

I will have to take a look through this.

iaglet, 2018-04-01:
I just want you to know, this is resolved. I was able to print an Octolapse, and I believe I may have also improved print quality settings in the process. I still have some things I need to address: 1) I am seeing the same issue reported in a separate thread where the first line prints before Octolapse takes the snapshot. It is always one line that prints, then the photo. 2) I am not sure it's an issue but I set the trigger to layer changes and it is taking two photos before the print starts. I need to scrub through the gcode on this test print and see where the 2nd trigger is occurring. 3) I had to make changes to the profile and basically tell Octolapse to use the previous speeds for all the movements. I am not sure the best way to determine what the right speeds are and I am afraid the printer is not explicitly updating the speeds. So if I tell Octolapse to use a speed, I am afraid it's going to keep printing at that speed after the snapshot.

But the nutshell is it's working: https://youtu.be/NEewxwwImrI

FormerLurker, 2018-04-01:
Yeah, it's probably because by default the layer trigger forbids snapshots on detract.  That's why it draws the line first before taking a snapshot, because the state of the extruder at the start of the line was Detracted=True.  

If you are sure you won't run into problems with detract + immediate retract (depends on the print and slicer settings, material, etc..), just go to the Snapshot profile settings and look at the Layer/Height trigger settings.  Expand 'Advanced Layer/Height Trigger Options' and set the three 'Detract' drop downs to ignore, and make sure 'Extrusion Starting', 'Extruding' and 'On Extruder Primed' are all set to Trigger.  While you're in there scroll to the very bottom under the advanced settings and check out the position restrictions.  This is another way to control exactly where and when snapshots are taken.  

You can see that there are lots of options for adjusting when Octolapse takes a snapshot.  Most of this is a compromise between quality and gcode compatibility.  If you want 100% control of when snapshots are taken, use the Gcode trigger and add the command:

Snap

to the file where you want snapshots taken (the exact code is configurable in the printer profile).  Warning:  This trigger has not been tested as much as the others.  Please use caution.

Let me know if that makes sense!

FormerLurker, 2018-04-01:
Hey, I also forgot to mention:  It is important that you set the correct speed for each axis else your quality will suffer and the speed of the travels/retract/detract made by octolapse will be of an unpredictable speed.  

Octolapse (is supposed to) keeps track of the current speeds/axis mode (relative/absolute), position, etc and makes sure they are reset back before your gcode continues.  If you have problems with your feedrate while using Octolapse please report this as a separate issue.  It's certainly possible (no software is bug free), but I've not had problems with this reacently.

Thanks!

iaglet, 2018-04-02:
I am going to mark this closed. The initial issue I reported is not an issue. It's an RTFM issue. I did not have success with tweaking the snapshot profile so that it would print immediately on the layer change, but I found that using the gcode trigger and using a search and replace to edit the layer number comment, I was able to successfully add the snap tag. I also found that the CR-10 profile actually worked with the only tweak being to set the priming distance to 15mm and redo my "start" gcode so that the G90 comes immediately after the extruder is primed.

closed 2018-04-02 by iaglet

FormerLurker, 2018-04-02:
Awesome!  I hope to simplify setup in the future.  I'm also going to set the priming height to 0.5mm for this profile in the future.  Glad things are working!

dosel79, 2019-02-17:
I was also having only taking first layer snapshot
Going to printer setting and matching the profile did the job
Thank you so much always

varoonm, 2019-03-05:
Hi FormerLuker....i am having a similar issue where octolapse is only capturing one image for the entire print. I am using CR-10s and have already set-up G90 .  Priming height in Octolapse is set to .75mm. Can you please take a look at my Gcode file to analyze what might be going wrong here. Thanks

[CCR10_Reb_Data_Stick.txt](https://github.com/FormerLurker/Octolapse/files/2932294/CCR10_Reb_Data_Stick.txt)

FormerLurker, 2019-03-05:
@varoonm,

Looks like you're running the gcode trigger probably.  Using 'G4 P1' is not the standard snapshot command also.  Having your settings.json file would be most helpful, btw.

Have you tried the 'Layer Change Compatability' trigger?  If not, I recommend trying that before moving to the gcode trigger.