FormerLurker/Octolapse #114: MP4 is flipped and upside down

makaper, 2018-04-01:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.2.5

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

#### What should have happened?
Rendered MP4 should have been correctly oriented.

#### What happened instead?
Rendered MP4 was upside down and flipped.

#### Operating System running OctoPrint and Octolapse

OS Name:Windows
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: MK3
Printer Firmware Version: Latest

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-04-01:
So it sounds like Octolapse does what you want it to, right?  The control window is showing you the current stream, but Octolapse only cares about snapshots.  Also, Octolapse does post processing on the snapshots to rotate them according to the camera profile settings (in Octolapse).

Have you checked here:

[image]

If this ends up being a bug, you should report it in the OctoPrint issues for sure.  No need to mention Octolapse in this case since I don't believe the two are related (yet, maybe I'm missing something).

Thanks for reporting your issue!

makaper, 2018-04-01:
Whether or not this is an Octolapse or an Octoprint bug is a matter of
perspective. I would say that octoprint "obeyed" the settings to flip the
image wheras Octolapse did not obey them. FWIW I still haven't gotten
octolapse to capture snapshots in the correct orientation!

Thanks for making an awesome tool! Sorry for being one!

FormerLurker, 2018-04-01:
I plan on allowing multiple cameras in the future, so it's not a good idea to rely on the Octoprint settings for that to work since you could have 2 cameras each with a unique orientation.  It may be annoying to configure, but.... it's a feature?   Lol.

FYI, I'm a bit confused on one part.  Are you using the 'Transpose' feature in Octolapse's camera profiles to get the proper orientation, or is it just right without that setting?  FYI, you can also transpose the rendering separately in the rendering settings.

makaper, 2018-04-01:
Prior to installing Octolapse, I had a fresh Octopi running on a pi 3B with
a pi cam v2 from adafruit. I experienced a flipped image (x and y) in the
live webcam stream and also in my (regular stock) Timelapse mpegs.

So then I flipped the image again (x and y) in the Octoprint settings and
this remedied my problems in the control tab and also the stock timelapse
tab.

But then I installed OctoLapse (which looks awesome btw) except that it has
always resulted in a flipped image and mp4 video.

Curiously, it seems that when I reboot the Pi now, it seems to always
forcibly change the settings back to having the image flipped (x and y).

So, to be more clear, if I.

1. Deselect the check boxes in OctoPrint settings for webcam x and y axis
flip.
2. Restart OctoPrint after Octolapse is installed.
3. Go back into OctoPrint settings and look. Now then settings I deselected
in step 1 are Selected again.

So no matter what I do OctoLapse will not let. E flip the orientation!

Best, Andreas

FormerLurker, 2018-04-01:
Octolapse does not obey these settings because it will support multiple cameras in the future.  There is a transpose option in the OCTOLAPSE camera profiles (settings->Octolapse->Camera) that should fix this.  See this: 
[image]

makaper, 2018-04-01:
Great! I’m not in front of my stuff right now but look forward to trying it
tomorrow.

Thanks!

FormerLurker, 2018-04-03:
@makaper , did this solve your issue?

FormerLurker, 2018-04-03:
> Yes it did. Working great now. I had an “out of print area” bug when doing
> a large MK3 print but I’ll submit a proper bug with logging etc if it comes
> up again.

That's great!  Regarding the out of bounds error, I think the mk3 x and y
axis start at -3mm, so you may want to check your octoprint printer
profile, or you can override it in octolapse printer profiles.  I should
probably add that to the default settings huh?

closed 2018-04-04 by FormerLurker

astronyu, 2024-10-02:
> Octolapse does not obey these settings because it will support multiple cameras in the future. There is a transpose option in the OCTOLAPSE camera profiles (settings->Octolapse->Camera) that should fix this. See this: [image]

Sorry to revive this thread. Is there a way to enable all three? (flip vertically, horizontally and left-right)
I have my camera mounted upside down so that I could have it point down more. I have set correct orientation in octoprint and my camera is showing correct but right now octolapse still showing upside down.