FormerLurker/Octolapse #116: Does Octolapse bites with ABL?

Blacksheep70, 2018-04-01:
#### If this is a feature request describe it here

Does Octolapse works with Auto Bed Leveling?

#### Version of Octolapse
 
Octolapse Version: Octolapse V0.2.5

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.6 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Enable Octolapse
2. Start print
3. Octolapse does not recognize that the print is working

#### What should have happened?
Octolapse shoud gather snapshots

#### What happened instead?
Nothing. The Octolapse screen is same as when print is inactive

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.14.0 

#### Printer model & used firmware incl. version

Printer Model: Creatlity 3D 10
Printer Firmware Version: Lates Marlin

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Win 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Does not matter. Happens with different Gcode

FormerLurker, 2018-04-01:
Yes it does.  The only printer I own is a Prusa MK2s, and it has ABL.  I'm guessing something else is going on, so I'll need some of the info that was requested in the issue template to solve this:

1.  plugin_octolapse.log (very important).  Please turn on one of the diagnostic profiles before you start a print, let it run for a few layers after the problem occurs, then download the .log and post it somewhere.  Please don't truncate if possible (I know it gets really big) since there's good information right at the beginning of the print!
2.  gcode file (also very important, even if it seems like it isn't)
3.  settings.json file (export from octolapse settings page)

Thanks!

Blacksheep70, 2018-04-01:
Thanks for the  feedback. I will manage the data collection by tomorrow

Blacksheep70, 2018-04-02:
Here you go:
https://blacksheeps.my-wan.de/nextcloud/index.php/s/4Mmw4gxJqeGKKqD

Note: The Nextcloud is running on an Cubietruck (Raspberry like). So it's not the fastest ;)

As suggested I just printed few layers and stopped the print.

FormerLurker, 2018-04-03:
I've seen this issue a few times now, and it's caused by a missing G90/G91 command.  Your startup gcode assumes your printer is always using absolute coordinates for XYZ movement, but I don't think this is safe to assume (though I could be mistaken), so I recommend doing the following:

Go into your slicer and edit your print start gcode.  Find these 2 lines:
M82 ;absolute extrusion mode
G28 ;Home

Replace those lines with these new ones:
G90 ; Set XYZ to absolute mode
M82 ;absolute extrusion mode
G28 ;Home

Alternatively if you don't want to mess with your start gcode, go into the Octolapse printer profile and set your XYZ axis to 'Absolute'.  For now I don't recommend this if you can avoid it.  

I may switch the default profile for the CR10 to absolute since I have seen this issue a few times.  It seems to be an issue with the default slicer profile.  I'm going to try to do some tests and see what happens if someone sends a G91 right before starting a print with with no G90 in the start gcode.  I suspect the XYZ axis will rampage around in a most unpleasant way.

You can see the problem in the position state info panel here:

[image]

Blacksheep70, 2018-04-03:
I am confused ;)

Cura 3.2.1 has in the profile for the CR-10 the following start gcode:

```
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
```

However, I followed your advise and changed the start gcode to the following:

```
G90 ; Set XYZ to absolute mode
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
```

However, no success. I uploaded a new set of data into the provided share. I also made a screen cast which should make more clear what the problem is: Octolapse is just doing nothing.

Maybe the start gcode in Octoprint itself plays a role? I need that start code to switch off the bed heating before ABL is running to get rid of the magnetic field during the ABL measurement.

By the way: Do you accept donations? Would be a pleasure for me to support your effort.

Blacksheep70, 2018-04-03:
Additional observation: The buildin time lapse function still works, even Octolapse is activated. I have no clue if it's normal.

FormerLurker, 2018-04-03:
That is unfortunate.  I suspect a parsing issue in that case.  I am in the middle of a re-write of the parsing engine.  The old method had issues that couldn't be solved so I switched approaches entirely.

However, assuming you were NOT using one of the 'Full Diagnostic' debug profiles during the last print (the one where you posted the logs and gcode), please try turning one on (test mode is cool, read about it [here](https://github.com/FormerLurker/Octolapse/wiki/Debug-Profiles#test-mode)  and re-running a test of the first few layers, this will generate data in the plugin_octolapse.log file, which is what I really need to debug this.  If you DID run the print in one of the diagnostic modes, please export and upload the plugin_octolapse.log file and I'll probably be able to diagnose it pretty quickly.

Finally, when you run the test print (if you do) try enabling the 'Position State' info panel and take a snapshot of it after the first layer has completed.  If you want you could turn on the 'Trigger' info panel too, it's fun to watch :)  Word of caution:  there is a known issue with the info panels.  They can generate a LOT of network traffic.  It can be enough to slow the print down depending on your network and processor.  This has been fixed already (rate limit of 1 per second), but the branch isn't ready for general use yet .

And, yes, the built in timelapse function works even when Octolapse is running, though it generates a series of somewhat confusing popup messages.  Making these more clear is on my list.

THANK YOU for offering a donation!  I'm going to setup a patreon account.  Enough donations and I'll be able to put more time into this plugin and others that are in the works.  This one was just a warm-up :)

Finally, I noticed the German in the gcode comments.  Sind Sie Deutscher?  Ich war einmal da ein austausch Student, aber das ist eine lange Zeit her.  Ich habe fast alles vergessen.

FormerLurker, 2018-04-03:
> Yes, I'm German, living close to Heidelberg ;) For the fact that you
> "forgot everything" your German language is still like a native speaker.
> Just only one gammar mistake: "Austauschstudent" instead of "austausch
> Student" ;-)
>
> If you dare to download the Screen Cast I made you see at 6:19 that Full
> Log was enabled during that print.
>
> The position button (and the other buttons as well) just change their
> color but nothing else happens. Seems that something in the code is saying
>
> if (user == 'blacksheep')
> then{
>    no_operation();
> }
> else {
>    doOctolapseStuff();
> }
>
>
> ;-)
>
> So I will be patient and follow your updates. Let me know if you want to
> debug the problem on my Octopi (in case of you don`t have family and you
> are feeling bored ;-) ). I can manage ssh access for you.
>
> Thanks for your efforts and don't forget to publish if you had setup your
> patreon account

Whoops :)  You live in a beautiful part of the country!  I've been to
Heidelberg 3 times and will be there again in June!  It will be my first
time back in the country in nearly 20 years, so I'm excited.  Will be in
Berlin, Basil, Freiburg, Stuttgart (went to this Uni), Heidelberg and
Frankfurt.  Might go to a few other places if time allows.

FormerLurker, 2018-04-03:
OH, and I nearly forgot to talk about the issue.  I TOTALLY MISSED that
folder on nextcloud.  Doah!  Downloading now (might take a while, lol!)

Blacksheep70, 2018-04-03:
Upload stream on my side is 20 Mbit. So should be in a feasible time downloadable.
Yeah, Heidelberg and the region is indeed beautiful. If you need tips where to go and where to stay dont hestistate to ask (in 20 years there might be changes :D)

FormerLurker, 2018-04-03:
Ok, so I see there is an octoprint.log file.  Can you also upload the file:  plugin_octolapse.log, or is it not getting generated?  If you ran with the debug profile, it should be full of data.

Blacksheep70, 2018-04-03:
Done

Blacksheep70, 2018-04-03:
An educated guess :-D

```
2018-04-03 18:10:04,460 Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 986, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'
```

FormerLurker, 2018-04-03:
Well, that was easy :)  This error:

2018-04-03 17:14:52,274 Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 986, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
KeyError: 'tags'

This is caused by running Octoprint V1.3.6 or lower.  I added a dynamic version check to 0.3.0, but it's not out yet unfortunately.  I don't know why I can't specify a minimum version in setup.py (I can for everything else), but that's another story.

So, go here:
[image]

Then change your settings like so:
[image]

Then reboot and try again.  Earlier I solved a problem you WOULD have had after fixing this one, so that's good!

Blacksheep70, 2018-04-03:
Yay! I owe you a beer! Or two...

Thanks a lot!

FormerLurker, 2018-04-03:
You are most welcome!  I have one of those refillable Growlers from some bar in Heidelberg somewhere, maybe from Vetter's?  I wonder if they would still honor it if I find it :)

FormerLurker, 2018-04-03:
Let me know if you start getting snapshots and I'll close the issu.

Blacksheep70, 2018-04-04:
I did a quick test yesterday but had issues (might be my own stupidness). It looked like Octolapse did not recognized the layer changes. But I was finally to tired to start again. So I will test this evening again and report back. Hopefully everything is fine and we can close this one. Exptect my results at roundabout 22:00 CEST

Blacksheep70, 2018-04-04:
We can close this issue as it is solved. Octolapse still does not act on layer change but this seems to be another issue. I will do some research on my owen and open a new issue if necessary.

Thanks again. Feel free to drop me a mail (m.marre web.de) if you are in Heidelberg. I always pay my duties ;)

closed 2018-04-04 by Blacksheep70

FormerLurker, 2018-04-04:
I'll probably take you up on the offer.  The wife and I do love German beer.  Thanks for self-closing the issue and please do open another issue if you have any additional issues!