FormerLurker/Octolapse #119: wait for rendering after cancel print job ?

fred2088, 2018-04-03:
plugin : 0.25
- cancel print job
- stop octolapse
- can't playback cancelled job, it wait for print job complete before rendering ( but the job is cancel ...)

FormerLurker, 2018-04-03:
Sorry, that is a most annoying bug. I've actually already fixed it, but the
branch isn't ready for release.

If you are breve, you could try installing from the devel branch which
contains the fix (and many others).  Otherwise you should get an update
notification when v0.3.0 drops.

FormerLurker, 2018-04-05:
Wow, I found like 8 more causes of that bug and fixed it.  Two issues were caused by missing a parameter within a json string (how embarrasing, lol), so those were straightforward.  The others were mostly caused by edge.  I now send a server status update after every printer/connection/client connection state change.  Should take care of it.

V0.3.0rc1 should fix this problem, hopefully in all cases.  [Read about it here.](https://github.com/FormerLurker/Octolapse/issues/121)  Thanks!

ralxnder, 2018-04-09:
I'm running V0.3.0rc1 and the problem persists. After canceling a job, Octoprint becomes unresponsive. I cannot control my printer or send a cooldown command. I have to reboot the Octoprint server.

FormerLurker, 2018-04-09:
So, first try installing RC3 (I've been running a bug-fix marathon) from here:

`https://github.com/FormerLurker/Octolapse/archive/v0.3.0rc3.zip`

There were a few more issues that I fixed involving cancel and disconnect.

Hopefully you won't have the issue again, but if you do try clicking the 'disconnect' button.  Then wait for disconnect and reconnect.  I think this might be slightly less annoying than having to reboot Octoprint!

There are some [known cancel errors](https://github.com/foosel/OctoPrint/issues/2509) in Octoprint at the moment, so make sure you also try cancel/emergency stop without Octolapse if you continue to have problems.  This way we can rule out (or implicate) the plugin.

Thanks again for submitting your issue!

ralxnder, 2018-04-09:
It still happens with rc3. If I pause my print, Octoprint pauses but doesn't honor the pause gcode that I have setup in Octoprint and I do not get a Resume button. The printer still goes off into never never land, but disconnnecting and reconnecting was quicker than rebooting the server.

FormerLurker, 2018-04-09:
This is somewhat mysterious.  I will do more debugging and will get back
with you.  In the meanwhile it might be useful if you could do a print with
Octolapse enabled while using one of the Diagnostic profiles.  Then you can
post your plugin_octolapse.log file, which should help me debug the issue
(hopefully).  In any case I'll be starting and cancelling a lot of prints
in the meanwhile to see if I can find yet another issue in the cancel
process.

Thanks again!

ralxnder, 2018-04-09:
After playing around with this some more, I think it may be a Octoprint bug and not Octolapse at all. I need to run some more tests to be definite though.

FormerLurker, 2018-04-09:
Well that would be good for me, so definitely let me know.  If it ends up being an Octoprint bug, we should let Gina know about it, especially if the issue hasn't been discovered yet!  I do know that one fix is in the works that won't drop until Octoprint 1.3.8, since it involves changing some pretty brittle code in the serial communication library.

If you determine the cause, I wouldn't mind looking through the OctoPrint issues and see if there is a workaround.

ralxnder, 2018-04-09:
Yeah, as far as I can tell, the problem is with Octoprint 1.3.7. If I pause a print and then cancel it, the UI goes into a mode where you really can't do anything. It reports a state of Cancelling but I can't control the printer at all.  Refreshing the page I can select Preheat again but that is it.

FormerLurker, 2018-04-10:
I'm going to close this issue since most of the pause issues I hear about seem to be Octoprint related.  If I find anything that contradicts this, I'll reopen the issue.

Thank you for trying the plugin and helping to make it better!

closed 2018-04-10 by FormerLurker