FormerLurker/Octolapse #120: feature request ? time based time-lapse function ...

fred2088, 2018-04-04:
is it possible to have time based time-lapse function without anyprint ? in order to try out the feature ? b'cos the 3d printer has sd card and touch lcd screen which is easy to operate ...
therefore need a time based timelapse function web interface... 

Thanks

FormerLurker, 2018-04-04:
If I'm correctly understanding you, you'd like to try out timelapse settings without actually printing anything, right?  You should give [Test Mode](https://github.com/FormerLurker/Octolapse/wiki/Debug-Profiles#test-mode) a try!  This is an experimental feature, so please watch your bed/extruder/fans to make sure none of them turn on during the test print.  The printer itself will move like it's printing, but there will be no warmup or fans, just the XYZ axis.  All extruder movement will be suppressed (hopefully?), but I STRONGLY recommend you remove your filament just in case and monitor the test just as you would a regular print.

However, Octolapse does not work when printing from the SD card on the printer.  You must print from Octoprint locally.  This is because the commands must be streamed through OctoPrint, otherwise Octolapse has no way of knowing what gcodes are being sent, and therefore no way to know when to take a snapshot or how to move the XYZE axes to stabilize the image :(  If you know of a way around this I'd like to hear about it!  Does that make sense?

FormerLurker, 2018-04-05:
@fred2088 , did this help with your question?  In my last reply I noticed I didn't mention the time based timelapse.  This feature already exists too :)

fred2088, 2018-04-06:
got it and will try it out, 

thanks a lot :)

closed 2018-04-06 by fred2088