FormerLurker/Octolapse #121: V0.3.0rc1 Feedback

FormerLurker, 2018-04-05:
Hot off the press, stable enough for testing (I hope)!  

I especially would like people who have already gotten Octolapse to work to test.  If you've never gotten it working, you're still welcome to try!

So, if you get a chance to install via the plugin manager at this URL:

and have problem, or even if you DON'T have any problems, comment here!

Thanks, and I hope you enjoy!

cityeyes, 2018-04-05:
Did a small print with Octolapse 0.3.0rc1 today with no issues at all. Haven't tested if canceling a print hangs up still, but I'll update if and when I check it out.

ejjenkins, 2018-04-05:
I ran a couple of short prints with OctoLapse 0.3.0rc1 today on my JG Aurora A3 (I3 clone), first time I have run OctoLapse on this printer.  Everything worked fine, some stringing, but I should be able to tweak the slicer settings to get rid of that.

FormerLurker, 2018-04-07:
Thanks @cityeyes  and @ejjenkins!  

I just pushed rc2 and will be closing this issue and creating a new one for that release.

closed 2018-04-07 by FormerLurker