FormerLurker/Octolapse #124: [Request] Add event hooks for using an external camera

brandonforty2, 2018-04-06:
#### If this is a feature request describe it here

When making timelapses, I use a Canon DSLR that is triggered by an arduino and relay to take the pictures.  Currently, I use the CameraStart event to send a serial command to the arduino.  This works well with the built in timelapse function of Octoprint but not with Octolapse as it doesn't trigger this event.  I have only tried to run a timelapse in test mode.  

Would it be possible to make Octolapse call an event when a picture is taken so external cameras can be used?

Information on events: http://docs.octoprint.org/en/master/events/
#### Version of Octolapse
 
Octolapse Version: Vv0.3.0rc1

#### Version of OctoPrint

OctoPrint Version: 1.3.7rc4 on Octopi0.13.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-04-06:
You are not the first person to mention adding camera hooks of some kind, but you are the first to request that I utilize the built in event.  I will look into it for sure.  Here are some initial thoughts:

1.  Octolapse does some post-processing (transpose, rotate, flip, create thumbnail, etc), but I can't think of any way to get the image from your SLR back to Octolapse.  I guess all of that stuff needs to be skipped.
2.  What about rendering?  Would you do this yourself?  It doesn't seem like Octolapse could render your timelapse in this case.
3.  I guess you'll need some delay between sending the event and Octolapse continuing the print.  I suppose the snapshot delay setting would still work for this.
4.  Octolapse currently does a camera check before starting a print, and if it's not working the print is cancelled.  I guess this would need to be disabled too.
5.  I'm wondering what, if anything Octoprint itself does with CameraStart event.  For example, when I send a MovieDone event, this triggers several things on the client.  Hopefully the CameraStart event won't interfere with anything that's already happening with the built-in timelapse functionality.

I guess this boils down mostly to settings.  How do I change the camera and rendering profiles without making it confusing for users.  Some other people wanted GPIO functionality for things like triggering an external flash, and such, so maybe I need to think about these things together.

That being said, I'd LOVE to capture some ultra high res timelapses.  Maybe there could be a way to use an IR emitter to trigger the camera?  I believe my SLR can be triggered via IR.

Thanks for the suggestion!

brandonforty2, 2018-04-06:
It wouldn't really need to use the post-processing and rendering as I can do all of that.  It also depends on the camera for getting the pictures from it to the plugin.  Mine cannot interface through USB while being used as it is pretty old.  

Camera check wise, maybe you could add an external camera option as a way to skip the camera check and disable post processing as well as rendering.  While testing, It wasn't much of a problem for me as I still had a webcam plugged into the pi to get around that.  I also set it to not render the pictures because I wasn't to worried about what the webcam was seeing.  

I also tried controlling the camera with an IR emitter but I couldn't really get it to work properly.  I don't think it was able to emulate the timings properly, but I was using an Arduino Uno to control the LED.  A Raspberry Pi would probably be able to do it more accurately.

rdu, 2018-04-27:
I would like to have this feature as well - although my use case is different.

My custom made led-lamp is controllable via mqqt. I don't want to have the lamp turned on all the the time, but only for the snapshot. It would be nice to have a event which fires (with a delay) before the snapshot is taken to switch the lamp on and after the snapshot to turn it off.

FormerLurker, 2018-04-27:
@rdu, that's interesting.  I've been thinking a lot about this, and I'm leaning towards implementing two separate solutions:

1.  GPIO pin control - turn gpio events on and off based on Octolapse events (snapshot starting, snapshot complete, etc...)
2.  Bash/Bat file execution - run bat/bash scripts based on Octolapse events.  We could build a library of default scripts that could be customized.  One of these could utilize mqqt.

rdu, 2018-04-30:
@FormerLurker I don't know about the octoprint api's, but would'nt it be possible to send an event to the octoprints internal event system? I use the MQTT-Publish plugin which forwards at least a lot (if not any) events to the mqtt server for further processing. So I receive a messages like this:

"{"_event": "CaptureStart", "file": "/home/pi/.octoprint/timelapse/tmp/Light_20180430050834-560.jpg", "_timestamp": 1525070515}"

octoprint/event/CaptureDone : msg.payload : string[125]
"{"_event": "CaptureDone", "file": "/home/pi/.octoprint/timelapse/tmp/Light_20180430050834-560.jpg", "_timestamp": 1525070515}"

That works great for my processing. The only downside is: there is no pre delay, because my lamp takes about 300ms from event to full brightness.

FormerLurker, 2018-05-01:
There is a delay setting for snapshot acquisition that applies after the stabilization motion, but before the snapshot is taken, I think this will account for it.

There is already an event being generated at the start and end of the stabilization/snapshot procdess, so it shouldn't be difficult to add one before and after the snapshot is taken.  I'll look into this a bit more in the next couple of weeks.

FormerLurker, 2018-08-11:
I know this thread is a bit old, but I did add the ability to execute a custom bat/bash after stabilization (with an optional delay).  It's in the latest commit on the devel branch.  I haven't yet integrated with OctoPrint's built in events, but I figured you still might be interested.

brandonforty2, 2018-08-11:
Thank you so much, I can't wait to test it!

FormerLurker, 2019-10-12:
So, this is a much belated update, but camera scripting is now fully supported since V0.3.4, and has been further enhanced in V0.4 alpha.  I'm going to go ahead and close this down, but feel free to create a new issue if you feel there is any unfinished business here.

Thanks for participating everyone!

closed 2019-10-12 by FormerLurker