FormerLurker/Octolapse #125: [Request] add custom gcode before snapshot

rooiejoris, 2018-04-06:
#### If this is a feature request describe it here
Is it possible to send some custom commands before taking a snapshot...?!
I am very enthusiastic about some more options when making time lapses, that's why I wanted to download this plugin. But not really in the first place to make [motion] stabilized time lapses, but stabilizing the light would be great!!
Would it be possible to turn on the light before making the snapshot and switch it of after the snapshot...  : ) This could be done with the gcode command M42 P... S...
Maybe with optional time on before snapshot and keep on after snapshot. Keep in mind the gcode buffer. [Especially with with long line segments, so maybe a field with buffer length is interesting as well?!]

Maybe there is an overlap with issue #124, mot sure 

also referring to issue #86 is it possible to add a 'generic' printer profile? I also want to use the plugin on a printer with bed size 1000x600mm... : )

#### Version of Octolapse
Octolapse Version: V0.3.0rc1

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.3.7rc4

And now the compliments: This Plugin seems very promising...!!!! Thanks for your effort...!
And although I probably will not use all the parking and hopping features (maybe later : ) I really like something where you can tune more things! The position of the head and bed is already pretty stable with the printers I work when the platform doesn't move [Ultimakers] and you use 'on z change'. 
 
cheers / joris

FormerLurker, 2018-04-06:
Hi!  There had definitely been some interest in controlling light.  The buffer isn't a problem because I need to issue an M400 and receive a response before any snapshots occur.  There is already a snapshot delay, which is the time between receiving a response from M400 and acquiring a snapshot, so I think that would take care of it, no?  I'll try to think of some way to add some start/end gcode capabilities to the settings.  The settings are always the hard part :)  

Regarding the 'Generic' profile, you can just add a new profile anytime you want.  You can also copy an existing profile.  Make sure you looking [here](https://raw.githubusercontent.com/FormerLurker/Octolapse/master/Extras/Wiki/assets/img/settings_tab_printer.png) and not at the main tab for the add/copy/delete functionality.  You are free to modify the bed size within Octolapse (if you enable override), or via the OctoPrint printer profile.

For fun you should try out the parking features a few times on some non-critical prints.  It's pretty satisfying.

Thanks for the ideas and the praise!  I will leave this open whilest I ponder.

rooiejoris, 2018-04-07:
> For fun you should try out the parking features a few times on some non-critical prints. It's pretty satisfying.
I will...!

>  There is already a snapshot delay, which is the time between receiving a response from M400 and acquiring a snapshot, so I think that would take care of it, no?

I think so... don't really know how it works, because you also want to make sure that the lights are turned off AFTER the picture is made
here is the difficulty that gcode doesn't work with time, but with lines...

>  I'll try to think of some way to add some start/end gcode capabilities to the settings. 
I was wondering if it should belong to a printer setting or not...?! In my case it is printer specific, but not sure in other cases, worth some discussion....

> The settings are always the hard part :)
agree, usability [userfriendlyness] is important...!! (speaking as a product design engineer, I am not a programmer...: )

FormerLurker, 2019-10-12:
Sorry I didn't update this earlier, but Octolapse has supported before and after snapshot scripts since V0.3.4 (further enhanced in V0.4 alpha).  The script settings are available within the camera profile, and you can have as many camera profiles running simultaneously as you like!

I'm going to call this one solved.  Thanks for posting and let me know if you have any questions!

closed 2019-10-12 by FormerLurker