FormerLurker/Octolapse #126: V0.3.0rc2 Feedback

FormerLurker, 2018-04-07:
The next [release candidate](https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.0rc2) is out!  If you have problems new problems, old problems, or no problems at all, please let me know.

Thanks for helping me out!

iaglet, 2018-04-08:
I am trying to see if I did something wrong but the gcode triggered octolapse profile is no longer capturing snapshots. I want to do some more poking before opening a ticket.

FormerLurker, 2018-04-08:
Did you try rc1?  If not maybe it is a problem with the new parser and the
snapshot command.  I will try it out today.

iaglet, 2018-04-08:
v0.3.0rc1

FormerLurker, 2018-04-08:
Ahh. Ok.  Feel free to submit a ticket, but I'm on it either way!

FormerLurker, 2018-04-08:
Ok, this was a a definite bug.  This is fixed on v0.3.0rc3.  I haven't added a release yet, but you can install via the plugin manager from this url:

https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip

Let me know if this fixed your problem.  If so I'll drop rc3.  If not more debugging!

Thanks for reporting!

iaglet, 2018-04-08:
Initial result is promising. It is now capturing images. This is a two or three hour print so I will update when done.

https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FFormerLurker%2FOctolapse%2Farchive%2Frc%2Fmaintenance.zip&data=02%7C01%7C%7Cd814e3f1ce3f4caf151708d59d6551d5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636587980843479314&sdata=ah%2F2Z3ZXDNt%2B9GiSVQTh07izWiz0cFGcqqrZEC4NP7Y%3D&reserved=0>

iaglet, 2018-04-09:
The octolapse was successful.

________________________________
From: Stephen Willson <s.r.willson@outlook.com>
Sent: Sunday, April 8, 2018 4:48:21 PM
To: FormerLurker/Octolapse; FormerLurker/Octolapse
Cc: Comment
Subject: Re: [FormerLurker/Octolapse] V0.3.0rc2 Feedback (#126)

Initial result is promising. It is now capturing images. This is a two or three hour print so I will update when done.

https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FFormerLurker%2FOctolapse%2Farchive%2Frc%2Fmaintenance.zip&data=02%7C01%7C%7Cd814e3f1ce3f4caf151708d59d6551d5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636587980843479314&sdata=ah%2F2Z3ZXDNt%2B9GiSVQTh07izWiz0cFGcqqrZEC4NP7Y%3D&reserved=0>

FormerLurker, 2018-04-09:
Thanks for letting me know!  Will add the new rc tomorrow.

FormerLurker, 2018-04-09:
Thanks for posting your results @iaglet !  I just dropped the stable version, so will be closing this issue.

closed 2018-04-09 by FormerLurker