FormerLurker/Octolapse #129: Logitech C270 and CR-10 - Not displaying video

Cowski4444, 2018-04-09:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse V0.2.6

#### Version of OctoPrint

OctoPrint Version: Version 1.3.7rc4

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

I tried all these settings with the settings on the "Timelapse" tab set to **OFF**/**Timed**/**Level Changed**
1. Used CR-10 Beta printer profile.  Did a Test and this came back with the message that it appears to be working.  NOTE:  My webcam works fine in the "Control" tab. 
2. Stabilization: Fixed - Extruder at Center. -- No changes to default settings.
3. Snapshot: Layer Change -- No changes to default settings.
    NOTE:  I also tried "Every 00:30 to no avail."
4. Rendering:  MP4 - 15FPS.  
5. Camera: Used my C270 setting.  NOTE:  I tried "Default Webcam" and "Logitech Webcam" with no results.
6.  Trigger State: ON | Extruder State: ON | Position State: ON | Position Changes: ON
   NOTE:  I tried with these values OFF as well.  No results.
7.  Ran the test per request in "Test Mode" on the DEBUG settings.

#### What should have happened?
I believe when I run a print I should get video.  Also I should get an incrementing "Snapshot Count".  That is staying at 0.

#### What happened instead?
No video.
No incrementing "Snapshot Count".
And after the print is done I get a error saying "OctoLapse Rendering Failed.  OctoLapse has failed to render a timelapse.  No frames were captured."

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.0

#### Printer model & used firmware incl. version

Printer Model: Creality CR-10
Printer Firmware Version: 

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Cowski4444/dcf04248d4ab1dfb429be6658595ef26.js

#### Link to settings.json

Link to settings.json with all passwords removed: Cannot find the settings.json file.  Please describe where this can be located at and I will send the file.

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Cowski4444/fb2fdecde730a808b7fd42897500aa22.js

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Cowski4444/2ef36534cfdd9fb994410c14b22d4d7d.js

#### Link to contents of Javascript console in the browser

Link to javascript console output: No UI issues other than video not showing up in OctoLapse but showing up in the "Control" tab.

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
https://pasteboard.co/HfKQKx5.png
https://pasteboard.co/HfKQXYk.png
https://pasteboard.co/HfKR8yz.png

Cowski4444, 2018-04-09:
Sorry about that.  That may be something I did wrong.  I'm attaching it to this reply.
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/1888380/plugin_octolapse.log)

Let me know if that isn't working either.  Sorry for the confusion.

Thanks again!

Cowski4444, 2018-04-09:
Sorry...I'm also using the most current version of Chrome as my browser.  Missed that part in the submission.

FormerLurker, 2018-04-09:
Ok, it looks like both the gcode file and the octoprint.log file is also messed up.  Also, the plugin_octolapse.log doesn't have very much in it.  I would have expected a lot more data in there.

Focus on getting me the gcode file for now, and if you can get more data in the plugin_octolapse.log file that would be great!  Maybe try this (slight variant from the last advice):

1. Delete the plugin_octolapse.log file from the logs setting screen in Octoprint settings.
2. Reboot OctoPrint
3. Print a few layers of a test print with the Diagnostic Debug profile enabled.
4. Stop the print and download plugin_octolapse.log and re-post.

Thanks!

Cowski4444, 2018-04-09:
Checked that area out.  I can say I didn't know that area existed so I didn't make any changes in there.
Here's what I found in there:  https://pasteboard.co/HfR7ZT8.png

I tried to delete the 1 in X2 and Y2 but when I saved it and went back in, they were there again.

FormerLurker, 2018-04-09:
It doesn't look like there are actually any restrictions enabled!  They show up in a little grid below the area in your screenshot (you can try adding one to see what happens).

Can you export and attach your settings.json file?  I'd like to see if maybe something got corrupted along the way.  You can export the file within the Octolapse settings page.  After you do this you might try restoring the default settings (also in the Octolapse settings page next to where you export the settings).  Some significant changes have been, so it wouldn't hurt to restore them anyway.

I really want to make this work for you, so hang in there!

Cowski4444, 2018-04-09:
I very much appreciate your patience and help with this. And I'm definitely hanging in there!! 

We must have done something to get it going in the right direction.  I went back to OctoLapse and found this waiting for me.
https://pasteboard.co/HfRoAHz.png

The file below is the Settings.json file but I couldn't post it in here.  Just rename it back to Settings.json and you are good to go.
[Settings.log](https://github.com/FormerLurker/Octolapse/files/1891563/Settings.log)

I reset the default settings, rebooted and the same older snapshot was still there.  Nothing I do now, can refresh that pic.  I tried F5 and tried to 0 out X2 and Y2 --> Save --> Save and same pic in there.

Incidently.  I got notification that v1.3.7 was released so did upgrade to that.

FormerLurker, 2018-04-09:
That's good, that means it took a snapshot while you were printing.  If you didn't start a print, you will not get a new snapshot.  Again, that image represents the last snapshot taken by Octolapse, and snapshots are only taken while printing.

I think I need to address this somehow, because you aren't the first one to be confused by this.  It seems fine as long as Octolapse works properly, but if it doesn't it is really a source of confusion for people.

Will you try running another test print via the 'Full Diagnostic - Test Mode' profile now that newer versions are installed and the default settings have been restored?  Try posting plugin_octolapse.log after the test completes and I'll take a look at your settings in the meanwhile.

FormerLurker, 2018-04-09:
Oh, I forgot to mention (if I haven't already before, sometimes I forget, lol), test mode prevents heating, fans and extrusion.  Unload your filament first just in case, and feel free to stop after 4 or 5 layers are complete.

Cowski4444, 2018-04-09:
Also, saw that my snapshot time calculation is disabled.  And you have a minor typo. :)  Sorry...I'm a database admin and I notice these things.  I don't like to blame OCD but I guess that's what it is.  lol
https://pasteboard.co/HfRIOmj.png

FormerLurker, 2018-04-10:
Any updates?  I'm curious if you were able to get the log file that I asked for, or if things are working now.  Thanks!

Cowski4444, 2018-04-10:
Yeah...a few messages up.  I had to put in the message as "Settings.log"  but you can rename it to "Settings.json".
Let me know if it isn't the one you are looking for.  And thanks again!!

FormerLurker, 2018-04-10:
Yes, I got the settings.json and that is helpful.  What I'm looking for now is plugin_octolapse.log. You sent it a few days ago but it didn't have anything useful in there, and looked corrupted.  I know there is a jumble in here now so I'll repost this:

1. Delete the plugin_octolapse.log file from the logs setting screen in Octoprint settings.
2. Reboot OctoPrint
3. Print a few layers of a test print with the Diagnostic Debug profile enabled.
4. Stop the print and download plugin_octolapse.log and re-post.

Thanks!

Cowski4444, 2018-04-10:
I'm sorry I missed that one.  Here is the file you requested.  And you aren't lying.  This is a much bigger file then before!  Hope this has the information you are looking for.
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/1896754/plugin_octolapse.log)

Cowski4444, 2018-04-11:
> Ok, It looks to me like there is no start gcode for setting the axis
> coordinate system (g90/g91). I just added a troubleshooting wiki a couple
> days ago and have addressed this issue.
>
> For now try going to your Octolapse printer profile settings and change
> XYZ Axis Mode to 'Absolute' like so:
>
> [image: xyzabsolutemode]
> <https://user-images.githubusercontent.com/33363128/38621008-890d1f36-3d65-11e8-96fe-16cab63cf55a.PNG>
>
> This will probably get it working for you, but I recommend adding an
> explicit G90 to your start gcode (Right above the M82 would be a good
> place) since it's possible for your printer to be in 'relative XYZ mode' at
> any given time, and it would be bad to start a print in relative mode
> without a G90 command in there.
>
> Let me know how it goes, and don't forget to star the repository if you
> think it's a good project!

I'm at work right now but will do this as soon as I get home.  Hopefully
this fixes the issue.

And count on a great rating from me regardless.  This looks to be a very
handy tool to use and look forward to taking advantage of it.

I'll keep you in the loop!

FormerLurker, 2018-04-11:
The 'play' button is for playing a preview of the timelapse.  The browser keeps (controlled by the main settings) several frames, and it animates the frames it's received on each snapshot, but ONLY if the tab is visible.  On your next print, stay on the 'Octolapse' tab, then after 10 or so layers click the play button.  You'll see what it does then.  I need to make this more obvious, and probably need to make it work even when you aren't on the Octolapse tab.

The MP4 are stored in the timelapse tab.  The 'Failed' tag on the front of the mp4 filename means you canceled the print before it finished.

17 frames at 30 FPS is a little under 0.5 seconds, so the length doesn't seem unreasonable.  It looks like you have 'Test mode' enabled, so change your 'debug' profile to 'No Logging' and try a full print.  I think it will work.

Regarding 'moving the extruder off to the side', you might want to change you Stabilization profile to 'Fixed - Extruder at back left'.  The stabilization profiles determine where the extruder will be when a snapshot is taken.

Let me know how it goes.

Cowski4444, 2018-04-12:
Okay.  I think we're in good shape!!  This is working now and I'm experimenting with it.  Thank you very much for the personalized support!!  This is generally unheard of.
I've attached a very quick, rather out of focus, video of my first test print!  As you can see it worked fine!!

Where can I go to give you high ratings on not only your plugin but your support?  I want to sing your praises on not only your support but your plugin as well.

Thanks again!!  If you need a beta-tester or "proofreader" (lol) let me know.  👍 

Keep up the great, great work and thank you again!!!

[Ring_C270_v2_20180412170530.zip](https://github.com/FormerLurker/Octolapse/files/1904307/Ring_C270_v2_20180412170530.zip)

closed 2018-04-12 by Cowski4444

FormerLurker, 2018-04-12:
It is my pleasure!  The video is looking good!  I figure if I don't get Octolapse working for people it won't get used as much!  Plus, it helps me see what needs improvement and what is confusing to people.

If you want to help I think spreading the word is the best thing!  Maybe post a few vids on youtube, reddit, facebook, some makes on Thingiverse, etc...  Link to my patreon account too since I need some $$ for new cameras, filament (I print a lot of benchys!  So many benchys....) , a logo, maybe a second printer, etc.  3d printing development is expensive :(

Here are all the relevant links:
Patreon - https://www.patreon.com/FormerLurker
Wiki: https://github.com/FormerLurker/Octolapse/wiki
PluginRepo: https://plugins.octoprint.org/plugins/octolapse/
Report Issues:  
https://github.com/FormerLurker/Octolapse/issues

I appreciate your offer for assistance!

Cowski4444, 2018-04-12:
I have a starter web page I've been working on [www.techtipsdigest.com](url)  Count on an endorsement there and I have several friends using Octoprint and hooking webcams up to it.  Count on them getting a rave endorsement from me as well.

$$$ is rough for me too, I can't become a monthly patron but I'd like to donate some money to you for your plugin and your help.  Is there somewhere I can do that?  It won't be a lot but it's something.  Let me know.
Thanks!

FormerLurker, 2018-04-12:
No worries!  I consider use and sharing to be just as good as $$, maybe better.  I'll let you know if could use another proofreader or someone with lots of DB experience.  I have been thinking about adding an sqlite db so I don't have to rely on XML or JSON files....