FormerLurker/Octolapse #130: [Request] Manual Timelapse

Railgunner13F, 2018-04-10:
#### If this is a feature request describe it here

Record a print that is running from the printer's control panel and not started through octoprint. It could be a simple manual start, timed photo, and manual stop button. That would allow you to print with Duet DWC and record time-lapse or stop motion animation. Or it could trigger from gcode available to octoprint for more automated operation.

#### Version of Octolapse
 
Octolapse Version: Octolapse Vv0.3.0rc1

#### Version of OctoPrint

OctoPrint Version: Fan off
OctoPrint 1.3.7rc4 running on OctoPi 0.14.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-04-10:
You can already trigger Octolapse with gcode!  Check the 'Snapshot' profiles for the Gcode trigger.  The actual command you use is determined by the 'Snapshot Command' (default = snap) within the printer profile.  When Octolapse receives this command it will initiate a snapshot (depending on the extruder and position restriction settings.  The default profile will just take a snapshot once it sees the snap command).

I'm not familiar with Duet DWC.  I did a bit of research on it, but still don't really understand what it is or how it works with Octoprint.  Can you elaborate?

Railgunner13F, 2018-04-11:
Duet DWC is a web interface to the 32 bit Duet controller. It allows you to upload and print your gcode, edit your configuration of the firmware, Control your machine directly, and view an ip webcam. But it wont interact with the camera or do time-lapse. I like it's interface much better than Octoprint so I don't want to use Octoprint. But I want to use Octolapse on a print started through DWC and get the best of both softwares. But it looks like I could enter some gcode into my layer change script tab in Simplify3D and Octolapse would respond to that. But I don't see what the gcode to enter would be.

FormerLurker, 2018-04-11:
The default is:
Snap

I think comments should be ok since footprint strips them.  Haven't tried.

I'd be interested in seeing a video of your process when you are finished.
Sounds really interesting.  One of the reasons i love this hobby is that
there are so many neat things to learn about!

If it doesn't work for whatever reason, let me know.

Railgunner13F, 2018-04-11:
When starting from Octoprint I get "Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves."  after heatup and home but before the extrusion starts and it just hangs. But If I turn Octolapse off it prints fine.

FormerLurker, 2019-10-12:
Whoops, I closed issue #132, but never got around to closing this issue!  Let me know if you want it reopened for any reason, and thanks for posting!

closed 2019-10-12 by FormerLurker