FormerLurker/Octolapse #133: [Request] Makergear M2 Profile

steve220, 2018-04-11:
#### If this is a feature request describe it here

Please create a profile for the MakerGear M2.

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-04-11:
Hi there!  Can you send me your slicer profile?  I can probably get you started.  Also, if you can look through the Octolapse printer profile settings and let me know ANY of the settings that you might know or be able to find easily, which will speed things along.

When we get a new profile working I'll add it to the defaults :)

steve220, 2018-04-11:
Sure, Here is the S3D profile.  I've created a custom one in Octolapse that appears to be working (at least I'm getting decent layer triggered vids), but frankly, I'm not sure of many of the settings. Is there a way to export the printer profile from octolapse? if so, I'll send you that too once I get home tonight.

[MakerGear M2 (24V V4 hotend) - PLA 205.zip](https://github.com/FormerLurker/Octolapse/files/1899086/MakerGear.M2.24V.V4.hotend.-.PLA.205.zip)

steve220, 2018-04-11:
I may have found the export button.  This is the octolapse printer profile I created.
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/1899113/Settings.zip)

FormerLurker, 2018-04-11:
Ahh, looks like you found it!  I've been on and off VPN any my local internet goes down.  I'll take a look after work, THANKS FOR THE PROFILE!

steve220, 2018-04-12:
Here is a link to my first Octolapse!
https://youtu.be/6ly0dUuk_pQ

Sadly, I don't know for sure what any of the setting are/should be.  Here is how it is currently configured:
Retraction Length: 1.6mm (I may increase this as I'm getting a wee bit of stringing)
z-hop height: 0

Axis Speed Display Units: mm per min
Movement Speed: 9000 (I just transferred this from my S3D profile)
Retraction Speed: 2400 (I just transferred this from my S3D profile)
Detraction Speed: 3000
Z Movement Speed: 1200 (I just transferred this from my S3D profile)

X/Y/Z Axis Mode: require Explicit G90/G91 (I have no idea, so I left it alone)
E Axis Mode: Require Explicit M82/M83 (I have no idea, so I left it alone)
G90/G91 Influences Extruder: Use Octoprint Settings

Default Units: Millimeters
Auto-Detect Position: Enabled is checked
Additional Location detection commands: Blank
Override Octoprint Printer Profile: Enabled is NOT checked
Abort Snapshot when out-of-bounds: Enabled is checked
Printer Position Tolerance: .01 mm
Snapshot Command: snap
Priming Height: 0mm

FormerLurker, 2018-04-12:
I love it, congratulations!!

If you adjust the retraction, do it in both Octolapse and s3d.  That's pretty much true for all the settings in the printer profile that are also in s3d.

Do you not ever use zlift?  Maybe not with your setup.  If so 0 is fine!  

The section about axis modes is sort of a safety feature to make sure Octolapse knows what coordinate system to use.  The default is to require a gcode (Octolapse makes no assumptions).  Read up on g90/g91 and m82/m83 to learn more if you care.  

You may want to turn the tolerance down to 0.001.  It shouldn't affect quality.

Thanks for putting this together!  I should have time this weekend to dig in and start to add this to the defaults.

steve220, 2018-04-12:
Thanks.
The only issue I saw is this dialog box. It occurs as the print starts and the printer head moves off to the right of the bed and extrudes some plastic to prime the extruder. The timelapse records fine though. Which setting would I tweak to avoid this?
[image]

FormerLurker, 2018-04-12:
Can you go into Settings-> Printer Profiles -> Print bed & build volume and take a snapshot of your Width/Depth/Height?  Or if you have custom bounding box enabled, take a snapshot of that.  It's probably telling you that the extruder is moving outside of this boundry.  

My guess is that your X axis is set to 200mm, but it prob should be 205.  FYI, you can also do this within the Octolapse printer profile by checking the Override Octoprint Printer Profile box and filling in the volume.

FormerLurker, 2018-04-12:
Also, been looking at your printer model online.  Wow, what a beast!

steve220, 2018-04-13:
Thanks.  I expanded the build volume X axis to 205 (from 200) and that got rid of the message.

steve220, 2018-04-14:
Hey FormerLurker. After several successful octolapses, I attempted about a 10 hour print. The print finished fine, but the web interface to octoprint indicated the server wasn't running. Thinking the timelapse was rendering, I left it for an hour or so, but no amount of refreshing the browser worked. I had to reboot the pi. I used filezilla to access the pi and found the snapshots at:
/home/pi/.octoprint/data/octolapse/tempsnapshots/HanSolo_LH100_205-PLA_152362190385
and was able to copy them to my desktop computer (see attached screenshot). 
I ended up with 1272 files taking up 81,555,557 bytes. I should be able to render a video from the jpgs.
I did not have any logging turned on, but would be interested in getting your thoughts on why that occurred and helping to troubleshoot if I can.
[image]

FormerLurker, 2018-04-14:
Hmmm.  Maybe check the timelapse tab and see if it rendered?  Someone else
reported that the rendering icon was stuck.  I'm going to debug that this
weekend.  If you find the mp4, I'll take that as confirmation that this is
related.

Also, you could run ffmpeg from the pi and try to render there.  I wonder
how long it would take...  hopefully not an hour.

FormerLurker, 2018-04-14:
Somehow I missed the fact that the server locked up!  Can you check octoprint.log?  Or better yet, post it.

steve220, 2018-04-14:
Timelapse did not render and I didn't find the mp4 anywhere on the pi. 
I was able to create a video from the jpgs and all of the jpgs seemed to be present through the very end of the print.  You can view it here:
https://www.youtube.com/watch?v=k0rlxSmwf_0
I subsequently printed a small print and everything completed and rendered appropriately.
I've attached the octoprint.log file
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/1910061/octoprint.log)

FormerLurker, 2018-04-14:
Wow, that print looked great!  I'll take a look at the log in the am and
see what I can find.

FormerLurker, 2018-04-14:
So, there's not much to go on, but the logs below seem to be of interest:
```
2018-04-14 00:53:05,201 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.26
2018-04-14 00:53:10,603 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-04-14 01:00:54,474 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 01:15:54,475 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 01:30:54,477 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 01:45:54,478 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 02:00:54,479 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 02:15:54,481 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 02:30:54,482 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 02:45:54,487 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 03:00:54,495 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-04-14 03:15:54,764 - octoprint.server.heartbeat - INFO - Server heartbeat <3
**2018-04-14 03:20:12,531 - octoprint.util.comm - INFO - Finished in 54108.732 s. ** Look at timestamp here!
2018-04-14 03:17:08,388 - octoprint.startup - INFO - ******************************************************************************
2018-04-14 03:17:08,391 - octoprint.startup - INFO - Starting OctoPrint 1.3.7
2018-04-14 03:17:08,391 - octoprint.startup - INFO - ******************************************************************************
2018-04-14 03:17:08,451 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
```

So the first line looks to be when you connected to the sever via a web browser, but the server sat idle for about 2 hours before reporting that the comm job finished, immediately before you rebooted.  In particular I noticed that the time stamps are out-of-order for those rows.  I'm not sure if that's significant.

Can you take a peek at plugin_octolapse.log and see if there are any exceptions in there?  Post them if there are.  Also, I wonder if updating to 1.3.8 made any difference.  I looked at the commits in that release and it seems to only have changed the required versions for some of OctoPrint's dependencies, so I'm not sure.

steve220, 2018-04-15:
The good news is I've done several more octolapses on the M2 and also on my Monoprice Select Mini and haven't seen a repeat of that issue.  However, the prints have been a max of 2.5 hours.  I dug out the plugin_octolapse logs. There are several on the pi.  The attached zip has the latest one and the one with the same date as the failed print.
[octolapselogs.zip](https://github.com/FormerLurker/Octolapse/files/1911690/octolapselogs.zip)

FormerLurker, 2018-04-15:
There's nothing terribly interesting in those logs.  Mostly errors from running Octolapse on OctoPrint v1.3.6, but those have obviously been solved.

I think I'm going to chalk it up to OctoPrint for the time being.  Let me know if you see a repeat, and let me know if OctoLapse was enabled or not.

FormerLurker, 2018-04-15:
Oh, one last thing, can you re-post your settings.json file now that the printer profile seems to be ironed out?  I'll extract it and add it to the defaults.

FormerLurker, 2018-04-18:
> Sure! Here you go.
> I ran into an issue on my monoprice select mini however. After aborting a
> print and restarting it, octolapse did not capture a time lapse. Has this
> been reported before?
>
> Settings.zip

Hmm.  No I haven't.  If you have a log file I'd like to see it.

Thanks for the settings!  I'll look them over soon!

steve220, 2018-04-18:
Here you go.  Not sure how coincidental the lack of a time lapse is or how valuable the following is.  This morning I had a failed print that I aborted.  The print failed because the bed didn't heat for some reason.  That's never happened before.  I raised the nozzle 30-40 mm, cleared the bed then immediately hit print again. After monitoring the first layer and making sure the bed was hot, i went to work.  The print was complete when I got home, but the only time lapse in the queue was for the failed print from the morning.  I checked for snapshots from the second print on the Pi, but none were found.
Hope you get some rest!

[octolapselogs.zip](https://github.com/FormerLurker/Octolapse/files/1922287/octolapselogs.zip)

FormerLurker, 2019-10-12:
@steve220, so I think this issue has been solved since Octolapse V0.3.4.  I have had a pretty big issue backlog (as you can tell by the late response), and never got back to updating you.  Sorry about that!

Anyway, if you have any additional problems, please create a new issue and I'll take a look!  Thank you for posting!

closed 2019-10-12 by FormerLurker