FormerLurker/Octolapse #134: No Pictures; no z homing

Leprechaun91, 2018-04-11:
Hey,

I think the problem why octolaps isnt taking any pictures is because im not homing the z axis before printing.
Therefore octolaps doenst know the z position.

Is there any posibility to solve this?

[image]

FormerLurker, 2018-04-11:
Yeah, currently you need to include a home command.  Why are you not doing this, if I might ask?  Maybe I can think of a workaround, or can work in a printer profile setting somehow if I know the details.

Leprechaun91, 2018-04-11:
Im homing; but only the x and y axis.
Because of some problems homing z I’m adapt it manually

FormerLurker, 2018-04-11:
Maybe I can add a feature that will set home=true if g92 is used?

Can you describe your current process for manually homing your z axis?

Leprechaun91, 2018-04-11:
If you start the printer it sets z = 0 automatically.
So either after finishing the print I move z back to 0 or I manually set it to the right height and „homing“ it via g92 Z 0.

FormerLurker, 2018-04-12:
Hey, will you try reinstalling octolapse from the plugin manager with the following url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

You should NOT have to uninstall first if you don't want to.

Watch your printer like a hawk after the g92 command is sent and let me know how it goes.  I'm working on a test script that will work with my printer, but I'm really strapped for time today.

Thanks!

Leprechaun91, 2018-04-12:
hey,
just watched closely.
The start is the same: z = unknown;
after finishing the first layer z still remains unknown; so this time, it doesnt recoginze z at all

FormerLurker, 2018-04-12:
So, you home X and Y, then set Z with G92 Z0 (or something similar), but the Z axis continues to report - Unknown?

Can you send me the gcode of the file you just printed?  I'd like to run it through the debugger and see what I can find out.

However, it sounds like Octolapse is able to pick up the coordinates at some point during the first layer (it can do this if it detects any absolute coordinates sent for any axis).  Did a timelapse render properly at the end of the print (or if you cancelled)?

Leprechaun91, 2018-04-12:
[Power_meter_cover.zip](https://github.com/FormerLurker/Octolapse/files/1905187/Power_meter_cover.zip)

Thats the gcode inside the zip.

Yesterday it didnt move to the middle after every layer or before; or maybe sometimes a bit random after one line

Leprechaun91, 2018-11-19:
Hey,

octolaps is still not working cuz of the missing z homing.
Do you know any solution?

FormerLurker, 2018-11-19:
Hey, OMG I never updated this, so sorry!  

If you are using the new Octolapse v0.3.4, you can try adding a

```
G92 0 ; Fake a home
```
Not sure if your printer will like that gcode.  If it complains, please let me know.  I'm going to be out of town for most of this week though, fyi.

FormerLurker, 2018-11-19:
Hey, I made a typo above (since corrected).  It's the letter O not zero, like so:
```
G92 O ; Fake a home
```

BotiB, 2019-01-08:
[camera_mount.zip](https://github.com/FormerLurker/Octolapse/files/2738587/camera_mount.zip)

This is a gcode that is without Z homing at the beginning. Have a corexy system with smoothieboard with s3d slicer and my start gcode is:
G90                            ;absolute positioning
G28 X Y                      ;Home X and Y axises
M375                          ;Load grid
G1 X150 Y150 F6000 ;Move to center of the bed
G4 P500                     ;Wait 5 seconds before seting Z offset
G30 Z0                       ;Set Z offset
G1 Z15.0 F6000         ;Move the platform down 15mm
G92 E0                       ;Prime the extruder
G1 F200 E10              ;Extrude 10 mm of filament
G92 E0                       ;Prime the extruder

FormerLurker, 2019-01-08:
@BotiB, thanks for your code, that will be helpful!  

Can you explain how exactly the scratch occurs?  Looking at your gcode at these lines:
```
G28 X Y ;Home X and Y axises
M375 ;Load grid
G1 X150 Y150 F6000 ;Does it scratch here?
```
Does it scratch at the G1 command?  If so, I'm trying to figure out how your nozzle would be close to the bed at all there, unless it started at Z=0.

Also, I see what the G30 command does, but can you explain how it operates?

Thanks!

BotiB, 2019-01-08:
In this gcode obviously is not scratching as its not homing Z, but to be able to use octolaps I have to home before G1

BotiB, 2019-01-08:
@FormerLurker, the other wierd thing is that if I put my hand when homing Z at the beginning, so basically doesn't even touches the bed, it still works fibe after that.
So Z position is taken from somewhere else, as this comes like a fake home. It looks like that all he wants is to see a Z home in start gcode. I might be wrong tho

FormerLurker, 2019-01-08:
@BotiB, OK, I think I understand.  So when you home Z, THEN it scratches at the G1 X150 Y150.  

First try to run this code manually one at a time via the octoprint terminal:
```
G28 X Y ;Home X and Y axises
M375 ;Load grid
G1 X150 Y150 F6000 ;Move to center of the bed
```

then try sending this command:
```
G92 X150 Y150 O
```
copy the printer's response out and send it to me.  Next run this (even if the previous command returned an error:
```
M114
```
again, copy the response out for me.  If the M114 reported your printer at X=150 and Y=150, Try changing this part of your start gcode:
```
G28 X Y ;Home X and Y axises
M375 ;Load grid
G1 X150 Y150 F6000 ;Move to center of the bed
```
to this:
```
G28 X Y ;Home X and Y axises
M375 ;Load grid
G1 X150 Y150 F6000 ;Move to center of the bed
G92 X150 Y150 O ; A hack to work around a potential bug
```

I may need to suppress the G92 command if only the fake O parameter appears, or at strip strip it out if multiple parameters are defined.  Your input here will help me figure out what to do, thanks!

FormerLurker, 2019-01-09:
@BotiB , I realize now that the above code will probably not work for you.  I've added a patch that you can install via the plugin manager using this URL:  

https://github.com/FormerLurker/Octolapse/archive/devel.zip

After you've installed and rebooted change your start gcode to this:
```
G90 ;absolute positioning
G28 X Y ;Home X and Y axises
G92 O ; A hack to tell octolapse that all axis are homed.  Will NOT be sent to the printer as long as Octolapse is enabled
M375 ;Load grid
G1 X150 Y150 F6000 ;Move to center of the bed
G4 P500 ;Wait 5 seconds before seting Z offset
G30 Z0 ;Set Z offset
G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0 ;Prime the extruder
G1 F200 E10 ;Extrude 10 mm of filament
G92 E0 ;Prime the extruder
```
Let me know if it works.

@Leprechaun91, you might also be interested in this.

BotiB, 2019-01-09:
@FormerLurker, so, I have half success. Octolapse does work, but...
I'm not sure how to explain, I experience this issues with the old Octolapse as well, but only with high detail organic models, that printhead moves kinda blocky/shaky, like he struggles. I put a picture where the left one is without, the right one is with octolapse enabled. On the skirt you can see the signs of the blocky movement of  the head, but when you print the whole part, is full of blobs.
Last time, always disabled octolapse to be able to print those models, thought that is to much load for the raspberry and struggles sending data fast enough. Now after this patch it does the same, but with simple models as well. Now idea whats happening, but if I switch off octolapse, prints fine.

[image]

FormerLurker, 2019-01-09:
@BotiB, what kind of pi are you using?  Stuttering is generally due to transmission problems, and Octolapse adds quite a bit of overhead.  I've not had any issues with My Pi2 or Pi3 B+.

What version of OctoPrint are you using?  The latest version will tell you if your cpu has been throttled due to undervoltage/overheating.  This is the most common issue I see.  Either that or trying to run Octolapse on a Pi Zero or (even worse) a Pi1.

BotiB, 2019-01-09:
@FormerLurker , in terminal I can see this a lot:
Send: N976 G1 Y133.364 X110.467*36
Recv: ok
Send: N977 G1 Y133.133 X110.195*45
Recv: ok
Recv: ok
Send: N978 G1 Y132.93 X109.897*27
Send: N979 G1 Y132.891 X109.824*40
Recv: ok
Send: N980 G1 Y127.964 X104.897*36
Recv: ok
Send: N981 G1 Y128.074 X104.621*33
Recv: ok
Send: N982 G1 Y128.136 X104.504*33
Recv: ok
Send: N983 G1 Y132.533 X108.9*46
Recv: ok

As far as I know this is ok

BotiB, 2019-01-09:
@FormerLurker by the way, why it adds octolapse overhead, is it not taking pictures only?
Can I disable some of the more CPU intensive aspects?

FormerLurker, 2019-01-09:
@BotiB, that's pretty unusual.  Can you try disabling all other plugins except for Octolapse?  Perhaps there is some kind of conflict going on.

I don't think there is anything you can disable really, but please send me a copy of your settings.json file (can be exported within OctoPrint settings->Octolapse.  There is a button to download), and I'll check it out.

What kind of printer are you using again?  Also, what all is plugged into your Pi's usb ports?  Are you using wifi or wired connection (wondering if there could be a bandwidth issue maybe).  Very mysterious in any case.

BotiB, 2019-01-09:
[image]
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/2742997/Settings.zip)

Above the settings and list with my plugins.
I have a corexy style custom printer and using a smothie board with it.
Pi is using wifi internet connection and its connected to smoothie with a usb cable. Also is connected to Pi a 7inch touch screen and my usb module camera, thats all.

FormerLurker, 2019-01-10:
Ill take a look.  In the meanwhile, try disabling print time genius.  If you have any logging turned on (including octolapse full diagnostic mode), turn that odd.  Ideally you would disable all non essential plugins and run another test.

FormerLurker, 2019-01-10:
I took a look at your settings and I see a few things you should change (not all of them are related to your problem).

1.  You might want to change your 'Printer Profile->E Axis Mode' setting to 'Default To Absolute' or 'Require Explicit M82/M83'.  I believe your printer defaults to absolute (if it's like the other enders).  I could be wrong here.
2.  Your 'Printer Profile->Snapshot Command' should NOT be 'G4 P1'.  Change it to something else like 'snap' or something similar.  The command should be suppressed by Octolapse, but this could confuse some other plugins.  Make sure you change this in your slicer too!  Alternatively you could just use the layer trigger (compatibility for now) until we get your problem resolved.
3.  Your 'Camera Profile->Snapshot Delay' seems high, but might be OK.  250MS is usually enough.  You can go lower still if you are running at a higher framerate.  The ideal delay is 1000/FPS, so if your camera is running at 30 frames per second your delay should be 1000/30 = about 34 milliseconds.  If your timelapse is blurry you can increase it.
4.  Disable ALL of the info panels while you are testing (trigger state, extruder state, position state, position changes).  
5.  Set your 'debug profile' to 'No Logging' OR 'Test Mode'.  Edit the debug profile and UNCHECK the enabled box under 'Log Additional Information'.  This shouldn't cause any problems, but it is better if it's unchecked.
6.  Make sure you disable logging in Octoprint too.  Delete any existing log entries (see the image below):
[image]
7.  The next time you run a print, close all browsers connected to the pi (including on your phone and any remote monitoring).  This is important because your issue could be bandwidth related (I've seen it before).  If this helps we'll debug further to solve the issue.

VERY IMPORTANT:  Did you disable all of the other plugins?  It may sound dumb, but please do this.  FYI, you do NOT need to reboot until you've disabled everything, but definitely reboot AFTER you've disabled everything.  That should save you time.  There is NO need to uninstall anything, disabling will do nicely.  Plus, this makes it easier to re-enable after we're done debugging!  I hear about compatibility issues all the time, so don't underestimate this step.  If it's an issue with another plugin I'll install it locally and see if I can figure out what's up.  I'll contact the dev if necessary.

If this doesn't work I've got some other ideas.

Also, I want you to know I've seen and dealt with this issue before, and it's almost always soved.  It's usually been one of the following issues:
1.  Bad PSU for a raspberry pi.
2.  Inadequate/no heatsink on pi, or pi is running in a very hot environment
3.  Old raspberry pi (not a problem in your case)
4.  Old/Bad/Defective/Unshielded USB cable
5.  Problem with other plugins (2 of the plugins you have installed have caused issues in the past).  These can usually be resolved.
6.  IO bandwidth issues (the PI only has so much)

We might want to open a new issue for this since your original problem has been solved.

FormerLurker, 2019-01-12:
@BotiB,  there is no doubt that Octolapse requires more CPU power than running without it.  I am planning to spend some time optimizing things and will test a few models with very high resolution (a high number of gcodes per mm).  

I'm glad I can help!  I await a progress report :)

scryharder, 2019-02-24:
I'd like to continue this thread, adding suggestions and my situation so you can see if this work around may be possible since I've been tearing my hair out for months because it worked great once, couldn't figure this requirement out, and could never find out why til today. Finally found it's this simple problem driving me crazy:

I'm using a lulzbot Taz 6, with the v3 dual extrusion upgrade. This is important because it adds time to the bed leveling and switches the Z offset a bit (and makes it worse when there's slight failures in reading the corners). It's pretty easy to set it up to just use one extruder to print if I don't want the extra colors. I can also quickly cancel a print and change the z offset or other setting through simplify 3d instead of re-leveling the bed. 

So I have a macro in Octoprint to level the bed once, then mess with scripting because dual extrusion is a pain in my book - especially as I switch back and forth between figurines, soluable supports, and rough prototypes regularly. Then I make my start gcode this:

G26                          ; clear potential 'probe fail' condition
G21                          ; set units to Millimetres
G90                          ; absolute positioning
G28 X Y                      ; home X and Y
G1 X-17 F3000                ; clear X endstop
G1 X20 Y200 Z5 F3000

(I tried adding an M114 command this morning but didn't seem to do anything to help). The printer then lowers the print heads after clearing the raised z button.

I have been using my own scripts for layer change to take over the octolapse portion:
G1 X140 Y140
snap 

It seems simple enough that you might question why I'd want to do it - but the real reason is to build those commands into my tool change scripts for dual extrusion as it's really good to do some custom wiping and I may as well combine in the timelapse capabilities. 

The problem is that I believe that the G28 z homing is throwing off my bedleveling - the taz dual extrusion has a raised button and I think that use overwrites the corner mesh leveling (at least visually it appears that way from my results). I have code in there to see the Z height, but I can't get octolapse to recognize it.

What I don't understand is why octolapse won't read the snap command without the z homing? My suggestion/request would be to allow a command or checkbox to disable the z homing requirement. Every snapshot is sequential, so it's not like it really NEEDS the z homing for any reason - when a print starts, snap 1 is the start of the timelapse, end of the print ends the snaps to make into a timelapse. Is there some reason for that requirement I can't see? Alternatively there could be some way to let it read the Z height from the code positioning as the tooltips suggest it does - but doesn't say anywhere that it doesn't without a new z probing! (The second idea seems much harder than a checkbox to ignore the z homing requirement though).

I'd love to add ideas or suggestions or anything to work with it that will make octolapse better, especially if it's easy! Or feel free to say it would cause problems with Z,X,y, and I'll try to think of some other way to do it.
Thanks!

FormerLurker, 2019-02-28:
@scryharder, Sorry for the late response, I have a bad case of the flu and have been out of commission for the last several days.

The reason Z-leveling is required is because it was an easy way to determine when the print was actually starting.  Since virtually any gcodes imaginable can be added to start gcode, and most people wouldn't want Octolapse to start taking snapshots without being homed on any axis (potentially hitting the top of the z axis, or sides of the X/Y axis, causing all kinds of problems).  Since virtually all printers home all 3 axes before printing/priming, this seemed like a reasonable thing to do.  Could it work differently?  I think it could, but I'd have to make sure it didn't negatively affect the majority of users who are homing X,Y and Z before printing.

However, there is a workaround.  It is not perfected (it has been enhanced in a development release, but that's nowhere near ready for consumption), but you could try adding this gcode to your start gocde whenever you know for CERTAIN that all axes are homed:

G92 O ; pseudo gcode to force Octolapse to believe it's been homed on all axis

Now, you probably were using the 'auto detect location' option in your octolapse printer profile, but since no homing is performed on Z, Octolapse will not know the Z position (assuming you actually do home x and y prior to adding G92 O), so you'll need to disable the 'autodetect position' option and manually enter your origin.  On many printers (mine included) the position after homing (mesh bed leveling) is not always the same, which is another reason homing is required by default.

Also, there is a reason this is a 'hidden' feature: it's not been tested very well, and was added for dealing with situations like yours.  I've not gotten enough feedback to determine if it works or not.

Lastly, Octolapse specifically does NOT work 100% with multiple extruders.  It will track your dual extruder as a single extruder, which will mess up retraction/detraction detection, will not take into account different retraction/detraction speeds, and perhaps more that I haven't even thought of.  I would try to stay away from prints where the material differences lead to big settings changes (especially in retraction/detraction speed/length) between extruders.

In any case, thanks for being brave and continuing to experiment.  The more I know about these sorts of printers and setups the more likely future versions will be more compatible.  At some point I plan on adding multi-extruder tracking, fyi.  It will probably be after the release of 1.0, however.

Ok, off to bed.  I apologize in advance if my responses are somewhat slower than usual.

scryharder, 2019-03-11:
@FormerLurker No worries on response time, I am busy working all the time, then sometimes want to mess with this or sometimes just want a working print thing to go without worrying about the camera.

I tried what you suggested with the G92 O and it didn't seem to work. I THINK I turned off auto position detection, but I can't be certain since I can't seem to find the spot again!

Let me give you a quick run down of some things I'd like to see/use this for and I think it could serve as a very easy starting point both to enable what I want and for dual extrusion lovers all over. Pardon me if I cover the basics you're well aware of.

1. So first of all, the simple way to implement this that is on our end with minimal programming. Some sort of command to initialize a timelapse print start (say G92 O), and then a command that tells octolapse to take a pic at a set time with say Snap. A far future version could add snap1, snap2, etc, or some such for multiple views/times/cameras (could be useful for error correcting or making cool dueling color/view timelapses, but that's probably only for a tiny audience), and just put them in different bins to add together. 

That's useful because then I can choose to put snap commands at layer changes (I tried having a move to X140 Y140 Snap for example). But I can also put them at tool change commands.

2. Snap shot capabilities for tool change can be really easy with adding in just a snap and a go to command. With dual or muliti tools, the software tends to pick up the tool, retract, and move it to a location aside from the piece. That eliminates the need for some of the programming you've had to do for other profiles since it's already being done in the slicer. If it's a prime pillar, the nozzle moves to a spot and starts heating, away from the print and making a nice snapshot. If it's an ooze shield, it still moves away from the print and starts heating (though tends to make less nice of a timelapse). In either case, the nozzle is moving away from the piece and becomes stationary for a time - perfect for a snapshot! So much of the work should be already done for you.

3. Or am I completely misunderstanding how octolapse and gcode triggering works for Octolapse? I thought it would be a simple matter to dumb it down a bit and do the hard thinking for it. That would prevent some of the worries of triggering that you might have?

Thanks for your response and ongoing work! I'm happy to elaborate more. And it's not like I think you will fix this overnight, I've been working against this for more than 6 months on and off and I'm just finally posting about it!

And just as a further impetus, lulzbot updated the firmware for the Taz series, pushing the autoleveling more and basically eliminating the z offset options (to detrimental effects when it's wrong!). So I keep running into issues where if I run their probe sequence, it will time octoprint out. If I offload it to a macro and get a leveled sequence, where I can then adjust a z offset in simplify3d (because they removed it from their forked cura version!!!), if I G28 Z, it removes the probed offset and my prints can fail to adhere as it thinks it's flat instead of off aligned a bit.

FormerLurker, 2019-03-11:
Thanks for your input!  I'll review the process a bit more once the latest round of changes are in place since I don't want to change too much at once :)

Regarding your timeout issue, you can go to Octoprint Settings (wrench/spanner icon), then to Serial Connection->Firmware & Protocol, then under the 'Protocol fine tuning' advanced section you can add your probe sequence command to the list of 'Long Running Commands', which should prevent your timeout.  Let me know if that helps!

scryharder, 2019-03-24:
Thanks, I ended up doing that, then working with the manufacturer to fix some problems with my leveling. Now it's not timing out, but I'm still running into the octolapse errors and problems noted. A great example is what I tried to print yesterday: https://www.thingiverse.com/thing:2797008 . I've done it before, but it's just got some flaws for many printers when printing some tiny lines that are disconnected at the beginning. So I had to stop and restart the print at least 5 times, then switched to raft which had it's own problems. Then the print managed to fail halfway because of heat warping upwards on a corner. While this isn't directly related to octolapse, it's indicative of why I don't want to re-level the bed each time when I need to wait on heating and cooling the nozzle, only to fail about a minute into the print. And when it finally works and it gets a great first layer - I don't want to jinx it and start over to get that timelapse haha.

But I'd definitely love to see it down the road sometime in the next year! 

Also, that file is a great example of something I'm wondering if it is in octolapse (offtopic, sorry): with a model like that, where it is already going to one corner of a big object, is there a way to get it to snapshot at a certain place? I know it is always going to hit a back corner of the model which will be out of the way of the camera, so it would be great to snap a picture there - no additional time or movement required. I thought there was something I saw about passing through a location but, not seeing how to use it.

Thanks for all your hard work!

FormerLurker, 2019-03-25:
Using position restrictions what you are talking about can be accomplished, but it's not exactly easy to configure.  I'm working on a more automatic way of doing this (it's sort of done, but needs testing and enhancement).  I've developed a new stabilization that will pick a position ON THE PRINT to stabilize.  It never moves away at all, only a brief retract/derectract.  Because of this the timelapse isn't perfectly stabilized, but is pretty darn close.  Think of a regular OctoPrint timelapse that magically lines up with a corner, that's pretty close to what this new mode does.  I've been able to get snapshots in under 250 ms, which means you can run Octolapse on an 800 layer print and only add 200 seconds :)  I'm working to bring that number down further too.  Unfortunately it requires that I pre-process the whole gcode file, which had been taking many minutes (over 4) for a 30MB gcode file.  I've got it down to 30 seconds on an RPI and 5 seconds on my PC, which is pretty good I think, but I'd like to get it even lower.

Also, I've been thinking about your issue while I work on the new version.  With any luck I'll find a way to make it work without any hassle for users.

FormerLurker, 2019-10-12:
@scryharder, you will be happy to know that Octolapse can now work without a home when using the new smart layer triggers that will be introduced in V0.4 (currently in alpha).  I've not tested this yet, but believe it should work.  I'm going to go ahead and close this issue, but feel free to post if you'd like to try the new version and I'll send some instructions.

Thank you for posting, and sorry for the long long delay :)

closed 2019-10-12 by FormerLurker