FormerLurker/Octolapse #135: feature request - profile for save setting ?

fred2088, 2018-04-12:
can have profiles to store related settings with different name to easy access ?

Thanks

FormerLurker, 2018-04-12:
You can create new profiles and even copy existing ones!  Go to Settings->Octolapse and select any of the profile tabs.  You'll see a copy icon next to each profile and an 'Add New Profile' button at the bottom.  Be sure to save your new profile when you are finished making changes, else it won't appear!

Here is an image of one of the profile settings screens so you know what you are looking for:

[image]

fred2088, 2018-04-13:
what I mean is to have a main profile to save all tabs profile settings, thanks

FormerLurker, 2018-04-13:
I see, like a sort of quick settings drop down that will adjust all Octolapse settings.  I will think about how this could best be accomplished.

Thanks for the suggestion!

FormerLurker, 2019-10-12:
Hey there Fred!  It's been a long while since you've created this issue, but I've been going through the backlog and wanted to give you an update.  The profile settings have taken a slightly different direction in the next version of Octolapse (v0.4), and I don't think the master profile drop down that you suggested is a good fit anymore.  However, changing the profiles has gotten a lot easier, so maybe this is now moot?  Anyway, I'm going to close this down, but if you'd like me to reconsider, let me know and i'll open this up again. 

Thanks for your suggestion!  I thought long and hard about it too, which proves it was pretty good!

closed 2019-10-12 by FormerLurker